ghdl/ghdl #2337: Bug with simple testbench involving 'open' array dimension.

creiter64, 2023-01-31:
**Description**
The attached sample causes the assertion to fail in GHDL, but manual code inspection and Questa say it should not fail.

**Expected behaviour**
The assertion passes.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.env.finish;

entity repro_tb is
end entity repro_tb;

architecture sim of repro_tb is
    type Test_Record_t is record
        test     : std_logic_vector(8 downto 0);
        expected  : std_logic_vector(8 downto 0);
    end record;

    type SLV_vector is array (natural range <>) of std_logic_vector;

    -- Want to create an array of test values without explicitely stating the size
    -- The following line breaks GHDL, if I replace "open" with "0 to 1" everything works as expected.
    constant TEST_VALUES : SLV_vector(open)(7 downto 0) := (
        x"AD",
        x"12"
    );

    constant TEST_REC : Test_Record_t := (
        test      => "0" & TEST_VALUES(0), -- The concatenation is somehow necessary to reproduce the issue
        expected  => "0" & TEST_VALUES(0)
    );
begin
    test_runner : process
    begin
        assert TEST_REC.test = TEST_REC.expected report "Should be equal" severity failure;
        finish;
    end process test_runner;

end architecture sim;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 repro_tb
```

**Context**

- OS: Windows 11
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources:`GHDL 3.0.0-dev (2.0.0.r1317.ga0321bd7a) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator
Written by Tristan Gingold.
`

**Additional context**
Putting fixed constraints to `TEST_VALUES` allows this to compile and run, but is less user friendly when values are added/removed.

tgingold, 2023-02-01:
It looks to fail only with the llvm backend, I am still investigating.

closed 2023-02-02 by tgingold

creiter64, 2023-02-06:
I can confirm the recent changes fixed this issue.

tgingold, 2023-02-06:
Thank you for the update!

commit b2cab4a6 (2023-02-02): translate: rework translate_object_subtype_indication.

Fix #2337