ghdl/ghdl #2340: Build fails for Debian

r2com, 2023-02-04:
I have installed GNAT latest version (it comes with its gcc 10.0), but I am using gcc 12.0

Inside ghdl folder I do run .configure with install prefix, but then at make I get this:
`/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: /lib/x86_64-linux-gnu/libm.so.6: unknown type [0x13] section `.relr.dyn'
/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: skipping incompatible /lib/x86_64-linux-gnu/libm.so.6 when searching for /lib/x86_64-linux-gnu/libm.so.6
/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: cannot find /lib/x86_64-linux-gnu/libm.so.6
/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: /lib/x86_64-linux-gnu/libmvec.so.1: unknown type [0x13] section `.relr.dyn'
/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: skipping incompatible /lib/x86_64-linux-gnu/libmvec.so.1 when searching for /lib/x86_64-linux-gnu/libmvec.so.1
/opt/GNAT/2021/bin/../libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ld: cannot find /lib/x86_64-linux-gnu/libmvec.so.1
collect2: error: ld returned 1 exit status
gnatlink: error when calling /opt/GNAT/2021/bin/gcc
gnatmake: *** link failed.
make: *** [Makefile:468: lib/libghdl-3_0_0_dev.so] Error 4
`

I also tried adding gcc which is packaged with GNAT install as default, but I still get same error.

I tried installing package responsible for libm.so.6 (libc i think) but still same thing.

what is the best way to sort it out?

tgingold, 2023-02-04:
If you are using debian, I suppose there is a gcc-ada or ada package that provides the compiler.  You should try with it.

Paebbels, 2023-02-04:
Looks like you installed a standalone GNAT compiler in `/opt`.

Debian has a standard package that offers you the Ada language compiler as a part of GCC. The package is called `gnat-XX.Y` in Debian. Where XX.Y is the same version as your GCC version. Both are always in sync. Some package managers call it `gcc-gnat-XX.Y`.

r2com, 2023-02-04:
alright, this issue can be closed, because I was supposed to follow "build with gcc" instruction to compile it, and as pointed by others who helped me, I was supposed to install gnat from package manager with latest version and rebuild with that.

following it, it worked OK.

closed 2023-02-04 by r2com