ghdl/ghdl #2343: ld-elf.so.1: /lib/libc.so.7: Undefined symbol "environ" (on FreeBSD)

yurivict, 2023-02-06:
**Description**

A simple hello world example fails:
```
$ ./hello_world 
ld-elf.so.1: /lib/libc.so.7: Undefined symbol "environ"
```

**How to reproduce?**

Follow https://gritbub-ghdl.readthedocs.io/en/latest/using/QuickStartGuide.html

The ```environ``` symbol has a quirk on FreeBSD that it isn't available from shared libraries. But in this case this isn't a shared library, and it should be available.

How can I reproduce the build/link commands that build ```hello_world```?

Version: 2.0.0
llvm-13
OS: FreeBSD 13.1

tgingold, 2023-02-07:
I don't have freebsd, so I cannot really investigate.
But that's weird because there is no use of the environ symbol.

Can you run ldd on the hello_world ?
Can you try to investigate where does the environ symbol is referenced ?

umarcor, 2023-02-24:
Closing as a duplicate of #2358.

closed 2023-02-24 by umarcor