ghdl/ghdl #2348: Strange linker error when trying to build with Xilinx UNIMACRO library

creiter64, 2023-02-08:
**Description**
I have a VHDL2008 testbench with VUnit that includes a module from the Xilinx UNIMACR/UNISIM library. I managed to sucessfully compile the Xilinx libraries using the [script from the repository](https://github.com/ghdl/ghdl/blob/master/scripts/vendors/compile-xilinx-vivado.sh), but when trying to run the unit tests GHDL crashes with a linker error.

This is the linker error:
```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/tmp/unimacro-min-test/build/evk__ebv-fifo_0.0.1/sim-vunit/vunit_out/ghdl/libraries/unisim\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__numeric_std__ELABORATED' does not match section name '//AArdLD'
C:/tmp/unimacro-min-test/build/evk__ebv-fifo_0.0.1/sim-vunit/vunit_out/ghdl/libraries/unisim\unisim_retarget_VCOMP.o: warning: COMDAT symbol '.rdata$.refptr.ieee__std_logic_1164__ELABORATED' does not match section name '//AArdLw'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/tmp/unimacro-min-test/build/evk__ebv-fifo_0.0.1/sim-vunit/vunit_out/ghdl/libraries/unisim\unisim_retarget_VCOMP.o:ortho:(//AArdLD[.refptr.ieee__numeric_std__ELABORATED]+0x0): multiple definition of `.refptr.ieee__numeric_std__ELABORATED'; C:\msys64\mingw64\lib/ghdl\ieee\v08\numeric_std-body.o:ortho:(.rdata$.refptr.ieee__numeric_std__ELABORATED[.refptr.ieee__numeric_std__ELABORATED]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:/tmp/unimacro-min-test/build/evk__ebv-fifo_0.0.1/sim-vunit/vunit_out/ghdl/libraries/unisim\unisim_retarget_VCOMP.o:ortho:(//AArdLw[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): multiple definition of `.refptr.ieee__std_logic_1164__ELABORATED'; C:\msys64\mingw64\lib/ghdl\ieee\v08\std_logic_1164-body.o:ortho:(.rdata$.refptr.ieee__std_logic_1164__ELABORATED[.refptr.ieee__std_logic_1164__ELABORATED]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

I also tried to just select the files needed for the `FIFO_DUALCLOCK_MACRO` macro and build them myself, but the error stayed the same. I passed the following options to GHLD:

- For analysis `-fsynopsys -frelaxed -Wno-hide` (just for the `unisim` and `unimacro`) targets
- For elab-run `-fsynopsys -frelaxed` for the testbench.

Do I need something else? Could this link error happen because I'm running the whole thing under MSYS2 instead of Linux or native Windows?

**Context**
Please, provide the following information:

- OS: Windows 11
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r1345.gbf0423ee4) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator
Written by Tristan Gingold.`

tgingold, 2023-02-08:
I think this is a know issue, that I haven't been yet able to fully investigate.

creiter64, 2023-02-09:
Anything I can do to help you find the root cause of this issue?

tasgomes, 2023-02-09:
@tgingold 

I was just about to report the same issue. Similar situation where a Xilinx library component is simulated using GHDL.

The error is exactly the same as described above. 

I can also add that the error does not happen on older versions of GHDL. I have two machines with different GHDL versions and only one crashes.

Crashing one:
```
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r870.g1cc85c578) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator
Written by Tristan Gingold.
```

Working one:
```
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r77.g4775ebbd6) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
Written by Tristan Gingold.
```

Unfortunately, it is a big gap on the versions but perhaps it still helps on finding the root cause.

tgingold, 2023-02-09:
@creiter64 according to @tasgomes, it was working with an older version of GHDL.  I suppose the issue is related to a change of llvm; so if you can rebuild a recent version of GHDL with an older version of llvm, that would be interesting.

It would also be useful to find a small reproducer (at least to put it in the testsuite).

tgingold, 2023-02-12:
I tried to reproduce the issue (using the test in #2182), without success.  I have installed ghdl with pacman (ghdl r670), I think llvm 15.0.2 is used.

creiter64, 2023-02-13:
The linked test case reproduces the issue for me too.

I have the following llvm version:
```
$ llvm-link.exe --version
LLVM (http://llvm.org/):
  LLVM version 15.0.7
  Optimized build.
  Default target: x86_64-w64-windows-gnu
  Host CPU: tigerlake
```

So I guess that shouldn't make much of a difference.

I am however running with a much newer GHDL build, from the following PKGBUILD file:
```
_realname=ghdl
pkgbase="mingw-w64-${_realname}"
pkgname='__placeholder__'
pkgver=2.0.0.r1345.gbf0423ee4
pkgrel=1
pkgdesc='GHDL: the open-source analyzer, compiler, simulator and (experimental) synthesizer for VHDL'
arch=('any')
mingw_arch=('mingw32' 'mingw64' 'ucrt64')
license=('GPL2+')
url='https://github.com/ghdl'
groups=("${MINGW_PACKAGE_PREFIX}-eda")
checkdepends=("${MINGW_PACKAGE_PREFIX}-python")
provides=("${MINGW_PACKAGE_PREFIX}-${_realname}")

_commit="bf0423e"
source=("${_realname}::git+https://github.com/ghdl/ghdl.git#commit=${_commit}")
sha512sums=('SKIP')

pkgver() {
  cd "ghdl"
  git describe --long "${_commit}" | sed 's/\([^-]*-g\)/r\1/;s/-/./g;s/^v//g'
}

build() {
  [[ -d "${srcdir}"/build-${MSYSTEM} ]] && rm -rf "${srcdir}"/build-${MSYSTEM}
  mkdir -p "${srcdir}"/build-${MSYSTEM} && cd "${srcdir}"/build-${MSYSTEM}

  if [ "$CARCH" = "i686" ]; then
    echo 'Configuring ghdl-mcode...'
    MSYS2_ARG_CONV_EXCL="--prefix" ../${_realname}/configure \
        --prefix=${MINGW_PREFIX} \
        --enable-checks \
        --enable-libghdl \
        --enable-synth \
        LDFLAGS="-static -Wl,--stack=8388608"
  fi

  if [ "$CARCH" = "x86_64" ]; then
    echo 'Configuring ghdl-llvm...'
    export CC=clang
    export CXX=clang++
    MSYS2_ARG_CONV_EXCL="--prefix" ../${_realname}/configure \
        --prefix=${MINGW_PREFIX} \
        --enable-checks \
        --enable-libghdl \
        --enable-synth \
        --with-llvm-config="llvm-config --link-static" \
        LDFLAGS="-static"
  fi

  make GNATMAKE="gnatmake -j$(nproc)"
}

check() {
  cd "${srcdir}/build-${MSYSTEM}"
  make install.vpi.local
  make test
}

_package() {
  cd "${srcdir}"/build-${MSYSTEM}
  _lib="${pkgdir}${MINGW_PREFIX}"/lib
  mkdir -p "${_lib}"
  make DESTDIR="${pkgdir}" install

  sed -i -e 's@.*\(/mingw.*\)@\1@' "${_lib}"/libghdl.link

  if [ "${CARCH}" = "x86_64" ]; then
    sed -i -e 's@^-L.*\(/mingw.*\)@-L\1@' "${_lib}"/ghdl/grt.lst
  fi

  # License
  install -Dm644 ${srcdir}/${_realname}/doc/licenses.rst ${pkgdir}${MINGW_PREFIX}/share/licenses/${_realname}/licenses.rst
}

if [ "${CARCH}" = "x86_64" ]; then
  pkgname=("${MINGW_PACKAGE_PREFIX}-${_realname}-llvm")
  makedepends=(
    "${MINGW_PACKAGE_PREFIX}-clang"
    "${MINGW_PACKAGE_PREFIX}-autotools"
    "${MINGW_PACKAGE_PREFIX}-gcc-ada"
    "git"
  )
  package() {
    pkgdesc="$pkgdesc (LLVM backend) (mingw-w64)"
    depends=(
      "${MINGW_PACKAGE_PREFIX}-gcc-ada"
      "${MINGW_PACKAGE_PREFIX}-clang"
      "${MINGW_PACKAGE_PREFIX}-llvm"
      "${MINGW_PACKAGE_PREFIX}-zlib"
    )
    options=(!emptydirs)
    _package
  }
fi
if [ "${CARCH}" = "i686" ]; then
  pkgname=("${MINGW_PACKAGE_PREFIX}-${_realname}-mcode")
  makedepends=(
    "${MINGW_PACKAGE_PREFIX}-cc"
    "${MINGW_PACKAGE_PREFIX}-autotools"
    "${MINGW_PACKAGE_PREFIX}-gcc-ada"
    "git"
  )
  package() {
    pkgdesc="$pkgdesc (mcode backend) (mingw-w64)"
    depends=(
      "${MINGW_PACKAGE_PREFIX}-gcc-ada"
      "${MINGW_PACKAGE_PREFIX}-zlib"
    )
    _package
  }
fi
```

built with: `makepkg-mingw -sCLf --nocheck`

tasgomes, 2023-02-13:
@tgingold In case it helps I also write down here the llvm version used for the working case vs failing case.

Crashing one:
```
$ llvm-link.exe --version
LLVM (http://llvm.org/):
  LLVM version 15.0.7
  Optimized build.
  Default target: x86_64-w64-windows-gnu
  Host CPU: alderlake

$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r870.g1cc85c578) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator
Written by Tristan Gingold.
```

Working one:
```
$ llvm-link.exe --version
LLVM (http://llvm.org/):
  LLVM version 14.0.0
  Optimized build.
  Default target: x86_64-w64-windows-gnu
  Host CPU: skylake
  
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r77.g4775ebbd6) [Dunoon edition]
 Compiled with GNAT Version: 11.2.0
 llvm code generator
Written by Tristan Gingold.
```

tasgomes, 2023-02-13:
@tgingold I had to setup a new workstation with GHDL today (that uses an old CPU) and for my surprise I was able to run the simulation without crashing using GHDL 2.0.0.r870. and LLVM 15.0.7.

In this new workstation, I have the following installed:

```
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r870.g1cc85c578) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 llvm code generator
Written by Tristan Gingold.

$ llvm-link.exe --version
LLVM (http://llvm.org/):
  LLVM version 15.0.7
  Optimized build.
  Default target: x86_64-w64-windows-gnu
  Host CPU: skylake
```

If you look closely, it has the same tool versions as the other setup where GHDL crashes but the "Host CPU" reported by the version of LLVM is different.

It seems that on `skylake` CPUs this error is not reproducible while on `anderlake` CPUs the problem is noticeable. As reported by @creiter64 , the `tigerlake` CPUs seems to also suffer from the same issue.

It looks to me LLVM has an issue with 11th Gen and 12th Gen Intel CPUs. This will be nasty to fix I guess.

tasgomes, 2023-02-13:
Well, I think I found what really triggers the error... and it is not the GHDL version, LLVM version or CPU version... it is actually the Vivado version.

If I use Vivado 2020.1 to compile the libraries then there is no error no matter the GHDL/LLVM version.
If I use Vivado 2022.1 to compile the libraries with GHDL 2.0.0.r870 and LLVM 15.0.7 then the error is visible!

tgingold, 2023-02-13:
Thank you for the investigation.  I can reproduce the issue.
I fear it is due to the size of `unisim_retarget_vcomp.vhd` which triggers a bug in the linker.

I am investigating.

tgingold, 2023-02-13:
It looks like an llvm bug.

I tried to generate the .s file for unisim_retarget_vcomp.vhd with ghdl1-llvm
If I assemble it with clang, the link fails.
If I assemble it with gcc, the link succeed.

So the workaround would be to use the gcc backend, or an older (and smaller) version of unisim.

Paebbels, 2023-05-06:
Please see also:
* 2182

When writing the precompile scripts, I selected `unisim_retarget_VCOMP.vhd`.  
There are also:
* `retarget_VCOMP.vhd`, and
* `unisim_VCOMP.vhd`

These have `1284`, `863` or `421` components.

Paebbels, 2023-05-14:
A bug report has been filed at binutils: https://sourceware.org/bugzilla/show_bug.cgi?id=30444

tasgomes, 2023-12-12:
FYI I upgraded to GHDL 3.0.0.r147 / LLVM 16.0.2:

```
$ ghdl --version
GHDL 4.0.0-dev (3.0.0.r147.g6c56631a7) [Dunoon edition]
 Compiled with GNAT Version: 13.1.0
 llvm 16.0.2 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

and I have no more that linker error so I think this bug has been fixed. 

I can now use GHDL 3.0.0 to compile Xilinx libraries from Vivado 2022.1.

creiter64, 2024-11-06:
I haven't hit that linker error anymore, I think this can be closed.

closed 2024-11-06 by creiter64