ghdl/ghdl #2351: Inconsistent error message formatting

he29-net, 2023-02-11:
**Description**
Error messages and warning follow the format:
`filename:line:column:type: message`
when printed to a terminal, but the changes to:
`filename:line:column: message`
(i.e. the 'warning:' or 'error:' string is missing) when printing to a file or another program.

**Expected behaviour**
Consistent formatting in both cases. This confused me for hours as I tried to customize a GHDL based linter for ALE, since the regular expression I was changing was presented with a different string than GHDL reported when running from a console.

If the difference is intentional and there is a reason for it, it could be useful to add an option to add the 'error:' and 'warning:' distinction to the output even when writing to a file, because as it stands, ALE has no choice but to report everything as an error.

**How to reproduce?**
```
[o] martin@pavelek:/tmp$ cat test.vhd 
library IEEE;
use IEEE.std_logic_1164.all;

entity demo is
	port (
		a : in std_logic;
		b : out std_logic
	);
end entity;

architecture test of demo is
begin
	a <= b;
end architecture;
[o] martin@pavelek:/tmp$ ghdl -a test.vhd 
test.vhd:13:9:error: port "a" can't be assigned
        a <= b;
        ^
test.vhd:13:14:error: port "b" cannot be read
        a <= b;
             ^
[x] martin@pavelek:/tmp$ ghdl -a test.vhd 2> output; cat output
test.vhd:13:9: port "a" can't be assigned
        a <= b;
        ^
test.vhd:13:14: port "b" cannot be read
        a <= b;
             ^
```
**Context**

- OS: Debian 11.4
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: `tarball_url`
```
GHDL 2.0.0 (tarball) [Dunoon edition]
 Compiled with GNAT Version: 10.2.1 20210110
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
  - [ ] Built from sources: `commit SHA`

tgingold, 2023-02-11:
According to errorout-console.adb, `error:` is printed if messages are colored.
This is the default when the output is a tty (so a console), and not the default when the output is a file.

Maybe knowing that is enough for you, but it looks like you want an option to always print `error:`.  Any suggestion for the option name ?

he29-net, 2023-02-11:
Thanks for the pointer; I can confirm that if I force enable colors with `-fcolor-diagnostics`, the 'error:' string consistently appears. Then I can strip the color codes with sed and parse the string without problems.

Since there is a usable workaround and it does not seem to be a feature that would be used often, perhaps just a short note about this behavior (error / warning being printed only with colors enabled) in the documentation would be sufficient?

If you prefer adding a new option to offer a cleaner solution, what about `-fdiagnostics-always-show-type` or just `-fdiagnostics-show-type`? Or `-fdiagnostics-show-msg-type` to make it more explicit.

tgingold, 2023-02-12:
If you are Ok with the option, that's fine and I won't add anything (so that your tool is compatible with older versions).  I keep that in mind.

I have also documented this behaviour.

commit 43fa0c0a (2023-02-12): doc: clarify diagnostic level output.

Close #2351
closed 2023-02-12 by tgingold