ghdl/ghdl #2353: Null Access Dereference

ncervino, 2023-02-15:
**Description**
When initializing OSVVM scoreboards whose definitions are contained in a package, GHDL fails with a `NULL access dereferenced` error.  Note that this same example runs without issue using ModelSim. 

**Expected behaviour**
The testbench should run without issue, initializing the scoreboard.  

**How to reproduce?**
Minimal working example: [ghdl_error_report.tar.gz](https://github.com/ghdl/ghdl/files/10745439/ghdl_error_report.tar.gz)

First build the OSVVM libraries:
```tcl
source <path to OSVVM>/OsvvmLibraries/Scripts/StartUp.tcl
build <path to OSVVM>/OsvvmLibraries/OsvvmLibraries.pro
```

Then build the attached sources and run the testbench:
```sh
ghdl -a -v --std=08 -P=./VHDL_LIBS/GHDL-3.0.0-dev/osvvm/v08/ pkg_a.vhd
ghdl -a -v --std=08 -P=./VHDL_LIBS/GHDL-3.0.0-dev/osvvm/v08/ pkg_b.vhd
ghdl -a -v --std=08 -P=./VHDL_LIBS/GHDL-3.0.0-dev/osvvm/v08/ atest_tb.vhd
ghdl -e -v --std=08 -P=./VHDL_LIBS/GHDL-3.0.0-dev/osvvm/v08/ --workdir=. atest_tb
ghdl -r -v --std=08 -P=./VHDL_LIBS/GHDL-3.0.0-dev/osvvm/v08/ --workdir=. atest_tb
```

**Context**

- OS: Ubuntu 22.04.1 running under WSL2
```
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy
```

- Origin:
  - [x] Built from sources: `f041824100c87e65ae8aaa92dff01988b6fbc94f`

GHDL:
```
GHDL 3.0.0-dev (2.0.0.r1369.gf04182410) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 mcode code generator
Written by Tristan Gingold.
```
Also tried with the `v2` tag, same result.

ModelSim:
```
ModelSim PE 2022.3
Revision: 2022.07
Date: Jul 18 2022
```

OSVVM: https://github.com/OSVVM/OsvvmLibraries 
Git hash:`0431512c30991304fa32ba4bf362a28888cd58a2`

**Behavior**
GHDL results: 
```
Linking in memory
Linking in memory
Starting simulation
ghdl:error: NULL access dereferenced
in process .atest_tb(behv).p1
  from: [unknown caller]
  from: work.b.scoreboard.scoreboardptype.newid
  from: work.b.scoreboard.newid
  from: process work.atest_tb(behv).p1
ghdl:error: simulation failed
```

ModelSim results:
```restart
run 1 us
simtime 
# {1 us} {0       } /atest_tb 0 0
```

Paebbels, 2023-02-15:
`NULL access dereferenced` errors are runtime errors, because a pointer (access type) contains the value `NULL` and it was used to dereference an object. This is in 99.99% of the cases a user error (testbench code or library code) but not a compiler or simulator error.

I suggest to first contact OSVVM (@JimLewis) and to investigate what are possible reasons for a null pointer error in that code section.

JimLewis, 2023-02-15:
@Paebbels   I ran it in RivieraPro and Questa and it runs fine.    

What I see are some local package instances that are referenced from outside the package using a selected name.   Cool.

JimLewis, 2023-02-17:
NewID on OSVVM scoreboards is a well tested feature.   There is nothing here that indicates this is an OSVVM error.   Especially since it is working in the other simulators.   

OTOH, this has a number of VHDL-2008 local package instances.   One of them in a package that is referenced in the design.

tgingold, 2023-02-19:
It's an issue with elaboration.  I am testing a fix for it.

commit 3ce84355 (2023-02-19): trans-chap2: elaborate dependencies of macro-expanded packages.

Fix #2353
closed 2023-02-19 by tgingold