ghdl/ghdl #2354: Integer ranges only up to 2**30

Detzi88, 2023-02-16:
**Description**
when i define a signal:
```vhdl
signal gain:                        integer     range 0 to 2**31:=2**30;
```
i get a "static range violates bounds" running ghdl.

**Expected behaviour**
I can define an integer at least up to 2**31 without issues.

**How to reproduce?**
run 
```
ghdl -a -g ./src/offsetGain.vhd
```
using the vhd provided below.

```vhd:file:
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity offsetgain is
Port (
    --Inputs
    reset_i           : in STD_LOGIC;
    mod_clk_i         : in STD_LOGIC
    );
end offsetgain;

architecture RTL of offsetgain is

signal gain:                        integer     range 0 to 2**31:=2**30;

begin
end RTL;
```

```console
ghdl -a -g ./src/offsetGain.vhd
./src/offsetGain.vhd:17:55:error: static range violates bounds
./src/offsetGain.vhd:17:55:error: static range violates bounds
```
**Context**
Please, provide the following information:

- OS: Ubuntu 20.04.5 LTS
- Origin:
  - [ ] Package manager: apt 2.0.9 (amd64)
  - [ x] Released binaries: `tarball_url`
  - [ x] Built from sources: `commit SHA`
```

diogratia, 2023-02-16:
Invoking `ghdl --disp-standard` provides
```none
type integer is range -2147483648 to 2147483647;
```
where ghdl has a low bound one less than the minimum low bound specified by the standard and a high bound equal to the minimum specified in the standard.

The subtype indication for gain high bound is 2**31 which equals 2147483648 and violates the high bound of type integer. 

In ghdl-0.37 the low bound may have been -2147483647 matching that specified by the standard in the minimum range of type integer.

ghdl has the ability to [display a line with a caret following an error message](https://ghdl.github.io/ghdl/using/InvokingGHDL.html#diagnostics-control) to point to where a reported occurs:

```none
%: ghdl -a -g offsetgain.vhdl
offsetgain.vhdl:17:61:warning: arithmetic overflow in static expression [-Wruntime-error]
signal gain:                        integer     range 0 to 2**31:=2**30;
                                                            ^
%: 
```

It depends on a fixed width font being used in the terminal display. I'd expect this feature to be enabled by default on later ghdl versions. Using ghdl-0.37 it provides no real useful information:

```none
%: ghdl -a -fcaret-diagnostics offsetgain.vhdl
offsetgain.vhdl:17:55:error: static range violates bounds
signal gain:                        integer     range 0 to 2**31:=2**30;
                                                      ^
offsetgain.vhdl:17:55:error: static range violates bounds
signal gain:                        integer     range 0 to 2**31:=2**30;
                                                      ^
%: 
```

Note the difference in diagnostic message content. The error is actually specified in the standard as being detected during elaboration and has been downgraded to a warning during analysis in later versions of ghdl along with indicating where in the declaration the error is found.

Version 0.37 from February 29, 2020 would likely be susceptible to numerous issues that have been since addressed. There have been numerous -2008 features implemented since as well. You are encouraged to use a newer version.

umarcor, 2023-02-28:
Closing since GHDL is working as expected.

Supporting 64 bit integers is being tracked in #950.

closed 2023-02-28 by umarcor