ghdl/ghdl #2355: Feature Request: Line numbers in assertion backtrace

creiter64, 2023-02-16:
**Is your feature request related to a problem? Please describe.**
I often get backtraces similar to this:

```
/usr/local/bin/ghdl:error: assertion failed
in process .bypass_tb(sim).foo_inst@foo(rtl).foo_p
  from: foo_function
  from:foo(rtl).foo_function
  from: processfoo(rtl).foo_p
/usr/local/bin/ghdl:error: simulation failed
```

If the process/function gets bigger, it's really hard to determine the exact call flow.

**Describe the solution you'd like**

It would be really helpful, if each of the lines in the backtrace had a file and line number printed.

**Additional context**
I'm also using Questa, where that seems to be the default. I have also looked through [the documentation](https://ghdl.github.io/ghdl/using/Simulation.html) and not found an option to turn that on.

tgingold, 2023-02-18:
Which backend are you using  (ghdl -v) ?

creiter64, 2023-02-21:
```
GHDL 3.0.0-dev (2.0.0.r1345.gbf0423ee4) [Dunoon edition]
 Compiled with GNAT Version: 9.4.0
 mcode code generator
```

creiter64, 2024-11-06:
With the LLVM backend there are now exact locations for assertions.

closed 2024-11-06 by creiter64