ghdl/ghdl #2357: Simulation hangs when constrain array of records using a generic

JimLewis, 2023-02-19:
**Description**
Simulation hangs when I constrain an array of records using a generic.

```
entity TbAddressBusMemory is
generic (
  NUM_INTERRUPTS       : integer := 1 
) ;
  signal InterruptRecArray : StreamRecArrayType(NUM_INTERRUPTS-1 downto 0)(   -- Does not work
    DataToModel(0 downto 0), DataFromModel(0 downto 0), ParamToModel(NULL_RANGE_TYPE), ParamFromModel(NULL_RANGE_TYPE)) ;
```

If instead the array is constrained with a literal or a range of another object (except one sized by the generic), it is ok:
```
  signal InterruptRecArray : StreamRecArrayType(1 downto 0)(   -- Works
    DataToModel(0 downto 0), DataFromModel(0 downto 0), ParamToModel(NULL_RANGE_TYPE), ParamFromModel(NULL_RANGE_TYPE)) ;
```

**Expected behaviour**
Run.  

**How to reproduce?**
With attached zip and OsvvmLibraries, after building OsvvmLibraries (requires both OSVVM and the verification IP), I ran the test by doing:

```
# analyze TestCtrl_e.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/default/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev TestCtrl_e.vhd

# analyze TbAddressBusMemory.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/default/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev TbAddressBusMemory.vhd

# RunTest TbAb_Interrupt3.vhd [generic NUM_INTERRUPTS 2]
# TestName TbAb_Interrupt3
# analyze TbAb_Interrupt3.vhd
ghdl -a --std=08 -Wno-library -Wno-hide --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/default/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev  TbAb_Interrupt3.vhd

# Simulation Start time 07:25:50
# simulate TbAb_Interrupt3 [generic NUM_INTERRUPTS 2]
ghdl --elab-run --std=08 --syn-binding --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/default/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev TbAb_Interrupt3 -gN
UM_INTERRUPTS=2
```
[GhdlHangConstrainArrayOfRecordsWithGeneric.zip](https://github.com/ghdl/ghdl/files/10777330/GhdlHangConstrainArrayOfRecordsWithGeneric.zip)

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
If only it got this far ...
```

tgingold, 2023-02-19:
I cannot reproduce the hang.  I tried with the master sources of OSVVM, but I had to slightly changes the sources of the testcase.
Even after that I have:

error: index (1) out of bounds (0 downto 0) at TbAb_Interrupt3.vhd:192
in process .***@***.***(interrupt3).geninterruptproc2
  from: work.testctrl(interrupt3).STMT_ELAB at TbAb_Interrupt3.vhd:187
  from: work.tbab_interrupt3.testctrl at TbAb_Interrupt3.vhd:72
  from: work.tbab_interrupt3
  from: Elaboration of design

Is it work in development ?
(a smaller reproducer would be helpful!)

JimLewis, 2023-02-19:
Sorry.   This is work in progress and the sources are on the Dev branch.    

Also it needs to be run with the generic, NUM_INTERRUPTS, set to 2.   Since you have a range of 0 downto 0, it appears that you are using the default value of the generic - which seems to work ok.   It seems to be impacted solely by the generic being set.   If I constrain the range manually in the testbench to either 1 downto 0 or put it at the maximum number of interrupts supported which can be extracted from gIntReq'range, then it works fine.    So it has everything to do with the range.   

ghdl --elab-run --std=08 --syn-binding --work=default --workdir=C:/tools/sim_temp/GHDL-3.0.0-dev/default/v08 -PC:/tools/sim_temp/GHDL-3.0.0-dev TbAb_Interrupt3 -gNUM_INTERRUPTS=2

I may be able to make a shorter reproducer.   I know it has to do with the array of records, but beyond that I am not sure.   Is it enough to just do the array of records or do I have to connect it up to something.   Not really sure.

tgingold, 2023-02-20:
Still using master branch of osvvm, with this simple change:

```
*** TbAddressBusMemory.vhd	Sun Feb 19 20:25:16 2023
--- TbAddressBusMemory.vhd~	Sun Feb 19 07:52:20 2023
***************
*** 329,335 ****
    ------------------------------------------------------------
    port map (
      -- Interrupt Input
!     IntReq       => IntReq(0),
  
      -- From TestCtrl
      TransRec     => ManagerRec,
--- 329,335 ----
    ------------------------------------------------------------
    port map (
      -- Interrupt Input
!     IntReq       => IntReq,
  
      -- From TestCtrl
      TransRec     => ManagerRec,
```

I get: 

```
ghdl -r --std=08 --syn-binding TbAb_Interrupt3 -gnum_interrupts=2
%% Log    ALWAYS    in Default,   IntReq(0) U    IntReq 000000000000000000000000000000UU at 0 ns
%% Log    ALWAYS    in Default,   IntReq(0) 0    IntReq 00000000000000000000000000000000 at 0 ns
...
```

What is the output of `ghdl -v` ?

tgingold, 2023-02-20:
Again:

Still using master branch of osvvm, with this simple change:

```
*** TbAddressBusMemory.vhd	Sun Feb 19 20:25:16 2023
--- TbAddressBusMemory.vhd~	Sun Feb 19 07:52:20 2023
***************
*** 329,335 ****
   ------------------------------------------------------------
   port map (
     -- Interrupt Input
!     IntReq       => IntReq(0),

     -- From TestCtrl
     TransRec     => ManagerRec,
--- 329,335 ----
   ------------------------------------------------------------
   port map (
     -- Interrupt Input
!     IntReq       => IntReq,

     -- From TestCtrl
     TransRec     => ManagerRec,
```

I get: 

```
ghdl -r --std=08 --syn-binding TbAb_Interrupt3 -gnum_interrupts=2
%% Log    ALWAYS    in Default,   IntReq(0) U    IntReq 000000000000000000000000000000UU at 0 ns
%% Log    ALWAYS    in Default,   IntReq(0) 0    IntReq 00000000000000000000000000000000 at 0 ns
...
```

What is the output of `ghdl -v` ?

Paebbels, 2024-12-19:
@JimLewis is this still relevant? Its marked with `User action required`.

JimLewis, 2025-03-01:
Not happening on current version

closed 2025-03-01 by JimLewis