ghdl/ghdl #2358: All tests fail on FreeBSD with ld-elf.so.1: /lib/libc.so.7: Undefined symbol "environ"

yurivict, 2023-02-23:
**Description**
For example, the test 000hello fails:
```
$ DEBUG=yes GHDL=/usr/local/bin/ghdl ./testsuite.sh 
analyze hello.vhdl
elaborate and simulate hello
ld-elf.so.1: /lib/libc.so.7: Undefined symbol "environ"
```

test.log is:
```
analyze hello.vhdl
elaborate and simulate hello
/usr/local/bin/ld: hello: local symbol `environ' in /usr/lib/crt1.o is referenced by DSO
/usr/local/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
/usr/ports/cad/ghdl/work/stage/usr/local/bin/ghdl: compilation error
```

Something is wrong with how the executable is built.
How can I see what command lines are used to build executables?

Revision: 7670533
OS: FreeBSD 13.1

tgingold, 2023-02-23:
Yes, this has been reported but I don't have a freebsd machine to investigate quickly.

You can see the full command line by running the test manually:
$GHDL -e -v -v hello

I don't know which library references "environ".  Is it libgrt.a ?
This symbol is not referenced on linux.

yurivict, 2023-02-23:
> You can see the full command line by running the test manually:
> $GHDL -e -v -v hello

This fails:
```
[yuri@yv /usr/ports/cad/ghdl/work/ghdl-7670533/testsuite/sanity/000hello]$ ghdl  -e -v -v hello
/usr/local/bin/ghdl1-llvm -v -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ --elab hello -l e~hello.lst -c -o e~hello.o e~hello
error: cannot find entity or configuration hello
ghdl:error: compilation error
```

yurivict, 2023-02-23:
> I don't know which library references "environ".  Is it libgrt.a ?
> This symbol is not referenced on linux.

The ```environ``` symbol in FreeBSD is special - it isn't referenced in any libraries. When encountered it is resolved by the runtime loader.

tgingold, 2023-02-23:
You first need to analyze the file: ghdl -a hello.vhdl

tgingold, 2023-02-23:
Did you have a look at https://github.com/ghdl/ghdl/issues/501 ?
In particular:
https://github.com/ghdl/ghdl/issues/501#issuecomment-352900046

yurivict, 2023-02-23:
Now I can run the compilation.

Does the project use the 'environ' symbol directly?

tgingold, 2023-02-23:
But but it is used by libc.so !

Do you think you could provide a patch for ghdl to support freebsd ?

yurivict, 2023-02-23:
> But but it is used by libc.so !

Regular programs don't fail with this error. There's something in how the executable is built that triggers this problem.

> Do you think you could provide a patch for ghdl to support freebsd ?

I am trying to find what is wrong first.

tgingold, 2023-02-23:
The executables are linked with grt.ver, which change the visibility of the symbols.

yurivict, 2023-02-23:
Removing -Wl,--version-script=/usr/local/lib/ghdl/grt.ver makes the error to go away.

yurivict, 2023-02-23:
Yes, all tests pass except for Python ones.

tgingold, 2023-02-24:
Ok, I have pushed a commit to not user --version-script on FreeBSD.
Python is a different matter.

commit 822e1031 (2023-02-24): grt/Makefile.inc: do not use version-script on FreeBSD.

Fix #2358
closed 2023-02-24 by tgingold