ghdl/ghdl #2359: Elaboration working fine one one machine, STORAGE_ERROR for same code the other

cweickhmann, 2023-02-24:
**Description**
I try to elaborate the following very simple VHDL file using vhdl 2008:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity discr_lin_block is
    port(
        clk: in std_logic;
        rst: in std_logic;
        sig_in: in std_logic_vector(1 downto 0) := (others => '0');
        sig_out: out std_logic_vector(1 downto 0) := (others => '0')
    );
end entity;

architecture behavioural of discr_lin_block is
begin
    process(clk)
    begin
        sig_out <= sig_in;
    end process;
end behavioural;
```

On `machine 1`, this works as expected:
```bash
$ ghdl -a -g --std=08 discr_lin_block.vhdl
$ ghdl -e -g --std=08 discr_lin_block
$ ls
discr_lin_block.vhdl   work-obj08.cf
```

On `machine 2`, this fails:
```bash
$ ghdl -a -g --std=08 discr_lin_block.vhdl
$ ghdl -e -g --std=08 discr_lin_block

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/home/localadmin_cweickhm/Dokumente/FPGA/GHDL/regler/
Command line:
/usr/bin/ghdl-mcode -e -g --std=08 discr_lin_block
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f275324251e 0x55e5174c924e 0x55e5174c976c 0x55e5174cb78e 0x55e5174cb7bf 0x55e5174fb27f 0x55e5174fb2fb 0x55e5174fb4da 0x55e5174f0cb7 0x55e5174d9e93 0x55e5174fb9cf 0x55e51776c7e9 0x55e517770212 0x55e5177705fc 0x55e517771f57 0x55e517771fec 0x55e5177ab2cc 0x55e5177be957 0x55e5177acbef 0x55e517797882 0x55e5177a8dbe 0x55e5177a30a0 0x55e5177e8b55 0x55e517729af5 0x55e51765f69e 0x55e5177ec4b3 0x55e517417b07 0x7f2753229d8e 0x7f2753229e3e 0x55e517416243 0xfffffffffffffffe
******************************************************************
```

This does not occur, when not using `--std=08`.

**Expected behaviour**
Both machines should behave the same.

**How to reproduce?**

Run

```bash
$ ghdl -a -g --std=08 discr_lin_block.vhdl
$ ghdl -e -g --std=08 discr_lin_block
```

with the VHDL file shown above.

**Context**
Please, provide the following information:

- `Machine 1` (working)
  - OS: Linux host1 5.15.0-30-generic <span>#</span>31-Ubuntu SMP Thu May 5 10:00:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
  - Origin:
    - [x] Package manager: `1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]`, `Compiled with GNAT Version: 10.3.1 20211117`, `mcode code generator`
    - [ ] Released binaries: `tarball_url`
    - [ ] Built from sources: `commit SHA`
- `Machine 2` (working)
  - OS: Linux host2 6.1.0-1006-oem <span>#</span>6-Ubuntu SMP PREEMPT_DYNAMIC Tue Jan 24 18:24:09 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
  - Origin:
    - [x] Package manager: `1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]`, `Compiled with GNAT Version: 10.3.1 20211117`, `mcode code generator`
    - [ ] Released binaries: `tarball_url`
    - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/home/localadmin_cweickhm/Dokumente/FPGA/GHDL/regler/
Command line:
/usr/bin/ghdl-mcode -e -g --std=08 discr_lin_block
Exception STORAGE_ERROR raised
Exception information:
raised STORAGE_ERROR : stack overflow or erroneous memory access
Call stack traceback locations:
0x7f275324251e 0x55e5174c924e 0x55e5174c976c 0x55e5174cb78e 0x55e5174cb7bf 0x55e5174fb27f 0x55e5174fb2fb 0x55e5174fb4da 0x55e5174f0cb7 0x55e5174d9e93 0x55e5174fb9cf 0x55e51776c7e9 0x55e517770212 0x55e5177705fc 0x55e517771f57 0x55e517771fec 0x55e5177ab2cc 0x55e5177be957 0x55e5177acbef 0x55e517797882 0x55e5177a8dbe 0x55e5177a30a0 0x55e5177e8b55 0x55e517729af5 0x55e51765f69e 0x55e5177ec4b3 0x55e517417b07 0x7f2753229d8e 0x7f2753229e3e 0x55e517416243 0xfffffffffffffffe
******************************************************************
```

tgingold, 2023-02-25:
So it fails with kernel 6.1 ?

Can you try to run within gdb ?

cweickhmann, 2023-02-27:
I should add: It's `ghdl-mcode` that I'm using. Don't know if that's of any use.

Here you go:

```
$ gdb /usr/bin/ghdl-mcode
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/ghdl-mcode...
(No debugging symbols found in /usr/bin/ghdl-mcode)
(gdb) run -e discr_lin_block
Starting program: /usr/bin/ghdl-mcode -e discr_lin_block
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x000055555561ae9f in ?? ()
```

I'll try two things now:
* Boot into a 5.x kernel and run it there.
* Try to compile with debug symbols.

Stay tuned.

tgingold, 2023-02-28:
If you do a backtrace with 'bt', does gdb output something interesting ?

cweickhmann, 2023-02-28:
Indeed...

```gdb
(gdb) run -e discr_lin_block
Starting program: ***/ghdl/kernel-6/ghdl-1.0.0/build/ghdl_mcode -e discr_lin_block
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x000055555561afe1 in binary_file.gen_8 (b=13) at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/binary_file.adb:364
364           Cur_Sect.Data (Cur_Sect.Pc) := B;
(gdb) bt
#0  0x000055555561afe1 in binary_file.gen_8 (b=13)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/binary_file.adb:364
#1  0x000055555564d358 in ortho_code.x86.emits.emit_const (val=54)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-x86-emits.adb:3173
#2  0x000055555564d4c4 in ortho_code.x86.emits.emit_const (val=7884)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-x86-emits.adb:3209
#3  0x000055555564d4c4 in ortho_code.x86.emits.emit_const (val=7882)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-x86-emits.adb:3209
#4  0x000055555564d61e in ortho_code.x86.emits.emit_init_value (decl=1611, val=7882)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-x86-emits.adb:3243
#5  0x0000555555642dfb in ortho_code.x86.abi.expand_init_value (decl=1611, val=7882)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-x86-abi.adb:216
#6  0x000055555562bfd7 in ortho_code.decls.new_init_value (decl=1611, val=7882)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_code-decls.adb:362
#7  0x000055555564db13 in ortho_mcode.finish_init_value (decl=1611, val=7882)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ortho/mcode/ortho_mcode.adb:142
#8  0x00005555558c3341 in trans.rtis.generate_object (decl=1430, rti=1611)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-rtis.adb:2017
#9  0x00005555558c4508 in trans.rtis.generate_declaration_chain (chain=1430, parent_rti=1610)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-rtis.adb:2264
#10 0x00005555558c5b0d in trans.rtis.generate_block (blk=1428, parent_rti=1170)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-rtis.adb:2652
#11 0x00005555558c7029 in trans.rtis.generate_unit (lib_unit=1428)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-rtis.adb:2927
#12 0x00005555558fe468 in trans.chap2.translate_package (decl=1428, header=0)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-chap2.adb:851
#13 0x00005555558fe6f5 in trans.chap2.translate_package_declaration (decl=1428)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-chap2.adb:906
#14 0x00005555558e9473 in translation.translate (unit=938, main=true)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/translation.adb:294
#15 0x00005555558fabda in trans.chap12.elaborate (config=2524, whole=true)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/trans-chap12.adb:621
#16 0x00005555558f4eee in translation.elaborate (config=2524, whole=true)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/vhdl/translate/translation.adb:2200
#17 0x000055555593a9a3 in ghdlrun.compile_elab (cmd_name=..., args=..., opt_arg=2)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ghdldrv/ghdlrun.adb:180
#18 0x000055555587b949 in ghdlcomp.perform_action (cmd=..., args=...)
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ghdldrv/ghdlcomp.adb:629
#19 0x00005555557b17e2 in ghdlmain.main ()
--Type <RET> for more, q to quit, c to continue without paging--
   lmain.adb:518
#20 0x000055555593e301 in ghdl_jit ()
    at ***/ghdl/kernel-6/ghdl-1.0.0/src/ghdldrv/ghdl_jit.adb:38
```

tgingold, 2023-02-28:
If you could also do:

 print cur_sect.all

that would be very useful.

Another point I missed: you are using ghdl-1.0, which was released 2 years ago.  You'd better to try master.

cweickhmann, 2023-03-01:
There you go:

```gdb
(gdb) print cur_sect.all
$1 = (next => 0x555555a46050, flags => 2, name => 0x555555a46038, link => 0x0, align => 2, esize => 0, pc => 8168, 
  insn_pc => 0, data => 0xffffffffffffffff, data_max => 16384, first_reloc => 0x555555a90a80, 
  last_reloc => 0x555555b73b00, nbr_relocs => 505, number => 0, vaddr => 0, seg => (
    base => (system.address) 0xffffffffffffffff, size => 16384))
```

The version question is indeed a good point. However, as mentioned above, 2.0.0 behaves exactly the same. I'll check out `master` and have a look at that as well.

cweickhmann, 2023-03-01:
Okay, here is the output with `master` at 42478d579 (checked out just now).

# Segfault

```gdb
(gdb) run -e discr_lin_block
Starting program: ***/Downloads/ghdl/kernel-6/ghdl/build/ghdl_mcode -e discr_lin_block
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x0000555555628f45 in binary_file.gen_8 (b=83) at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/binary_file.adb:365
365           Cur_Sect.Data (Cur_Sect.Pc) := B;
```

# Backtrace

```gdb
(gdb) bt
#0  0x0000555555628f45 in binary_file.gen_8 (b=83)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/binary_file.adb:365
#1  0x000055555565bed0 in ortho_code.x86.emits.emit_const (val=387)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3273
#2  0x000055555565bfb7 in ortho_code.x86.emits.emit_const (val=7793)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3300
#3  0x000055555565c196 in ortho_code.x86.emits.emit_init_value (decl=1473, val=7793)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_code-x86-emits.adb:3343
#4  0x000055555565160d in ortho_code.x86.abi.expand_init_value (decl=1473, val=7793)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_code-x86-abi.adb:233
#5  0x000055555563a4fb in ortho_code.decls.new_init_value (decl=1473, val=7793)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_code-decls.adb:362
#6  0x000055555565c80b in ortho_mcode.finish_init_value (decl=1473, val=7793)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ortho/mcode/ortho_mcode.adb:142
#7  0x0000555555730a7d in trans.chap10.create_global_const (name=1913, vtype=1486, storage=o_storage_private, 
    initial_value=7793) at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans.adb:507
#8  0x0000555555941bbe in trans.chap7.translate_static_implicit_conv (expr=7793, expr_type=5832, res_type=672)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap7.adb:93
#9  0x0000555555943d4b in trans.chap7.translate_composite_literal (str=5834, res_type=672)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap7.adb:517
#10 0x0000555555953063 in trans.chap7.translate_expression (expr=5834, rtype=672)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap7.adb:4708
#11 0x000055555598e4ad in trans.chap8.translate_report (stmt=4047, subprg=627, level=364)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:1266
#12 0x000055555598e662 in trans.chap8.translate_assertion_statement (stmt=4047)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:1322
#13 0x000055555599e990 in trans.chap8.translate_statement (stmt=4047)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:5021
#14 0x000055555599eb3f in trans.chap8.translate_statements_chain (first=4047)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:5106
#15 0x000055555598b2c0 in trans.chap8.translate_if_statement_direct (stmt=4041)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:375
#16 0x000055555598b36b in trans.chap8.translate_if_statement (stmt=4041)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:398
#17 0x000055555599e981 in trans.chap8.translate_statement (stmt=4041)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:5019
#18 0x000055555599eb77 in trans.chap8.translate_statements_chain_has_return (first=4041)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap8.adb:5119
#19 0x000055555596ffa4 in trans.chap2.translate_subprogram_body (subprg=4014)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap2.adb:683
#20 0x0000555555983de3 in trans.chap4.translate_declaration_chain_subprograms (parent=2434, 
    what=subprg_translate_spec_and_body)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap4.adb:2663
#21 0x000055555597111b in trans.chap2.translate_package_body (bod=2434)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap2.adb:979
#22 0x000055555595b882 in translation.translate (unit=920, main=true)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/translation.adb:180
#23 0x000055555596d2fe in trans.chap12.elaborate (config=2524, whole=true)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/trans-chap12.adb:625
#24 0x000055555596754e in translation.elaborate (config=2524, whole=true)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/vhdl/translate/translation.adb:2141
#25 0x00005555559b217b in ghdlrun.compile_elab (cmd_name=..., args=..., opt_arg=2)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ghdldrv/ghdlrun.adb:139
#26 0x00005555558ec4cc in ghdlcomp.perform_action (cmd=..., args=...)
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ghdldrv/ghdlcomp.adb:634
#27 0x00005555558d77ee in ghdlmain.main ()
    at ***/Downloads/ghdl/kernel-6/ghdl/src/ghdldrv/ghdlmain.adb:518
#28 0x00005555559b35ec in ghdl_jit () at ***/Downloads/ghdl/kernel-6/ghdl/src/ghdldrv/ghdl_jit.adb:40
```

# cur_sect

```gdb
(gdb) print cur_sect.all
$1 = (next => 0x555555b5ef90, flags => 2, name => 0x555555b5ef78, link => 0x0, align => 2, esize => 0, pc => 8136, 
  insn_pc => 0, data => 0xffffffffffffffff, data_max => 16384, first_reloc => 0x555555b21740, 
  last_reloc => 0x555555bba150, nbr_relocs => 498, number => 0, vaddr => 0, img_off => 0, seg => (
    base => (system.address) 0xffffffffffffffff, size => 16384))
```

tgingold, 2023-03-02:
It's weird.

Can you confirm that the same binary is working well with kernel 5, but fails with kernel 6 ?

Next action: output of:
  strace ghdl_mcode -e discr_lin_block

Thanks,
Tristan.

cweickhmann, 2023-03-02:
Yes, I can confirm: No matter what kernel was running when compiling (6.1 or 5.15), either way the same binary will run without problems on 5.15 but crash on 6.1.

# strace on 5.15 

```
$ strace ~/Downloads/ghdl/kernel-6/ghdl/build/ghdl_mcode -e discr_lin_block
execve("***/Downloads/ghdl/kernel-6/ghdl/build/ghdl_mcode", ["***/Downlo"..., "-e", "discr_lin_block"], 0x7fffa4394f60 /* 71 vars */) = 0
brk(NULL)                               = 0x55d2d25e0000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffd3b825fc0) = -1 EINVAL (Das Argument ist ungültig)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f2f8da50000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (Datei oder Verzeichnis nicht gefunden)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=235799, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 235799, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f2f8da16000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=940560, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 942344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f2f8d92f000
mmap(0x7f2f8d93d000, 507904, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe000) = 0x7f2f8d93d000
mmap(0x7f2f8d9b9000, 372736, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8a000) = 0x7f2f8d9b9000
mmap(0x7f2f8da14000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe4000) = 0x7f2f8da14000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=108936, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 110776, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f2f8d913000
mprotect(0x7f2f8d915000, 98304, PROT_NONE) = 0
mmap(0x7f2f8d915000, 69632, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f2f8d915000
mmap(0x7f2f8d926000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x13000) = 0x7f2f8d926000
mmap(0x7f2f8d92d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19000) = 0x7f2f8d92d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgnarl-10.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=288032, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 302064, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f2f8d8c9000
mprotect(0x7f2f8d8e4000, 155648, PROT_NONE) = 0
mmap(0x7f2f8d8e4000, 106496, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b000) = 0x7f2f8d8e4000
mmap(0x7f2f8d8fe000, 45056, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x35000) = 0x7f2f8d8fe000
mmap(0x7f2f8d90a000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x40000) = 0x7f2f8d90a000
mmap(0x7f2f8d910000, 11248, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f2f8d910000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgnat-10.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=3726048, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 3774416, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f2f8d52f000
mmap(0x7f2f8d640000, 1859584, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x111000) = 0x7f2f8d640000
mmap(0x7f2f8d806000, 671744, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2d7000) = 0x7f2f8d806000
mmap(0x7f2f8d8aa000, 81920, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x37a000) = 0x7f2f8d8aa000
mmap(0x7f2f8d8be000, 42960, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f2f8d8be000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=125488, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 127720, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f2f8d50f000
mmap(0x7f2f8d512000, 94208, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f2f8d512000
mmap(0x7f2f8d529000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7f2f8d529000
mmap(0x7f2f8d52d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1d000) = 0x7f2f8d52d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\237\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0i8\235HZ\227\223\333\350s\360\352,\223\340."..., 68, 896) = 68
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=2216304, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
[... cut ...]
mremap(0x7f2f8da32000, 32768, 65536, MREMAP_MAYMOVE) = 0x7f2f8da22000
mremap(0x7f2f8da22000, 65536, 131072, MREMAP_MAYMOVE) = 0x7f2f8d281000
mmap(NULL, 167936, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f2f8d258000
brk(0x55d2d277c000)                     = 0x55d2d277c000
mmap(NULL, 135168, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f2f8da1d000
exit_group(0)                           = ?
+++ exited with 0 +++
```

# strace on 6.1 (stay tuned, need to reboot)

...

cweickhmann, 2023-03-02:
Okay, uhm, 🤯

Ubuntu had a kernel update today (6.1.0-1006-oem --> 6.1.0-1007-oem), i.e. something very minor.

And now the binary is running. This *is* really weird.

Still: Thanks for all the help!

So from a crash point of view, this issue can be closed now. But I find it very dissatisfying to no know what the hell was going on.

tgingold, 2023-03-02:
Yes, it's weird.  Do not hesitate to reopen if the issue appears again.

tgingold, 2023-03-03:
The problem was:
```
mremap(0x7fcd9e81d000, 8192, 16384, MREMAP_MAYMOVE) = -1 ENOMEM (Nicht genügend Hauptspeicher verfügbar)
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x1fc7} ---
```

closed 2023-03-03 by cweickhmann