ghdl/ghdl #2360: rerror eading output signal of procedure

gilbertwaltoon, 2023-02-24:
Hi,

I have a VHDL procedure that assigns a value to an output, then later attempts to read the output :

```
procedure myproc (signal clk : in; signal a : out std_logic) is 
begin
    a <= '1';
    wait until rising_edge(clk);
    if a='1' then  -- etc 
)
```

When I try to compile I get  **error : signal interface "a" cannot be read**

Is this a bug? I thought reading outputs was legal in VHDL-2008 ?

Is there any way to downgrade the error to a warning (I tried -frelaxed but without success)

(I realise the above is bad practice, but its not my code and I can't easily modify. 
I'm running GHDL 3.0.0-dev (2.0.0.r689.gff1ef30e8) [Dunoon edition], set for VHDL-2008)

diogratia, 2023-02-25:
IEEE Std 1076-2008  
6.5.2 Interface object declarations  
>An interface object has one of the following modes:  
>>…  
>>— **out**. The value of the interface object is allowed to be updated and, provided it is not a signal parameter, read. …

Here a is a signal *parameter* of the procedure.

tgingold, 2023-02-25:
Yes indeed, GHDL is correct here.

You should change out to inout.

gilbertwaltoon, 2023-02-25:
Thanks @diogratia  

I was puzzled because Ashenden in his "VHDL2008" textbook says _"VHDL-2008 removes the restriction on reading the value of an out-mode parameter of a procedure. Out-mode parameters can be either signal parameters or variable parameters."_ 

But since you are quoting directly from the standard, you are clearly correct.  I'll close the issue.

closed 2023-02-25 by gilbertwaltoon