ghdl/ghdl #2361: raised TYPES.INTERNAL_ERROR : synth-vhdl_stmts.adb:80

smaslovski, 2023-02-24:
Hi,

While playing with the attached VHDL source I run into the following internal error in ghdl

% ghdl synth --std=08 jkff.vhd -e
/usr/bin/ghdl-mcode:note: top entity is "jkff"

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0 (Debian 2.0.0+dfsg-6.2) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/home/stas/vhdl/
Command line:
/usr/bin/ghdl-mcode synth --std=08 jkff.vhd -e
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-vhdl_stmts.adb:80
Call stack traceback locations:
0x55611be15ff7 0x55611be1846d 0x55611be1a273 0x55611be1ad63 0x55611be1dfa7 0x55611be1e228 0x55611be34d0c 0x55611be3032a 0x55611be4ab71 0x55611bdbe56c 0x55611bebe9b0 0x55611bc36857 0x7fa0e1446188 0x7fa0e1446243 0x55611bc368bf 0xfffffffffffffffe
******************************************************************

Here is the content of the file jkff.vhd:

----------------------------------------------------------------------
entity jkff is
  port (
    j_n, k_n, clk : in bit;
    q, q_n : inout bit);
end entity;

architecture behavioral of jkff is
begin
  flip_flop : process (j_n, k_n, clk) is
  begin
    case j_n & k_n & clk is
      when "010" | "011" => q <= '1';
      when "100" | "101" => q <= '0';
      when "001" => q <= q_n;
      when others => q <= unaffected;
    end case;
  end process;
  q_n <= not q;
end architecture;
----------------------------------------------------------------------

The command to reproduce the bug can be seen above. Please let me know if you need more information.

smaslovski, 2023-02-24:
Hi,

I experimented a bit and found that the problematic line is this one:

`      when others => q <= unaffected;`

If I replace it by

`      when others => null;`

the bug does not appear.

tgingold, 2023-02-25:
I am not sure why you want to have a dummy assignment in the `others` clause, that's not usual.
I will push a fix soon.

Thank you for the issue.

commit b7c5831f (2023-02-25): synth: handle unaffected in simple sequential signal assignment.

Fix #2361
closed 2023-02-25 by tgingold