ghdl/ghdl #2368: choice must be locally static expression on CASE/WHEN statement

davidel, 2023-03-01:
## Description

GHDL (v2.0 on Debian Bookworm) rejects both the following snippets, while, e.g. Vivado 2022.2 accepts both:

```VHDL
entity MatchEnt is
  port (
    A : in unsigned(4 downto 0);
    B : in unsigned(4 downto 0);
    XOUT : out unsigned(4 downto 0)
  );
end entity;

architecture behavior of MatchEnt is
begin
  tester : process (A, B)
  begin
    case A is
      when to_unsigned(17, 5) =>
        XOUT <= A + 1;
      when to_unsigned(21, 5) =>
        XOUT <= A + B;
      when to_unsigned(34, 5) =>
        XOUT <= A - B;
      when others =>
        XOUT <= resize(A * B, 5);
    end case;
  end process;
end architecture;
```

```VHDL
entity MatchEnt is
  port (
    A : in unsigned(4 downto 0);
    B : in unsigned(4 downto 0);
    XOUT : out unsigned(4 downto 0)
  );
end entity;

architecture behavior of MatchEnt is
  constant K17 : unsigned(4 downto 0) := to_unsigned(17, 5);
  constant K21 : unsigned(4 downto 0) := to_unsigned(21, 5);
  constant K34 : unsigned(4 downto 0) := to_unsigned(34, 5);
begin
  tester : process (A, B)
  begin
    case A is
      when K17 =>
        XOUT <= A + 1;
      when K21 =>
        XOUT <= A + B;
      when K34 =>
        XOUT <= A - B;
      when others =>
        XOUT <= resize(A * B, 5);
    end case;
  end process;
end architecture;
```

AFAICT the [VHDL 2008 Reference Manual](https://faculty-web.msoe.edu/johnsontimoj/Common/FILES/VHDL_2008.pdf) at section **9.4.2** seems to imply that both should be accepted (one because it is a **constant**, and the other because it is a  function in an **IEEE-2008** standard package whose inputs are all **constant**).

## Expected behaviour

GHDL should accept the posted code in *--std=08* mode.

## How to reproduce?

```Shell
$ ghdl -a --std=08 -frelaxed $VHDL_FILE
```

## GHDL Version
```Shell
davide@linux1:~/tmp/misc$ ghdl --version
GHDL 2.0.0 (Debian 2.0.0+dfsg-6.2) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

diogratia, 2023-03-02:
The issue of 9.4.2 Locally static primaries "e) A function call whose function name denotes an implicitly defined operation or an operation defined in one of the packages STD_LOGIC_1164, NUMERIC_BIT, NUMERIC_STD, NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE and whose actual parameters are each locally static expressions" has been addressed since release 2.0. 

Adding the missing context clauses does not replicate the issue for any of  the to_unsigned calls as case statement choices using either package numeric_bit or numeric_std for a more recent ghdl release (GHDL 3.0.0-dev (2.0.0.r1367.g803edb716)).

```vhdl
constant K34 : unsigned(4 downto 0) := to_unsigned(34, 5);
```
will result in the value 2, the range constraint and to_unsigned size parameter preclude the value 34.

davidel, 2023-03-03:
I am confused, the version shown above is *GHDL 2.0.0* ... is that not representative for the real GHDL version?

As for 34 on 5 bits, the value does not change the outcome:
```Shell
davide@linux1:~/tmp/misc$ !3253
for F in *.vhd; do ghdl -a --std=08 -frelaxed $F; rm -f work*; done
case1.vhd:35:12:error: choice must be locally static expression
      when to_unsigned(17, 5) =>
           ^
case1.vhd:37:12:error: choice must be locally static expression
      when to_unsigned(21, 5) =>
           ^
case1.vhd:39:12:error: choice must be locally static expression
      when to_unsigned(25, 5) =>
           ^
case2.vhd:38:12:error: choice must be locally static expression
      when K17 =>
           ^
case2.vhd:40:12:error: choice must be locally static expression
      when K21 =>
           ^
case2.vhd:42:12:error: choice must be locally static expression
      when K25 =>
           ^
```

```Shell
davide@linux1:~/tmp$ ghdl --version
GHDL 2.0.0 (Debian 2.0.0+dfsg-6.2) [Dunoon edition]
 Compiled with GNAT Version: 12.2.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2022 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

So I assume version *3.0.0* fixed the issue?
Super!

tgingold, 2023-03-03:
Yes, I confirm that master handles this correctly.

closed 2023-03-03 by tgingold

umarcor, 2023-03-03:
@davidel for completeness, version 3.0.0 does not exist yet. Tristan will typically create a release tag once a year (around feb-march). During the whole year, the version in "master" is the `*-dev` of the next version. So, during all of 2022, the "stable" release was 2.0.0, but the "master" was `3.0.0-dev`.
In the following days/weeks (I did not talk to him about it this year) Tristan will tag `3.0.0` and "master" will change to `4.0.0-dev`, in preparation for next year.

Overall, GHDL is rolling, and there is no particular reason to use an "stable" tag, instead of any other. Hence, our recomendation is to pick the most recent build available on your system and; should you build it yourself, use "master".

Packagers who are not aware of this (or who have other distro imposed constraints) are always up to one year behind the development of GHDL. We should expect Debian and others to receive a meaningful update when `3.0.0` is tagged, since 1.4k commits were contributed in the last 12 months.

agamez, 2023-04-13:
Right... So is this the same case than [this thread](https://github.com/ghdl/ghdl/issues/2254) and can't be solved if following the VHDL standard?

If that's the case... that's disappointing.