ghdl/ghdl #2371: Tagging, semver and the vertag function in ci-run.sh

umarcor, 2023-03-04:
Earlier today, @tgingold tagged `3.0.0-rc1`. I saw two issues with the syntax:

- All other version tags in this repo start with `v`.
- According to https://semver.org/, a dot is required after `rc`.

Hence, I created `v3.0.0-rc.1` and removed `3.0.0-rc1` (both pointing to the same commit).

To my surprise, CI fails: https://github.com/ghdl/ghdl/actions/runs/4330739145. It's ok, since that's what `rc`s are for, and we don't create releases/pre-releases for `-rc*` tags anyway. Nonetheless, I run some tests to understand what's going on.

See the following function in `scripts/ci-run.sh`:

https://github.com/ghdl/ghdl/blob/0d2363dc400eebbbdb06cd93a2ae87e4bfaaf1e8/scripts/ci-run.sh#L93-L107

If the tag starts with `v`, it requires the string to match the ghdl_version in `configure` (https://github.com/ghdl/ghdl/blob/master/configure#L24). We can find that Tristan explicitly changed it before tagging release candidates two years ago:

- https://github.com/ghdl/ghdl/blob/v1.0.0rc3/configure
- https://github.com/ghdl/ghdl/blob/v1.0.0rc4/configure

I see two solutions:

- I create a commit setting the versions in https://github.com/ghdl/ghdl/blob/master/configure#L24 and https://github.com/ghdl/ghdl/blob/master/pyGHDL/__init__.py#L54 to `3.0.0-rc.1`, then change the tag to point to that commit and last do another commit to set them back to `3.0.0-dev`.
- Change the `vertag` function to ignore the content after the hyphen in the comparison so that `x{v3.0.0-[rc.1]} == xv{3.0.0-[dev]}` would return true.

The first approach will ensure that users will see the rc tag when they run `ghdl version`; while the second approach will always show `-dev`. However, the first approach requires creating two commits in order to publish an rc tag, while the second approach allows what I think Tristan meant this morning: just tag and push.

/cc @tgingold @Paebbels @tmeissner @Xiretza

umarcor, 2023-03-04:
Note that there is a regexp in semver.org that we can use to break the two strings (the tag and the one read from configure) into pieces: https://semver.org/#is-there-a-suggested-regular-expression-regex-to-check-a-semver-string.

tgingold, 2023-03-05:
Thank you for the issue, I wanted to test the release process as previously no assets were generated.

I will try the first approach soon.

umarcor, 2023-03-05:
@tgingold note that no release is generated and no assets are uploaded for tags with an appendix.

- Releases are only generated for non rc tags.
- Assets are only uploaded to the nightly pre-release.

We set that behaviour in January 2021. See https://ghdl.github.io/ghdl/#ghdl-v1-0-0rc1-was-tagged.
That's why in https://github.com/ghdl/ghdl/releases:

- Releases `v1.0.0` and `v2.0.0` don't have assets.
- Tags `v1.0.0-*` do not have a (pre)-release.

Hence, all these `v3.0.0-*` tags are expected to do nothing (no prerelease, no assets). When `v3.0.0` is tagged, a release will be created with no assets.

Do you want to change that behaviour? Do you want prereleases and/or releases to also have assets? See https://github.com/pyTooling/Actions/tree/main/releaser#snapshots and https://github.com/ghdl/ghdl/blob/master/.github/workflows/Test.yml#L277-L278.

tgingold, 2023-03-05:
Yes, I'd like to have a different behaviour: pushing a tag should create a release, whatever the tag is.

I am trying to do it directly.

umarcor, 2023-03-05:
Do you want the release to:

- Be a prerelease or a release depending on the tag? Or always be a release?
- Include assets or be just an empty release?

tgingold, 2023-03-05:
Prerelease is ok, as I think we can move from prerelease to release from the web interface.
The next step would be to do a release if the tag is a release tag.

It must always includes the assets.  I don't see the point of doing a (pre)release without all the assets.

umarcor, 2023-03-05:
Then, just adding `snapshots: true` in https://github.com/ghdl/ghdl/blob/master/.github/workflows/Test.yml#L291 should make it. That will create pre-releases without assets.

tgingold, 2023-03-05:
Ok, but I also want the assets!

umarcor, 2023-03-05:
Sorry, it's inverse logic. Needs to be `snapshots: false`.

umarcor, 2023-03-05:
> Ok, but I also want the assets!

Then, also remove these two lines: https://github.com/ghdl/ghdl/blob/master/.github/workflows/Test.yml#L277-L278

umarcor, 2023-03-05:
> It must always includes the assets.  I don't see the point of doing a (pre)release without all the assets.

We discussed that pre-release tags were meant for packagers so that they could more easily know which commit to checkout from the repo. However, we did not want assets because otherwise users tend to try those instead of the nightlies and we get reports for issues that are solved already. So, the recommendation was: use the version your package manager provides or use the master/nightly.

tgingold, 2023-03-05:
Yes, I understand this point of view.  But sometimes you want a more recent version than the packaged one.

umarcor, 2023-03-05:
IMHO believe the solution for that would be to release more frequently (e.g. `3.1.0` in June, `3.2.0` in September, `3.3.0` in December), rather than adding assets to pre-releases. Anyway, I'm testing a modification for the behaviour you want: https://github.com/umarcor/ghdl/commit/8fddb7e327962483631d4419d9fd5ad8aa483bcb. I will push to master after CI passes in my fork: https://github.com/umarcor/ghdl/actions/runs/4337772692,

umarcor, 2023-03-05:
@tgingold, done: https://github.com/ghdl/ghdl/commit/6697d3502aba1923d900f6a9c781b392f1f4c9c6. From now on, any tag will create a (pre)-release and assets will be uploaded. If the version has an appendix, it will be a pre-release; otherwise, it'll be a release.

closed 2023-03-05 by umarcor

tgingold, 2023-03-06:
Thank you for the help.