ghdl/ghdl #2372: to_01 is not supported

rhinton, 2023-03-04:
**Description**
Synthesis fails on my call to `to_01 [std_logic] return std_logic`.  I get the following error.

``` 
../hdl/to01.vhd:8:11: unhandled dyn operation: IIR_PREDEFINED_IEEE_1164_TO_01_LOG_LOG
  dout <= to_01(din);
```

**Expected behaviour**
For synthesis, `to_01` should be a "no-op".

**How to reproduce?**
```vhd :file: to01.vhd
library IEEE; use IEEE.std_logic_1164.all;
entity to01 is 
  port (din : in std_logic; dout : out std_logic);
end;
architecture test of to01 is begin
  dout <= to_01(din);
end;
```

and execute

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl --synth --std=08 --out=vhdl --no-formal to01.vhd -e to01
```

**Context**

- OS: Ubunto 22.04
- Origin:
  - [X] Released binaries: `https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-03-04/oss-cad-suite-linux-x64-20230304.tgz`

Output from version command:

```sh
$ ghdl --version
GHDL 3.0.0-dev (2.0.0.r1400.gc22d8419d.dirty) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 llvm 10.0.0 code generator
Written by Tristan Gingold.
```

closed 2023-03-05 by tgingold

rhinton, 2023-03-07:
Thanks for the incredibly fast turnaround!

rhinton, 2023-03-07:
Sorry to trouble you.  Any guess when this will show up in the OSS CAD suite?  
https://github.com/YosysHQ/oss-cad-suite-build/releases

tgingold, 2023-03-08:
No idea, I wasn't aware they were distributing ghdl.

commit 184901af (2023-03-05): synth-vhdl_oper: handle to_01.  Fix #2372