ghdl/ghdl #2373: PSL "p[+]" not working as expected

Topi-ab, 2023-03-05:
**Description**
The PSL sere p[+] and p[*1 to inf] behave differently.

**Expected behaviour**
The code with p[+] and p[*1 to inf] should behave the same way. They do not. See the example.

**How to reproduce?**
https://github.com/Topi-ab/async_formal_tester/commit/85359b8553e394acb2f8133f5eeb81877dd92333

run on project root:
`docker run -v $PWD:/root --rm -it hdlc/formal:all bash -c "cd /root; ./run.sh"`

**Context**
Please, provide the following information:

- OS: Linux + Docker
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
Not a SW crash.

**Additional context**

When the lines 45 & 51 are not commented, the tool analyzes all 50 cycles and reports no assertion.

When lines 45 & 51 are commented, and lines 46 & 52 are uncommented, the tools reporst stability_2 assertion on clock step 1.

tgingold, 2023-03-06:
I confirm this issue.

tgingold, 2023-03-07:
The fact you don't use "always" makes the assertion very special.
I am not even sure what would be the outcome.  Is it really what you wanted to write ?

Topi-ab, 2023-03-07:
The non-always assert is intended.

The idea is to compare two copies of the same dut, and to verify they behave the same way upto the first clock cycle the inputs between the two duts are different. 
I.e. clock cycles 0 to 3 with identical input,s and clock cycle 4 with different inputs. Then on clock cycle 4 the outputs do not match only if the output has asynchronous path from input (violation of the requirement tested). 

And running this with all possible lengths of equal inputs.

See an update here:
https://github.com/Topi-ab/async_formal_tester/commit/325714e2c703723174d3d1bb0ed86c781fabb4d1

***

But the reason for this issue is not the intent of my code, but to note that GHDL might work unintended way. 

The tool cannot be used as a formal checker if the user cannot rely on the tool to work as specified.

tgingold, 2023-03-07:
ok, but you have a reset problem in the design (and that's why there is a failure).

At the first clock edge, the signals are not initialized.
One of the assertion is never activated (and therefore never fails).
The other is activated and therefore can be incorrect.

But I agree that both should be equivalent.

closed 2023-03-09 by tgingold