ghdl/ghdl #2379: Some way to get PSL asserted signals out from the synthesized design

Topi-ab, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
With this, one could automatically generate HW-property checkers to be implemented into FGPA design.
E.g. Design which is too complicated to be formally verified, could have runtime checkers to increment counter every time an assert happens.

**Describe the solution you'd like**
Some synthesis option or similar, which routes PSL asserted signals in synthesized netlist as output signal. 
And consistent naming of the signals (hierarchy_architecture_assert) so that the names will stay the same when design contents (not hierarchy) changes.

**Describe alternatives you've considered**
1. One can make a parser of synthesized netlist to locate asserts and have the parser to generate wanted results.

2. One can use VHDL hierarchical signal probing, if the names of the signals used in assert were known (e.g. having the signal name always to be derived from assert name, or the line number if the assert name is missing). Or if one extra signal is generated with derived name from assert name/linenumber, and that signal is driven by asssert value (boolean).

**Additional context**
nope.

tgingold, 2023-03-07:
I think the easiest way is to use yosys and redirect pins.

PSL assertions create a special gate.  A yosys script could be written to transform those gates into pins.

Topi-ab, 2023-03-07:
I am not familiar with yosys. But if it can be  used to do the trick, it makes it possible to do CICD ready runtime (as in FPGA runtime) checkers utilizing PSL notation.

Will start studying yosys...

Topi-ab, 2023-03-11:
This
https://github.com/YosysHQ/yosys/discussions/3701#discussioncomment-5275990
does the required trick.

closed 2023-03-11 by Topi-ab