ghdl/ghdl #2380: Documentation how PSL model checkers are generated

Topi-ab, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
Yes. GHDL might generate erratic model checkers.
See eg this: https://github.com/ghdl/ghdl/issues/2373

Asking if the model checker generation algorithm would be properly documented. Then the GHDL synthesized checker could be formally verified (alson as regression testing) against known solutions. 

**Describe the solution you'd like**
1. [This](https://github.com/ghdl/ghdl/issues/2379) as an enabler for formal equivalence checking. Without signal to compare, there is not much to do.
2. Clear and understandable documentation how the conversion of PSL to netlist is done.

**Describe alternatives you've considered**
None

**Additional context**
None

tgingold, 2023-03-07:
I fear this is a research project.

There is no one unique way to generate a checker.

Topi-ab, 2023-03-07:
I didn't ask how to generate checkers as a general interest.

But I am interested how GHDL specifically generates checkers. 
Maybe it is taken from a research paper (in that case, can we have the link to it). 
Maybe it is "self invented". In such case there might be interests to peer-review the algorithm (to assist GHDL to be reliable part of formal checking toolchain).

tgingold, 2023-03-07:
No, it is not self-invented.  There are several papers on how to generate automata from PSL.  I am pretty sure I use many of them.

closed 2023-03-15 by tgingold