ghdl/ghdl #2381: Compilation error when using 'with select' inside a 'case' statement

sigmuha, 2023-03-07:
**Description**
The compiler gives out an error when using a 'with select' statement inside of a 'case' statement.  

**Expected behaviour**
When compiling with vsim, no error or warnings are outputted. The expected output from the ghdl compiler should be the same.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity;

architecture rtl of test is
    signal a,b,c,d,e: std_logic;
begin

    COMBINATORIC: process( all ) is
    begin
        case a is
            when '0' =>
                with b select c <=
                    '0' when '1',
                    '1' when '0',
                    '0' when others;
            when others =>
                null;
        end case;
    end process;

end architecture rtl;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -c std=08 test.vhd
```

**Context**

- OS:
- Red Hat Enterprise Linux 8.7 (Ootpa)
- Origin:
  - [ ] Package manager: `version`
  - [x ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
test.vhd:20:17:error: 'end' is expected instead of 'with'
                with b select c <=
                ^
test.vhd:20:17:error: 'case' is expected instead of 'with'
                with b select c <=
                ^
test.vhd:19:24:error: missing ";" at end of statement
            when '0' =>
                       ^
test.vhd:24:13:error: 'end' is expected instead of 'when'
            when others =>
            ^
test.vhd:24:13:error: "end" must be followed by 'process'
            when others =>
            ^
test.vhd:23:37:error: ';' expected at end of process
                    '0' when others;
                                    ^
test.vhd:23:37:error: (found: 'when')
test.vhd:24:13:error: 'end' is expected instead of 'when'
            when others =>
            ^
test.vhd:23:37:error: missing ";" at end of architecture
                    '0' when others;
                                    ^
test.vhd:26:9:error: missing entity, architecture, package or configuration
        end case;
        ^
test.vhd:27:5:error: missing entity, architecture, package or configuration
    end process;
    ^
test.vhd:29:1:error: missing entity, architecture, package or configuration
end architecture rtl;
^
******************************************************************
```

** Additional context**
GHDL version used: GHDL 3.0.0-dev (2.0.0.r823.g5726f0ecc) [Dunoon edition]

sigmuha, 2023-03-11:
After some more testing, I isolated the issue further to having a "with-select" statement inside of a process, not just nested inside of a case. 
It seems to be an issue when having a 'with -select' as anything other than a concurrent statement.

tgingold, 2023-03-11:
It is now supported.

umarcor, 2023-03-11:
@tgingold, I saw https://github.com/ghdl/ghdl/commit/38bfed041aa054b11716e832e750ea0ac788c2ca, https://github.com/ghdl/ghdl/commit/e82c258ec20e9261cf9b76cbaa2e60ab33d61d9a and https://github.com/ghdl/ghdl/commit/82e970042ce5133df5de05410d7d0d5a43ace90b. Is there anything missing to close this issue?

tgingold, 2023-03-11:
No, I thought I closed it through a commit.

closed 2023-03-11 by tgingold