ghdl/ghdl #2388: Wrong analysis order when using make

c-thaler, 2023-03-09:
**Description**
When using "make" command, the following two files lead to an error:

top_e.vhd
```vhd :file: top_e.vhd
library ieee;
use ieee.std_logic_1164.all;

entity unit_a is
    port (
        clk   : in  std_logic;
        reset : in  std_logic;

        a     : in  std_logic;
        b     : out std_logic
    );
end entity;

library ieee;
use ieee.std_logic_1164.all;

entity unit_b is
    port (
        clk   : in std_logic;
        reset : in std_logic;

        a     : in  std_logic;
        b     : out std_logic
    );
end entity;

architecture rtl of unit_a is
begin
    my_inst: entity work.unit_b
      port map (
        clk => clk,
        reset => reset,
        a => a,
        b => b
      );
end architecture;
```

top_a.vhd
```vhd :file: top_a.vhd
architecture rtl of unit_b is
begin

    b <= not a;

end architecture;
```

It seems that GHDL is unable to determine the correct analysis order when using the "make" command. The problem might be that there is a kind of cyclic dependency between two files.
GHDL exits with:

```
ghdl -i --workdir=build top_a.vhd top_e.vhd
ghdl -m --workdir=build unit_a
analyze top_a.vhd
top_a.vhd:1:14:error: unit entity "unit_b" has not been analyzed
architecture rtl of unit_b is
             ^
ghdl:error: compilation error
```

If I remove the instantiation of ```work.unit_b``` in top_e.vhd, make works as expected.

**Expected behaviour**
The "top_e.vhd" should be analyzed before "top_a.vhd" since it contains the entity declarations.

**How to reproduce?**
See sample code above or run "make" in the attached MWE:
[mwe.zip](https://github.com/ghdl/ghdl/files/10932988/mwe.zip)

**Context**
Please, provide the following information:

- OS: Linux
- Origin:
  - [x] Built from sources: `578c1b7fcae79be63224d8d595de9a2fdc7cde2c`

c-thaler, 2023-03-09:
BTW, analyzing both files manually in the correct order works:

```
ghdl -a --workdir=build top_e.vhd top_a.vhd
```

tgingold, 2023-03-09:
The way you distribute design units to files is unusual!

c-thaler, 2023-03-09:
I know, sorry :) It's actually not me but a code generation tool that generates a wrapper (basically what's in top_e.vhd) around the original module to do some protocol checking and automatically adding additional registers for decoupling. This has never been a problem so far, since I never used the "make" command. But I started using cocotb which unfortunately uses "make".
I might be able to change the code generation tool instead.

commit 5d90b04e (2023-03-13): ghdllocal.adb(Build_Dependence): rebuild file dependencies.

vhdl: remove Get/Set_File_Dependence_List.
Fix #2388
closed 2023-03-13 by tgingold