ghdl/ghdl #2390: raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:770

smaslovski, 2023-03-13:
**Description**

Hi, here is another synthesis bug. I tried reducing the code, however I am not sure which lines trigger the bug:

```
% ghdl synth --std=08 bug.vhd -e top
bug.vhd:78:21:note: found ROM "n48", width: 12 bits, depth: 256
      IR <= rom_data(11 downto 8);
                    ^
bug.vhd:125:10:note: found RAM "mem", width: 8 bits, depth: 256
  signal mem : memory := (others => x"00");
         ^

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r12.gc2a8dee97) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-linux-gnu
/home/stas/vhdl/ucpu/bug/
Command line:
/home/stas/app/ghdl/bin/ghdl synth --std=08 bug.vhd -e top
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : netlists.adb:770
Call stack traceback locations:
0x7f7bece8ea91 0x55761f2cccc7 0x55761f3939ec 0x55761f397abb 0x55761f3993d1 0x55761f439b4b 0x55761f47ee1e 0x55761f47eeb4 0x55761f439a66 0x55761f52c5f2 0x55761f500904 0x55761f5e1077 0x55761f195f11 0x7f7beca46188 0x7f7beca46243 0x55761f1943df 0xfffffffffffffffe
******************************************************************
```

Here is the code (combined to one file bug.vhd, for simplicity)

```
-- types.vhd -----------------------------------------------------------
library ieee;
context ieee.ieee_std_context;

package uCPUtypes is
  alias logic is std_ulogic;
  alias logic_vec is std_ulogic_vector;
  subtype unsigned_byte is unsigned(7 downto 0);
  subtype code_word is unsigned(11 downto 0);
end package uCPUtypes;library ieee;

-- top.vhd -------------------------------------------------------------
context ieee.ieee_std_context;
use work.uCPUtypes.all;

entity top is
  port (
    rst, clk : in logic;
    display : out unsigned_byte;
    stb : out logic
  );
end entity top;

architecture mixed of top is
  signal rom_addr, ram_addr, ram_data : unsigned_byte;
  signal rom_data : code_word;
  signal wr_en : logic;
begin

stb <= '1' when wr_en = '1' and ram_addr = x"00" else '0';

display <= ram_data;

CPU_instance: entity work.uCPU(RTL) port map (rst, clk, rom_addr, rom_data, ram_addr, ram_data, wr_en);

ROM_instance: entity work.ROM(RTL) port map (rom_addr, rom_data, '1');

RAM_instance: entity work.RAM(RTL) port map (clk, ram_addr, ram_data, wr_en);

end architecture mixed;

-- ucpu.vhd ------------------------------------------------------------
library ieee;
context ieee.ieee_std_context;
use work.uCPUtypes.all;

entity uCPU is
  port (
    rst, clk : in    logic;
    rom_addr : out   unsigned_byte;
    rom_data : in    code_word;
    ram_addr : out   unsigned_byte;
    ram_data : inout unsigned_byte;
    wr_en    : out   logic
  );
end entity uCPU;

architecture RTL of uCPU is
  signal IR : unsigned (3 downto 0);
  signal A, PC : unsigned_byte := x"00";
begin

rom_addr <= PC;

ram_addr <= A;

ram_data <= x"ZZ";

wr_en <= '0';

process (clk) is
begin
  if rising_edge(clk) then
    if rst then
      PC <= x"00";
      A  <= x"00";
    else
      IR <= rom_data(11 downto 8);
      A <= rom_data(7 downto 0);
      PC <= PC + 1;
    end if;
  end if;
end process;

end architecture RTL;

-- rom.vhd -------------------------------------------------------------
library ieee;
context ieee.ieee_std_context;
use work.uCPUtypes.all;

entity ROM is
  port (
    abus : in  unsigned_byte;
    dbus : out code_word;
    en   : in  logic
  );
end entity ROM;

architecture RTL of ROM is
  type memory is array (0 to 255) of code_word;
  constant mem : memory := (x"777", others => x"000");
begin

dbus <= mem(to_integer(abus)) when en else x"ZZZ";

end architecture RTL;

-- ram.vhd -------------------------------------------------------------
library ieee;
context ieee.ieee_std_context;
use work.uCPUtypes.all;

entity RAM is
  port (
    clk   : in logic;
    abus  : in unsigned_byte;
    dbus  : inout unsigned_byte;
    wr_en : in logic
  );
end entity RAM;

architecture RTL of RAM is
  type memory is array (0 to 255) of unsigned_byte;
  signal mem : memory := (others => x"00");
begin

dbus <= mem(to_integer(abus)) when not wr_en else x"ZZ";

syn_write: process (clk)
begin
  if rising_edge(clk) then
    mem(to_integer(abus)) <= dbus when wr_en else unaffected;
  end if;
end process syn_write;

end architecture RTL;

```

With best regards,
Stanislav

tmeissner, 2023-03-13:
Considering the latest bugfixes for `unaffected`, have you tried to remove that from the `syn_write` process? Especially because it has no real use there, as
```
mem(to_integer(abus)) <= dbus when wr_en else unaffected;
```
is similar to
```
mem(to_integer(abus)) <= dbus when wr_en;
```

tgingold, 2023-03-14:
The current workaround is to write:
```vhdl
syn_write: process (clk)
begin
  if rising_edge(clk) then
    if wr_en then
      mem(to_integer(abus)) <= dbus; -- when wr_en; -- else unaffected;
    end if;
  end if;
end process syn_write;
```

commit e4740a99 (2023-03-14): synth_conditiona_signal_assignment: handle simple case directly.

Fix #2390
closed 2023-03-14 by tgingold