ghdl/ghdl #2392: p[+] not working according to PSL specification

Topi-ab, 2023-03-14:
**Description**
PSL "p[+]" doesn't catch an assert it is supposed to catch.

This is a continuation to: https://github.com/ghdl/ghdl/issues/2373

**Expected behaviour**
catch the assert at cycle 20.

**How to reproduce?**
https://github.com/Topi-ab/async_formal_tester/commit/736c7a32814e07c0d135096962333f11676df1b8

The asserts "b2_nonasync_1" and "b2_nonasync_2" should be asserted at clock cycle 20. They are not.

The commented asserts (tb_dut.vhdl) at lines 87, 90 and 93 do catch asserts at clock cycle 20.

**Context**
Please, provide the following information:

- OS:
- Origin: docker hub "hdlc/formal:all" (as of March 14, 2023).
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
Not a SW crash.

**Additional context**

tmeissner, 2023-03-14:
Both assertions are asserted when I add `always` to the properties. So I assume that it have something to do with behavior at the start of the formal run.

Topi-ab, 2023-03-14:
I managed to create a formal verification to check the synthesized PSL.

https://github.com/Topi-ab/formal_verificaition_of_ghdl_psl

Comparing PSL assertion 
`p_plus_psl: assert {a_in; b_in} |-> {c_in};`
to manual version.

running `cd src/psl/psl_p_plus; ./run.docker.sh`

shows mismatch at cycle 2. According to "spec", c_in should be '1' at last cycle, but is not.
nda_res drops low (saying manual checker found violation), but p_plus_psl stays high (not detecting violation using PSL sequence).

This assuming the hand-crafted reference model is correctly created.

[image]

Topi-ab, 2023-03-14:
Actually the previous PSL was faulty ;)

Should be
`p_plus_psl: assert {a_in[+]; b_in} |-> {c_in};`

Now fixed in https://github.com/Topi-ab/formal_verificaition_of_ghdl_psl/commit/d72c69ee34d6847838b0810067c0aae487c3ff9d

And still fails, with the same waveform as in previous comment.

Topi-ab, 2023-03-14:
Tried to cover PSL generated property for "being able to detect error of original PSL (p[+]) at clock cycle 2"
by:
`cover_psl: cover {true; true; p_plus_psl = '0'};`

cover failed.

https://github.com/Topi-ab/formal_verificaition_of_ghdl_psl/commit/4258f0f0b54fbee9c134ead00634f96e45b09663

tgingold, 2023-03-14:
I am not sure to understand what is the issue.

If I run sby for b2_nonasync_1, I get the following waveform:

So yes, it is possible to make this assertion failed.

For me, both b2_nonasync_1 and b2_nonasync_2 fail to be proved.

What did I miss ?

Topi-ab, 2023-03-14:
@tgingold you are missing (or my browser is missing) the image.

tgingold, 2023-03-14:
The picture:
[image]

Topi-ab, 2023-03-14:
Could it be that hdlc/formal:all image is not uptodate with the latest correction?

Did you run in the Docker?

I added run.docker.sh script to replicate setup.

My run ends with (didn't fail the two non-commented asserts):

[image]

tgingold, 2023-03-14:
No, I don't use docket; I directly use the ghdl plugin.
So maybe the docker image is not updated.  What is the version ?

Topi-ab, 2023-03-14:
root@e3aed969e87a:/# ghdl --version
GHDL 4.0.0-dev (v3.0.0-4-g578c1b7fc) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tgingold, 2023-03-14:
It contains the changes for your previous issue.  But what about the yosys plugin ?

Topi-ab, 2023-03-14:
How do I see the plugin version?

Topi-ab, 2023-03-14:
yosys> ghdl -v

2. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (v3.0.0-4-g578c1b7fc) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:862
******************************************************************
ERROR: vhdl import failed.

tgingold, 2023-03-14:
So it's the same version.

Weird.

Can you file the output of: ghdl synth --std=08 --formal dut.vhdl tb_dut.vhdl -e

Topi-ab, 2023-03-14:
```
root@62b4474927a0:~# ghdl synth --std=08 --formal dut.vhdl tb_dut.vhdl -e > res.vhdl
ghdl:note: top entity is "tb_dut"
root@62b4474927a0:~# exit
exit
root@docker-base:~/work/formal/async_formal_tester# cat res.vhdl
library ieee;
use ieee.std_logic_1164.all;
entity tb_dut is
  port (
    clk_in: in std_logic;
    dut1_a1_in: in std_logic;
    dut1_b1_out: out std_logic;
    dut1_a2_in: in std_logic;
    dut1_b2_out: out std_logic;
    dut2_a1_in: in std_logic;
    dut2_b1_out: out std_logic;
    dut2_a2_in: in std_logic;
    dut2_b2_out: out std_logic
  );
end;
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity dut is
  port (
    clk_in : in std_logic;
    a1_in : in std_logic;
    a2_in : in std_logic;
    b1_out : out std_logic;
    b2_out : out std_logic);
end entity dut;

architecture rtl of dut is
  signal cnt : std_logic_vector (4 downto 0);
  signal a2_prev : std_logic;
  signal n98_o : std_logic_vector (31 downto 0);
  signal n100_o : std_logic;
  signal n101_o : std_logic_vector (31 downto 0);
  signal n103_o : std_logic_vector (31 downto 0);
  signal n104_o : std_logic_vector (4 downto 0);
  signal n111_o : std_logic;
  signal n112_o : std_logic_vector (31 downto 0);
  signal n114_o : std_logic;
  signal n115_o : std_logic;
  signal n121_o : std_logic_vector (31 downto 0);
  signal n123_o : std_logic;
  signal n124_o : std_logic;
  signal n125_o : std_logic;
  signal n127_o : std_logic_vector (4 downto 0);
  signal n128_q : std_logic_vector (4 downto 0) := "00000";
  signal n129_q : std_logic := '0';
  signal n130_q : std_logic := '0';
begin
  b1_out <= n130_q;
  b2_out <= n125_o;
  -- dut.vhdl:17:16
  cnt <= n128_q; -- (isignal)
  -- dut.vhdl:19:16
  a2_prev <= n129_q; -- (isignal)
  -- dut.vhdl:24:32
  n98_o <= "000000000000000000000000000" & cnt;  --  uext
  -- dut.vhdl:24:32
  n100_o <= '1' when n98_o /= "00000000000000000000000000010100" else '0';
  -- dut.vhdl:25:44
  n101_o <= "000000000000000000000000000" & cnt;  --  uext
  -- dut.vhdl:25:44
  n103_o <= std_logic_vector (unsigned (n101_o) + unsigned'("00000000000000000000000000000001"));
  -- dut.vhdl:25:40
[... 245 lines cut ...]
  n75_o <= n74_q (0);
  -- tb_dut.vhdl:84:9
  n76_o <= n75_o and all_inputs_equal;
  -- tb_dut.vhdl:84:9
  n77_o <= n74_q (1);
  -- tb_dut.vhdl:84:9
  n78_o <= n77_o and a2_differ;
  -- tb_dut.vhdl:26:16
  n79_o <= not b2_equal;
  -- tb_dut.vhdl:26:16
  n80_o <= n79_o and a2_differ;
  -- tb_dut.vhdl:84:9
  n81_o <= n77_o and n80_o;
  n85_o <= n81_o & n78_o & n76_o & '0';
  -- tb_dut.vhdl:84:9
  n86_o <= n85_o (3);
  -- tb_dut.vhdl:84:9
  n87_o <= not n86_o;
  -- tb_dut.vhdl:84:9
  b2_nonasync_2: postponed assert n87_o = '1' severity error; --  assert
  n88_o <= n85_o (2);
  -- tb_dut.vhdl:84:9
  b2_nonasync_2_cover: assert n88_o = '1' severity note; --  assert_cover
end rtl;
```

Topi-ab, 2023-03-14:
In the synthesized result in previous comment:

```
entity dut is
  port (
    clk_in : in std_logic;
    a1_in : in std_logic;
    a2_in : in std_logic;
    b1_out : out std_logic;
    b2_out : out std_logic);
end entity dut;
```

The startup values of '0' are lost (b1_out and b2_out).

tgingold, 2023-03-14:
That's ok.  Other difference: I use the z3 solver (I don't have yices-smt2 installed).

Topi-ab, 2023-03-14:
Can you paste the generated test bench. I can try if I can make it to catch the assert.

tgingold, 2023-03-14:
Humm, good luck:

`ifndef VERILATOR
module testbench;
  reg [4095:0] vcdfile;
  reg clock;
`else
module testbench(input clock, output reg genclock);
  initial genclock = 1;
`endif
  reg genclock = 1;
  reg [31:0] cycle = 0;
  reg [0:0] PI_dut2_a2_in;
  reg [0:0] PI_dut1_a2_in;
  reg [0:0] PI_clk_in;
  reg [0:0] PI_dut2_a1_in;
  reg [0:0] PI_dut1_a1_in;
  tb_dut UUT (
    .dut2_a2_in(PI_dut2_a2_in),
    .dut1_a2_in(PI_dut1_a2_in),
    .clk_in(PI_clk_in),
    .dut2_a1_in(PI_dut2_a1_in),
    .dut1_a1_in(PI_dut1_a1_in)
  );
`ifndef VERILATOR
  initial begin
    if ($value$plusargs("vcd=%s", vcdfile)) begin
      $dumpfile(vcdfile);
      $dumpvars(0, testbench);
    end
    #5 clock = 0;
    while (genclock) begin
      #5 clock = 0;
      #5 clock = 1;
    end
  end
`endif
  initial begin
`ifndef VERILATOR
    #1;
`endif
    // UUT.$auto$ghdl.\cc:806:import_module$17  = 4'b0001;
    // UUT.$auto$ghdl.\cc:806:import_module$26  = 4'b0001;
    // UUT.$auto$ghdl.\cc:806:import_module$35  = 4'b0001;
    UUT.dut_1.a2_prev = 1'b0;
    UUT.dut_1.b1_out = 1'b0;
    UUT.dut_1.cnt = 5'b00000;
    UUT.dut_2.a2_prev = 1'b0;
    UUT.dut_2.b1_out = 1'b0;
    UUT.dut_2.cnt = 5'b00000;

    // state 0
    PI_dut2_a2_in = 1'b1;
    PI_dut1_a2_in = 1'b1;
    PI_clk_in = 1'b0;
    PI_dut2_a1_in = 1'b1;
    PI_dut1_a1_in = 1'b1;
  end
  always @(posedge clock) begin
    // state 1
    if (cycle == 0) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 2
    if (cycle == 1) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 3
    if (cycle == 2) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 4
    if (cycle == 3) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 5
    if (cycle == 4) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 6
    if (cycle == 5) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 7
    if (cycle == 6) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 8
    if (cycle == 7) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 9
    if (cycle == 8) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 10
    if (cycle == 9) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 11
    if (cycle == 10) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 12
    if (cycle == 11) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 13
    if (cycle == 12) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 14
    if (cycle == 13) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 15
    if (cycle == 14) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 16
    if (cycle == 15) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b1;
    end

    // state 17
    if (cycle == 16) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 18
    if (cycle == 17) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b1;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 19
    if (cycle == 18) begin
      PI_dut2_a2_in <= 1'b0;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b0;
      PI_dut1_a1_in <= 1'b0;
    end

    // state 20
    if (cycle == 19) begin
      PI_dut2_a2_in <= 1'b1;
      PI_dut1_a2_in <= 1'b0;
      PI_clk_in <= 1'b0;
      PI_dut2_a1_in <= 1'b1;
      PI_dut1_a1_in <= 1'b0;
    end

    genclock <= cycle < 20;
    cycle <= cycle + 1;
  end
endmodule

Topi-ab, 2023-03-14:
`yosys -m ghdl.so` didn't work. Used `-m ghdl` instead.

Also added -e to the end of ghdl command.

```
root@3ea0aa93a6ba:~# yosys -m ghdl

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.27+3 (git sha1 b58664d44, clang 11.0.1-2 -fPIC -Os)

yosys> ghdl --std=08 --formal tb_dut.vhdl dut.vhdl -e
1. Executing GHDL.
note: top entity is "tb_dut"
Importing module tb_dut.
Importing module dut.

yosys> prep -top tb_dut

2. Executing PREP pass.

2.1. Executing HIERARCHY pass (managing design hierarchy).

2.1.1. Analyzing design hierarchy..
Top module:  \tb_dut
Used module:     \dut

2.1.2. Analyzing design hierarchy..
Top module:  \tb_dut
Used module:     \dut
Removed 0 unused modules.
Module tb_dut directly or indirectly contains formal properties -> setting "keep" attribute.

2.2. Executing PROC pass (convert processes to netlists).

2.2.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

2.2.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2.2.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 0 assignments to connections.

2.2.4. Executing PROC_INIT pass (extract init attributes).

2.2.5. Executing PROC_ARST pass (detect async resets in processes).

2.2.6. Executing PROC_ROM pass (convert switches to ROMs).
Converted 0 switches.

[... 358 lines cut ...]
  assign _19_ = _17_[1] & a2_differ;
  assign _20_ = _17_[1] & _14_;
  assign _21_ = ~ _20_;
  always @* if (1'h1) assert(_09_);
  always @* if (1'h1) cover(_05_);
  always @* if (1'h1) assert(_16_);
  always @* if (1'h1) cover(_12_);
  always @* if (1'h1) assert(_21_);
  always @* if (1'h1) cover(_19_);
  dut dut_1 (
    .a1_in(dut1_a1_in),
    .a2_in(dut1_a2_in),
    .b1_out(dut1_b1_out),
    .b2_out(dut1_b2_out),
    .clk_in(clk_in)
  );
  dut dut_2 (
    .a1_in(dut2_a1_in),
    .a2_in(dut2_a2_in),
    .b1_out(dut2_b1_out),
    .b2_out(dut2_b2_out),
    .clk_in(clk_in)
  );
endmodule
```

Topi-ab, 2023-03-14:
I added simulation tb to the project.

[image]

No asserts during simulation. Can you confirm on your side.

`./run.simu.sh`

Topi-ab, 2023-03-14:
How do you backtrack the ghdl version to the git commit?

Is there any change that the hdlc/formal build is bricked, and actually using older codebase?

umarcor, 2023-03-14:
Containers from hdl/containers are updated once a week. Last time was 4 days ago. I triggered the workflows now.

Topi-ab, 2023-03-14:
Does the binary fingerprint give any glue?

```
root@14821bf3e719:/# md5sum `which ghdl`
44f140b5a92a5a352e8fb5ef0bfccac7  /usr/local/bin/ghdl
```

Topi-ab, 2023-03-14:
> I managed to create a formal verification to check the synthesized PSL.
>
> https://github.com/Topi-ab/formal_verificaition_of_ghdl_psl
>
> Comparing PSL assertion `p_plus_psl: assert {a_in; b_in} |-> [...]

@tgingold can you try this out? run `run.formal.sh`.

Topi-ab, 2023-03-15:
new build of hdlc/formal didn't help.

also chaning solver to
```
[engines]
smtbmc z3
```
didn't make it to trigger the assert.

Simulation runs without any asserts printed out.

tgingold, 2023-03-15:
It doesn't fail but: I am not using docker and I use an older version of yosys.

But wait, I also have local changes.

Topi-ab, 2023-03-15:
I ran the simulator based check with Riviera Pro in EdaPlayground.

https://www.edaplayground.com/x/dmS5

It asserted the two problematic asserts:
```
# ASSERT: Error: ASRT_0005 design.vhd(81): Assertion "b2_nonasync_1" FAILED at time: 210ns, scope: simu_tb.dut, start-time: 10ns
# ASSERT: Error: ASRT_0005 design.vhd(84): Assertion "b2_nonasync_2" FAILED at time: 210ns, scope: simu_tb.dut, start-time: 10ns
```

tgingold, 2023-03-15:
I have pushed the fix.  It should now be working correctly.

Topi-ab, 2023-03-15:
@tgingold What was the reason you didn't face the problem?

tgingold, 2023-03-15:
I was working with a version that already contained the fixes I have committed...

closed 2023-03-19 by tgingold