ghdl/ghdl #2396: Instantiation of Generic Package inside of Package Raises Error During Analysis

bpadalino, 2023-03-20:
**Description**
Instantiating a generic package inside of a package raises an error during analysis.

**Expected behaviour**
Don't raise the error.

**How to reproduce?**
```vhd :file: frequency.vhdl
library ieee ;
    use ieee.std_logic_1164.std_logic ;
    use ieee.std_logic_1164.std_ulogic ;
    use ieee.std_logic_1164."not";

package frequency is

    type frequency is range 0 to 2e9 units
        Hz ;
        kHz = 1000 Hz ;
        MHz = 1000 kHz ;
        GHz = 1000 MHz ;
        THz = 1000 GHz ;
    end units ;

    function half_period(freq : frequency) return time ;
    function period(freq : frequency) return time ;

    -- GHDL workaround to make a generic package and instantiate it?
    package gen_generate_clock is
      generic (
        type t ;
        function "not"(x : t) return t is <>
      ) ;
        procedure generate_clock(signal clock : inout t ; freq : frequency ; count : natural := 0) ;
    end package ;

    package generate_clock_bit is new gen_generate_clock generic map(t => bit) ;
    package generate_clock_std_logic is new gen_generate_clock generic map(t => std_logic) ;
    package generate_clock_std_ulogic is new gen_generate_clock generic map(t => std_ulogic) ;

    alias generate_clock is generate_clock_bit.generate_clock[bit, frequency, natural] ;
    alias generate_clock is generate_clock_std_logic.generate_clock[std_logic, frequency, natural] ;
    -- Dedeclaration?
    --alias generate_clock is generate_clock_std_ulogic.generate_clock[std_ulogic, frequency, natural] ;

    -- Doesn't work in GHDL
    --procedure generate_clock generic (
    --    type t ;
    --    function "not"(x : t) return t is <>
    --) parameter (signal clock : inout t ; freq : frequency ; count : natural := 0) ;

    --procedure generate_clock is new generate_clock generic map(t => std_ulogic) ;
    --procedure generate_clock is new generate_clock generic map(t => std_logic) ;
    --procedure generate_clock is new generate_clock generic map(t => bit) ;

end package ;

package body frequency is

    function period(freq : frequency) return time is
    begin
        return 1 sec / frequency'pos(freq) ;
    end function ;

    function half_period(freq : frequency) return time is
    begin
        return period(freq) / 2.0 ;
    end function ;

    package body gen_generate_clock is
        procedure generate_clock(
            signal clock : inout t ;
            freq : frequency ;
            count : natural := 0
        ) is
            constant hp : time := half_period(freq) ;
            variable downcount : natural := count ;
        begin
            -- count = 0 means forever, otherwise we look at the downcount
            while count = 0 or downcount > 0 loop
                clock <= not clock after hp ;
                downcount := downcount - 1 ;
            end loop ;
        end procedure ;
    end package body ;

    --procedure generate_clock generic(
    --  type t ;
    --  function "not"(x : t) return t is <>
    --) parameter (
    --    signal clock : inout t ;
    --    freq : frequency ;
    --    count : natural := 0
    --) is
    --    constant hp : time := half_period(freq) ;
    --    variable downcount : natural := count ;
    --begin
    --    -- count = 0 means forever, otherwise we look at the downcount
    --    while count = 0 or downcount > 0 loop
    --        clock <= not clock after hp ;
    --        downcount := downcount - 1 ;
    --    end loop ;
    --end procedure ;

end package body ;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 frequency.vhdl
```

**Context**
```
$ ghdl --version
GHDL 4.0.0-dev (3.0.0.r1.gc68528225) [Dunoon edition]
 Compiled with GNAT Version: 10.4.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r1.gc68528225) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/home/bpadalino/work/vhdl/vhdl-ideas.git/sim/
Command line:
ghdl -a --std=08 frequency.vhdl
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:3776
Call stack traceback locations:
0x7f8cf2882542 0x562b558cc627 0x562b559791ac 0x562b55a48ab3 0x562b55a48d77 0x562b55a48e15 0x562b55a49d07 0x562b55a3592c 0x562b55ab0adb 0x562b55a9c81a 0x562b55b7862c 0x562b55730ed5 0x7f8cf2402d8e 0x7f8cf2402e3e 0x562b5572f353 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2023-03-22:
Should now be fixed.  Thank you for the issue!

umarcor, 2023-03-22:
Ref: #1246.