ghdl/ghdl #2402: Exception CONSTRAINT_ERROR raised when using uvvm bitvis_vip_gpio

fransschreuder, 2023-03-30:
**Description**
Whenever I use bitvis_vip_gpio in a testbench with VUnit and GHDL, there is a CONSTRAINT_ERROR raised

**Expected behaviour**
Simulation should succeed without error.

**How to reproduce?**
```
git clone https://github.com/fransschreuder/ghdl_bug_report.git
cd ghdl_bug_report
git submodule update --init
./run.py
```
Error output:
```
Re-compile not needed

Starting lib.uvvm_gpio_vunit_tb.all
Output file: /home/franss/ghdl_bug_report/vunit_out/test_output/lib.uvvm_gpio_vunit_tb.all_d3d0601184347b27d061bcef2f83dadd389c6073/output.txt

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r48.ge2d74e1f3) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/home/franss/ghdl_bug_report/
Command line:
/usr/local/bin/ghdl --elab-run --std=08 --work=lib --workdir=/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/lib -P/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/vunit_lib -P/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/lib -P/home/franss/ghdl_bug_report/bitvis_vip_axistream/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_clock_generator/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_error_injection/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_i2c/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_sbi/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_scoreboard/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_wishbone/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_gpio/v08 -P/home/franss/ghdl_bug_report/uvvm_util/v08 -P/home/franss/ghdl_bug_report/uvvm_vvc_framework/v08 -frelaxed-rules uvvm_gpio_vunit_tb func -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : /home/franss/ghdl_bug_report/vunit_out/test_output/lib.uvvm_gpio_vunit_tb.all_d3d0601184347b27d061bcef2f83dadd389c6073/,tb path : /home/franss/ghdl_bug_report/,use_color : true --assert-level=error
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:2071 access check failed
Call stack traceback locations:
0x55cce0ac7ad0 0x55cce0ac9e14 0x55cce0aca122 0x55cce0b343c9 0x55cce0b27576 0x55cce0b3457c 0x55cce0b217d8 0x55cce0b21041 0x55cce0b210e1 0x55cce0b0b919 0x55cce0b1d5c0 0x55cce0b17842 0x55cce0b62585 0x55cce0a99644 0x55cce0a87124 0x55cce0b639f6 0x55cce071aed5 0x7f4237e15d8e 0x7f4237e15e3e 0x55cce0719353 0xfffffffffffffffe
******************************************************************
fail (P=0 S=0 F=1 T=1) lib.uvvm_gpio_vunit_tb.all (0.9 seconds)

==== Summary ======================================
fail lib.uvvm_gpio_vunit_tb.all (0.9 seconds)
===================================================
pass 0 of 1
fail 1 of 1
===================================================
Total time was 0.9 seconds
Elapsed time was 0.9 seconds
===================================================
Some failed!
```

**Context**

- OS: Ubuntu 22.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x] Built from sources: `4.0.0-dev (3.0.0.r48.ge2d74e1f3)`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r48.ge2d74e1f3) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/home/franss/ghdl_bug_report/
Command line:
/usr/local/bin/ghdl --elab-run --std=08 --work=lib --workdir=/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/lib -P/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/vunit_lib -P/home/franss/ghdl_bug_report/vunit_out/ghdl/libraries/lib -P/home/franss/ghdl_bug_report/bitvis_vip_axistream/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_clock_generator/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_error_injection/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_i2c/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_sbi/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_scoreboard/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_wishbone/v08 -P/home/franss/ghdl_bug_report/bitvis_vip_gpio/v08 -P/home/franss/ghdl_bug_report/uvvm_util/v08 -P/home/franss/ghdl_bug_report/uvvm_vvc_framework/v08 -frelaxed-rules uvvm_gpio_vunit_tb func -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : /home/franss/ghdl_bug_report/vunit_out/test_output/lib.uvvm_gpio_vunit_tb.all_d3d0601184347b27d061bcef2f83dadd389c6073/,tb path : /home/franss/ghdl_bug_report/,use_color : true --assert-level=error
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : trans-chap3.adb:2071 access check failed
Call stack traceback locations:
0x55cce0ac7ad0 0x55cce0ac9e14 0x55cce0aca122 0x55cce0b343c9 0x55cce0b27576 0x55cce0b3457c 0x55cce0b217d8 0x55cce0b21041 0x55cce0b210e1 0x55cce0b0b919 0x55cce0b1d5c0 0x55cce0b17842 0x55cce0b62585 0x55cce0a99644 0x55cce0a87124 0x55cce0b639f6 0x55cce071aed5 0x7f4237e15d8e 0x7f4237e15e3e 0x55cce0719353 0xfffffffffffffffe
******************************************************************
```

**Additional context**

fransschreuder, 2023-03-30:
I created a branch: https://github.com/fransschreuder/ghdl_bug_report/tree/only_bitvis_vip_gpio_library in which I simplified the VHDL file. The only thing that seems to be necessary to trigger this bug is:

```
library vunit_lib;
  context vunit_lib.vunit_context;
```

tgingold, 2023-04-08:
I can reproduce a similar error, I am investigating.

closed 2023-04-08 by tgingold

fransschreuder, 2023-04-08:
Perfect! I can confirm that it works now!

tgingold, 2023-04-08:
Thank you for the issue and the status update.

commit 8336360c (2023-04-08): trans-chap2: defer body elaboration if non-immediate body

Fix #2402