ghdl/ghdl #2404: crash when using certain constant arrays

MJoergen, 2023-04-03:
**Description**
Ghdl crashes with a "GHDL Bug occurred".

**Expected behaviour**
A user-friendly error message, in case the source code is invalid.

Note. ModelSim provides the following message:
`** Warning: tb_min_example.vhd(26): (vcom-1362) Array expressions of array aggregate have an unknown index range.`

**How to reproduce?**

```vhd :file: tb_min_example.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb_min_example is
end entity tb_min_example;

architecture simulation of tb_min_example is

   type byte_vector is array (natural range <>) of std_logic_vector(7 downto 0);
   constant C_TEST1 : byte_vector := (
      X"43", X"36", X"34", X"20", X"43", X"41", X"52", X"54",
      X"52", X"49", X"44", X"47", X"45", X"20");

   constant C_TEST2 : byte_vector := (
      X"43", X"36", X"34", X"20", X"43", X"41", X"52", X"54",
      X"52", X"49", X"44", X"47", X"45", X"20", X"20", X"21",
      X"00", X"00", X"00", X"40", X"01", X"00", X"00", X"13");

   constant C_TEST3 : byte_vector := (
      X"43", X"36", X"34", X"20", X"43", X"41", X"52", X"54",
      X"52", X"49", X"44", X"47", X"45", X"20", X"20", X"20",
      X"00", X"00", X"00", X"40", X"01", X"00", X"00", X"13");

   type test_vector is array (natural range <>) of byte_vector;
   constant tests : test_vector := (C_TEST1, C_TEST2, C_TEST3);

begin

end architecture simulation;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 tb_min_example.vhd
```

**Context**
Please, provide the following information:

- OS: Ubuntu 22.04.1 LTS
- Origin:
  - [X] Package manager: `1.0.0+dfsg-6`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/home/mfj/git/PRIVATE/MJoergen/C64MEGA65/CORE/vhdl/
Command line:
/usr/lib/ghdl/gcc/libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ghdl1 --ghdl--std=08 -P/usr/lib/ghdl/gcc/vhdl/ieee/v08/ -P/usr/lib/ghdl/gcc/vhdl/ -quiet -o tb_min_example.s tb_min_example.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap3.adb:60
Call stack traceback locations:
0x7fca71df3542 0x750a18 0x7566cf 0x756801 0x759f54 0x78efd3 0x78f162 0x793afe 0x7ad532 0x7837f5 0x77f99b 0x7b1f4b 0x73a768 0x63da58 0xc05c81 0x6330ff 0x642cee 0x7b2279 0x630d17 0x7fca718acd8e 0x7fca718ace3e 0x6389d3 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/ghdl/gcc/libexec/gcc/x86_64-pc-linux-gnu/10.3.1/ghdl1 terminated by unhandled exception
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap3.adb:60
Call stack traceback locations:
0x7fca71df3542 0x750a18 0x7566cf 0x756801 0x759f54 0x78efd3 0x78f162 0x793afe 0x7ad532 0x7837f5 0x77f99b 0x7b1f4b 0x73a768 0x63da58 0xc05c81 0x6330ff 0x642cee 0x7b2279 0x630d17 0x7fca718acd8e 0x7fca718ace3e 0x6389d3 0xfffffffffffffffe
/usr/bin/ghdl-gcc:error: compilation error
```

**Additional context**
This is consistently reproducible, i.e. fails every time.

diogratia, 2023-04-03:
A more recent ghdl version:
```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r12.gc2a8dee97) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```
produces warnings but no exception:

```none
ghdl -a --std=08 tb_min_example.vhdl
tb_min_example.vhdl:26:46:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
   constant tests : test_vector := (C_TEST1, C_TEST2, C_TEST3);
                                             ^
tb_min_example.vhdl:26:55:warning: expression subtype doesn't match aggregate element subtype [-Wruntime-error]
   constant tests : test_vector := (C_TEST1, C_TEST2, C_TEST3);
                                                      ^
%: 
```
The good news is the ghdl bug causing the exception has been corrected in the last year or two - likely in the 3.0.0-dev releases.

Here the warnings indicate the second and third elements of the aggregate expression providing the constant tests value have a subtype that are not identical to the first element and would rely on aggregate elements of the type of array aggregate and not an element type (9.3.3 Aggregates). The message serves to point out a potential error in the first element.

The standard contains no semantic warnings.

The Modelsim warning:
```none
** Warning: tb_min_example.vhd(26): (vcom-1362) Array expressions of array aggregate have an unknown index range.
```
can be evaluated in light of it's Verror entry:

```none
vcom Message # 1362:
In an array aggregate in which the element is of an array type (array
type is array of arrays), or in which the element is a subarray (array
type is a multidimensional array), the compiler was not able to compute
the index range of any of the expressions at a particular subelement
depth or index position.  The check that each of the array expressions
is of the same array subtype or of the same length cannot be performed.
The matching element check (array length check) will be deferred until
the simulator evaluates this aggregate.
```
The 'deferred until' bit refers to determining the subtypes  of constants C_TEST1, C_TEST2, and C_TEST3 during elaboration. See IEEE Std 1076-2008 14.4 Elaboration of a declarative part, 14.4.2.5 Object declarations. Modelsim is disclaiming the determination of the validity of constant tests during analysis (compile) and not evaluating the subtypes of constants C_TEST1, C_TEST2, and C_TEST3. The warning notifies the user that an error could potentially result depending on the the elaboration of tests. At arms length with the ability for a user to see the three array constants have a different number of elements of the same subtype 

ghdl provides a greater delineation between analysis, elaboration and simulation and attempts to delve deeper during analysis. Instead of quitting evaluation subtypes when encountering positional associations of constants C_TEST1, C_TEST2, and C_TEST3 it warns their subtypes don't match for no purpose defined in the standard.

The usefulness of either the vcom or the ghdl analysis warnings is dependent on the skill of the end user and require the user to evaluate the warnings before discounting any significance. It'd be nice if all of ghdl's semantic warnings could be individually masked preferably from a loaded file instead of relying on -Wruntime-error. Individually masking warnings might be useful examining someone else's design descriptions.

MJoergen, 2023-04-04:
Thank you for your detailed analysis. Your explanation is very clear.

Since the GHDL bug is apparently fixed in later versions (and I'm apparently using a very old version), I'll close this issue.