ghdl/ghdl #2410: bound check failure, regression

avelure, 2023-04-13:
**Description**
I updated to latest nightly fb21840 and there seem to have been some breaking changes. Previously I was running with nightly 82e9700. 

**Expected behaviour**
Working as before

**How to reproduce?**
The behaviour is different on Linux and Windows. On Windows I get a "bound check failure" on some code that looks similar to the example below, the failure is pointing to the assignment on the last line in the procedure (this example does not fail though)
On Linux the same testbench gets further, but fails on some similar looking code that has the same kind of assignment.
If I print the vector lengths before the assignment I get v_sig'length=1 and sig'length=601. 1 is the correct length.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end entity test;
architecture beh of test is
   type    t_slv_array is array (natural range <>) of std_logic_vector;
   subtype t_word_array is t_slv_array(open)(15 downto 0);

  procedure test_proc(
    variable sig : out t_word_array)
  is
    variable v_sig : t_word_array(0 to sig'length);
  begin
    v_sig := (others => x"AAAA");
    sig := v_sig(1 to sig'length);
  end procedure;
begin
  
  process
    variable v_sig : t_word_array(0 to 0);
  begin
    test_proc(v_sig);
    wait;
  end process;

end architecture beh;
```

diogratia, 2023-04-14:
Not having tried the latest ghdl-llvm nightly build ( fb21840 ) on MacOS but noting the change switches from llvm@14 to llvm@15 there is another possibility.

There is no version checking in loading the LlibLLVM.dylib. The last time encountering an issue of running a new nightly it turned out the llvm version had been changed (from 13 to 15) and there was an incompatibility causing inconsistent behavior. That's likely the case again here. Noting the default llvm Cellar brew install is llvm@16, see [Homebrew Formulae llvm](https://formulae.brew.sh/formula/llvm). 

I had asked for visibility in the ghdl --version report:

```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r12.gc2a8dee97) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```

You can check which version of llvmLIB.dylib your ghdl1-llvm uses by checking /usr/local/opt/llvm (a link to  ../Cellar/llvm/15.0.7_1 for llvm@15 installed). 

The version used in ghdl1-llvm can be checked using otool -L from the command line tools:

```none
%: otool -L ghdl1-llvm
ghdl1-llvm:
	/usr/local/opt/llvm/lib/libLLVM.dylib (compatibility version 1.0.0, current version 15.0.7)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
%: 
```

If the cellar and otool -L version don't match it'd be prudent to brew install llvm@15 which commit fb21840 shows as being used in the production of the nightly build. You may or may not have to change the link /usr/local/opt/llvm to point to the llvm 15 version. I wouldn't uninstall an llvm version until I was sure the whole think worked correctly. (And you I know if you're using an M1 or M2 based Mac with a small SSD you just watch the disk space get eaten up, it's 1,420,404,000 bytes for the llvm@15). The actual dylib  is 127472 kbytes (127 Mib). If you're not planning on building your own ghdl llvm version you can skeleton-ize all or part of homebrew while leaving the libLLVM.dylib reachable through the path given by otool -L which shows no other dependencies on homebrew:

```none
%: otool -L ghdl
ghdl:
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311
%: 
````

It'd be nice if the dynamic library were integrated with the ghdl release and had explored in the past that only about half of libLLVM.dylib is necessary and the homebrew llvm install contains all the it's individual component dynamic libraries as well. It would require a license file and mention of dylib(s) specifically against that license, potentially using GPL 3.0 for the overall release and mentioning all the things staying GPL2, although that may not be the case. There seems to be lots of misinformation floating around about compatibility with the llvm.org license.  The entire homebrew installation to provide that one library is around 2.3 GiB. (And I mention it because I also installed the arm64 homebrew and llvm and have build a native ghdl llvm version using an arm64 gcc with Ada that can be found on github). Since switching to the new APFS file system Homebrew has a lot of stuff stored in a .git flle taking up have a GB itself.

Anyway tell us what /usr/local/opt/llvm links to, it may provide another incompatibility datum. I didn't keep llvm@14 around long enough to check it out thoroughly, although I do have the arm64 version still.

tgingold, 2023-04-14:
I cannot reproduce a crash, so I am just adding this test and I will check the results.

avelure, 2023-04-17:
I don't have a mac to test on.
The Windows MSYS has LLVM 16.0.1, the Linux docker runs LLVM 10.0.0, so this might explain the differences, but not the fault itself as I can roll back the version and it works again with the same LLVM.

@tgingold As I commented, I  also could not get the MWE to fail, so it was more meant as a reference in case you were able to spot a change you did that could impact this kind of code. If not, I will try to change the MWE so that it fails.

tgingold, 2023-05-16:
A few related issues have been fixed.  CLosing this one; to be reopened if the issue is still present.

closed 2023-05-16 by tgingold

avelure, 2023-05-16:
Thanks, it works well now.