ghdl/ghdl #2411: GHDL LSP Constraint error

JJ-Gaisler, 2023-04-14:
**Description**
Hi! I am currently trying to get the GHDL LSP running on GRLIB and I have had some success. At the moment I have hit a hurdle which I haven't been able to figure out. I have created a hdl-prj.json file which is based on the files that the Makefile includes when compiling one of our designs. Running GHDL in this make environment works fine and I end up with an executable that works. However, when starting the LSP with the same file list, the LSP immediately crashes due to a constraint error (listed below).

I did however get the LSP going if I first compile everything with GHDL and then left the files sections of the hdl-prj.json empty (keeping analysis untouched). This indicates to me that this must be something specific to the LSP? As soon as I include a single file in the files section ghdl-ls crashes (I haven't tested this extensively but including one of our most used design files one by one crashes the lsp).

I am mainly here to ask where I can start to debug this issue, or perhaps if someone has an idea of why this is happening and if there is any additional information I can provide to help debug this issue.

**Expected behavior**
LSP would launch and possibly produce a warning about an incorrect setup.

**How to reproduce?**
Unfortunately when I tried this on our GPL release things worked fine, this makes things harder to debug since this only happening on one of the development branches. 

**Context**
Please, provide the following information:

- OS: Linux 5.14.21-150400.24.55-default 
- Origin:
  - [ ] Built from sources: `7de967c51f352fe2d724dbec549b71a392e5ebae`
  - [ ] Neovim `v0.10.0-dev-46+ge23c6ebed`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
[ERROR][2023-04-14 09:04:11] .../vim/lsp/rpc.lua:734	"rpc" "ghdl-ls" "stderr" "\nraised CONSTRAINT_ERROR : str_table.adb:61 index check failed\n"
```

**Additional context**
I also tried this with the latest ghdl code-base and that didn't have any effect.
```json
{
  "options": {
    "ghdl_analysis": [
      "--ieee=synopsys",
      "--work=work",
      "--mb-comments",
      "--warn-no-binding",
      "--std=93",
      "-PA_PATH/gnu",
      "-PA_PATH/gnu/grlib",
      "-PA_PATH/gnu/eth",
      "-PA_PATH/gnu/opencores",
      "-PA_PATH/gnu/gaisler",
      "-PA_PATH/gnu/work",
      "--workdir=A_PATH/gnu/work"
    ]
  },
  "files": [
    {
      "file": "A_PATH/grlib/lib/gaisler/noelv/core/iunv.vhd",
      "language": "vhdl"
    }
  ]
}

```

tgingold, 2023-04-18:
That's certainly unexpected.

Can you do:
  export GHDL_LS_TRACE=ghdl-trace
before starting the editor ?  That will create trace files.  Please post the .in file

JJ-Gaisler, 2023-04-19:
> That's certainly unexpected. Can you do: export GHDL_LS_TRACE=ghdl-trace before starting the editor ? That will create trace files. Please post the .in file

Hi and thank you for the response. I created two log files one where the files section of the hdl-prj.json was empty and when it was filled. Launching the design file in question once again works fine if the files section is empty. 

Note: GitHub didn't allow the .in extension
[ghdl-trace-no-files.txt](https://github.com/ghdl/ghdl/files/11269667/ghdl-trace-no-files.txt)
[ghdl-trace-with-files.txt](https://github.com/ghdl/ghdl/files/11269668/ghdl-trace-with-files.txt)

tgingold, 2023-04-19:
I cannot see something obviously wrong or strange.

If you want to investigate by yourself, here are some tips:
- you can now replay; you should be able to reproduce the crash doing simply:
  $ ghdl-ls -i ghdl-trace.in
- Then you can replay invoking python:
  $ python /path/to/ghdl-ls -i ghdl-trace.in
- Finally you can run under gdb:
  $ gdb --args python /path/to/ghdl-ls -i ghdl-trace.in
  You don't have python symbols, but you don't care.
  Set a breakpoint on __gnat_debug_raise_exception and when it is hit, use 'bt' to display the backtrace.

Ideally a reproducer would be helpful but with such a big source base it would be difficult without having an idea of the issue.
You can also try to add some '-v' option, it could give some hints.

JJ-Gaisler, 2023-04-20:
> 

Hi I did what you asked and this is the back trace, there were quite a few jsonrpc prints which I have removed to reduce the verbosity.  When I ran with the -v option, the only difference I could see was the printing of several `2023-04-20 08:37:22,261 [INFO] loading /home/jonathan/Work/grlib_git/lib/gaisler/noelv/core/noelvint.vhd` and the analysis options being printed.
The loading section seems to work fine, all files I except are in there.

```
Starting program: /usr/bin/python3.10 /home/jonathan/.local/bin/ghdl-ls -i ghdl-trace-no-files.txt
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Content-Length: 384

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///home/jonathan/Work/grlib_git/lib/grlib/sparc/cpu_disas.vhd","diagnostics":[{"source":"ghdl","range":{"start":{"line":10,"character":0},"end":{"line":10,"character":0}},"message":"entity \"cpu_disas\" was also defined in file \"/home/jonathan/Work/grlib_git/lib/work/debug/cpu_disas.vhd\"","severity":2}]}}Content-Length: 391

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///home/jonathan/Work/grlib_git/lib/grlib/riscv/cpu_disas.vhd","diagnostics":[{"source":"ghdl","range":{"start":{"line":14,"character":0},"end":{"line":14,"character":0}},"message":"entity \"cpu_disas\" was also defined in file \"/home/jonathan/Work/grlib_git/lib/grlib/sparc/cpu_disas.vhd\"","severity":2}]}}Content-Length: 388

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///home/jonathan/Work/grlib_git/lib/work/debug/cpu_disas.vhd","diagnostics":[{"source":"ghdl","range":{"start":{"line":8,"character":0},"end":{"line":8,"character":0}},"message":"entity \"cpu_disas\" was also defined in file \"/home/jonathan/Work/grlib_git/lib/grlib/riscv/cpu_disas.vhd\"","severity":2}]}}Content-Length: 469

{"jsonrpc":"2.0","method":"textDocument/publishDiagnostics","params":{"uri":"file:///home/jonathan/Work/grlib_git/designs/noelv-generic/tb/testbench.vhd","diagnostics":[{"source":"ghdl","range":{"start":{"line":3,"character":0},"end":{"line":3,"character":0}},"message":"entity \"testbench\" was also defined in file \"./tb/testbench.vhd\"","severity":2}]}}Content-Length: 421

{"jsonrpc":"2.0","id":1,"result":{"capabilities":{"textDocumentSync":{"openClose":true,"change":2,"save":{"includeText":true}},"hoverProvider":true,"definitionProvider":true,"implementationProvider":true,"referencesProvider":false,"documentHighlightProvider":false,"documentSymbolProvider":true,"codeActionProvider":false,"documentFormattingProvider":false,"documentRangeFormattingProvider":true,"renameProvider":false}}}
Breakpoint 1, <__gnat_debug_raise_exception> (e=0x7ffff64be620 <constraint_error>, message=...) at s-excdeb.adb:43
43	s-excdeb.adb: No such file or directory.
@(gdb) bt
#0  <__gnat_debug_raise_exception> (e=0x7ffff64be620 <constraint_error>, message=...) at s-excdeb.adb:43
#1  0x00007ffff5f5b869 in ada.exceptions.complete_occurrence (x=x@entry=0x555555c231f0) at a-except.adb:929
#2  0x00007ffff5f5b879 in ada.exceptions.complete_and_propagate_occurrence (x=x@entry=0x555555c231f0) at a-except.adb:940
#3  0x00007ffff5f5bc78 in ada.exceptions.raise_with_location_and_msg (e=0x7ffff64be620 <constraint_error>, f=(system.address) 0x7ffff6823180, l=61, c=c@entry=0, m=m@entry=(system.address) 0x7ffff61e20b0) at a-except.adb:1151
#4  0x00007ffff5f5b83c in <__gnat_raise_constraint_error_msg> (file=<optimized out>, line=<optimized out>, column=column@entry=0, msg=msg@entry=0x7ffff61e20b0 <ada.exceptions.rmsg_05>) at a-except.adb:895
#5  0x00007ffff5f5bd61 in <__gnat_rcheck_CE_Index_Check> (file=<optimized out>, line=<optimized out>) at a-except.adb:1219
#6  0x00007ffff65cbadf in str_table.element_string8 (id=0, n=1) at /home/jonathan/Work/ghdl/src/str_table.adb:61
#7  0x00007ffff65d392b in files_map.is_eq (l=0, r=337537) at /home/jonathan/Work/ghdl/src/files_map.adb:1042
#8  0x00007ffff67a50d6 in vhdl.sem_lib.load_parse_design_unit (design_unit=2071332, loc=17473088) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_lib.adb:210
#9  0x00007ffff67a5b06 in vhdl.sem_lib.load_design_unit (design_unit=2071332, loc=17473088) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_lib.adb:379
#10 0x00007ffff67a5e7a in vhdl.sem_lib.load_design_unit (design_unit=2071332, loc=2064503) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_lib.adb:476
#11 0x00007ffff67a5eb4 in vhdl.sem_lib.load_primary_unit (library=2066663, name=32048, loc=2064503) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_lib.adb:487
#12 0x00007ffff67f2bcb in vhdl.sem_names.sem_selected_name (name=2064503, keep_alias=false, soft=false) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_names.adb:2487
#13 0x00007ffff67f855a in vhdl.sem_names.sem_name (name=2064503, keep_alias=false) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_names.adb:4566
#14 0x00007ffff67f9795 in vhdl.sem_names.sem_denoting_name (name=2064503) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem_names.adb:5028
#15 0x00007ffff67ce930 in vhdl.sem.sem_use_clause_name (clause=2064501) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem.adb:3236
#16 0x00007ffff67cec31 in vhdl.sem.sem_use_clause (clauses=2064501) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem.adb:3308
#17 0x00007ffff67cef16 in vhdl.sem.sem_context_clauses (unit=2064494) at /home/jonathan/Work/ghdl/src/vhdl/vhdl-sem.adb:3404
[... cut ...]
#47 0x00007ffff7cfcdb7 in ?? () from /usr/lib64/libpython3.10.so.1.0
#48 0x00007ffff7cfd15a in _PyRun_SimpleFileObject () from /usr/lib64/libpython3.10.so.1.0
#49 0x00007ffff7cfd300 in _PyRun_AnyFileObject () from /usr/lib64/libpython3.10.so.1.0
#50 0x00007ffff7e72868 in Py_RunMain () from /usr/lib64/libpython3.10.so.1.0
#51 0x00007ffff7e72b09 in Py_BytesMain () from /usr/lib64/libpython3.10.so.1.0
#52 0x00007ffff79fa24d in __libc_start_main () from /lib64/libc.so.6
#53 0x000055555555479a in _start ()

```

tgingold, 2023-04-20:
Thank you for the full trace.

I think the issue is due to the fact that one file is in hdl-prj.json and also analyzed.  I will try to reproduce the issue and avoid the crash.

tgingold, 2023-04-27:
I have committed a path to avoid the crash.
The hdl-prj.json file now also supports libraries.

closed 2023-04-27 by tgingold

JJ-Gaisler, 2023-04-28:
> I have committed a path to avoid the crash. The hdl-prj.json file now also supports libraries.

I can confirm that the LSP no longer crashes, thanks for the patch

commit 6c56631a (2023-04-27): libraries: avoid a crash when replacing a file without checksum.

Fix #2411