ghdl/ghdl #2413: Cannot create WAVE file for specified records that are defined within  a package

bilkes, 2023-04-17:
**Description**

When writing a GHW file for a simulation source that specifies  a record defined in a package, the output cannot be used. There are 2 different incarnations of the apparently same issue  
  (1) The produced GHW file appears to be invalid referencing an undefined type (line 38 enabled, 39 disabled)
  (2) The generated executable crashes with SIGSEGV (line 39 enabled, 38 disabled)

This bug is similar to #2395 (marked fixed and closed), indeed that bug can still be reproduced by moving the "General" package into the "Abstract" package in the source-code below.

**Expected Behavior**
The generated executable does not throw an Exception AND the produced WAV file is readable.

**How to reproduce?**
The following file causes the error when running the executable with a.out --wave=out.ghw. Depending on which of the lines 46/47 us used, either phenotype 1 or 2 is produced). There is a third phenotype (the compiler crashed) which likely is a duplicate of #2395, that occurs if the package "Configured" is nested within Abstract (not included in this source)

```
library IEEE;
library std;
use IEEE.STD_LOGIC_1164.all;
package Abstract is
    type Base is record
        A : std_logic_vector;
    end record;
end package;

use work.Abstract;
package General is generic
(
    l : integer
);
subtype T is Abstract.Base (
  A(l - 1 downto 0)
);
end package;

package Configured is new work.General generic
        map (l => 8);
    
library IEEE;
library std;
use IEEE.STD_LOGIC_1164.all;
use work.Abstract;
use work.Configured;

entity CrashTest is
end entity;

architecture Behavioral of CrashTest is
    subtype Specialized is Abstract.Base (
      A(7 downto 0)
    );

     --signal Dummy : Specialized; -- mangled GHW file
    signal Dummy : Configured.T; -- produced executable crashed 
begin
    process is
    begin
        Dummy.A <= "01101100";
        wait for 20 ns;
        wait;
    end process;

end architecture;
```
```
ghdl -a --std=08 CrashTest.vhdl`
ghdl -e  CrashTest
./crashtest --wave=aha.ghw
```

**Context** 
OS: Linux

```
ghdl --version
GHDL 4.0.0-dev (3.0.0.r64.g70c21cb6b) [Dunoon edition]
 Compiled with GNAT Version: 11.3.0
 GCC 12.2.0 code generator
```

Phenotype (1):
```
GTKWave Analyzer v3.3.104 (w)1999-2020 BSI

ghw_read_type: unknown type 37
ghw_read_base: error in section TYP
Error in ghw file 'aha.ghw'.
```

Phenotype (2):
```
./crashtest:error: NULL access dereferenced
in process .crashtest(behavioral).P0
./crashtest:error: simulation failed
```

umarcor, 2023-04-19:
@bilkes can you please try whether the GHW can be opened with a more recent version of GtkWave (latest is 3.3.115)?

diogratia, 2023-04-26:
You don't specify using declaration of Dummy belongs to which 'phenotype' clearly.
```none
ghdl -e  CrashTest
```
should contain --std=08

It might be more useful when providing a tofer to have distinct test cases.

There's an extraneous back tick that shouldn't be present: 
```none
ghdl -a --std=08 CrashTest.vhdl`
```

With 

```vhdl
     signal Dummy : Specialized; -- mangled GHW file
    -- signal Dummy : Configured.T; -- produced executable crashed 
```
using
```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r92.gd8ac5d642) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
```
a nightly build for MacOS and GTKWave Analyzer v3.3.107 (the latest available [gtkwave.app](https://gtkwave.sourceforge.net/) for MacOS):

[image]

Success.

Note the gtkwave [Changelog](https://github.com/gtkwave/gtkwave/blob/master/gtkwave4/ChangeLog):
```none
3.3.106	31jul20	Fix Shift-Up/Down highlight to traverse inside groups.
		Resync ghwlib to handled unbounded arrays.
```
which has a -2008  change affecting package Abstract.

As far as 
```vhdl
    -- signal Dummy : Specialized; -- mangled GHW file
    signal Dummy : Configured.T; -- produced executable crashed 
```

That crashed:
```none
%: ./crashtest --wave=aha.ghw
Abort
%: 
```
The difference in error indications left to vagaries of the back end and likely relating to the platform. The Abort may be an artifact of not using the native crash handler.

There's every expectation there was something not handled correctly in ghdl, perhaps through package instantiation.

Using a revision of gtkwave greater than 3.3.105 should solve at least one of your three issues.