ghdl/ghdl #2420: vpi_put_value with vpiVectorVal not working

r2com, 2023-04-22:
So the above code works OK:
```
static vpiHandle clk;
  s_vpi_value val;
  int a = 1;
  val.format = vpiIntVal;
  val.value.integer = a;
  vpi_put_value(clk, &val, NULL, vpiNoDelay);
```
It drives DUT port "clk" to 1. 

However, the below code does NOT work:
```
static vpiHandle clk;
  s_vpi_value val;
  s_vpi_vecval vecVal;
  vecVal.aval = 1;
  vecVal.bval = 0;
  val.format = vpiVectorVal;
  val.value.vector = &vecVal;
  vpi_put_value(clk, &val, NULL, vpiNoDelay);
```
All I am doing, is same thing but using aval/bval vector pair, I decided to do this since its faster, because internally if I am correct the logic value in simulator is represented in this structure format, thus it should be faster than converting from integer.

But why it does not work? 

Also, during the run, the console prints out:
vpi_put_value: vpiVectorVal

any hints?

tgingold, 2023-04-22:
Yes, according to the hint, it is just because the VectorVal is not supported (probably for a single bit).

r2com, 2023-04-22:
so what would be the most effective in terms of performance way to drive single bit or vector bits then?

r2com, 2023-04-22:
can you tell me in which format the single and vector nets/ports are saved within a simulation structure?
(I am trying to do the read/write in a same format to avoid unnecessary conversion which hits performance)

r2com, 2023-04-22:
Looking further into the vpi_put_value source I found it is actually ignoring the aWhen and aFlag and does not return anything.

But aside from that I see it only supports vpiBinStrVal and vpiIntVal (the one I used and saw code working).

Now I understand.

However, my question then is - isnt it ineffective/performance slow if I constantly try to toggle signals every clock on many instances by passing integer for single vector bits? 

If I am correct, for Verilog (what VPI was originally developed for) the most effective way of passing was using aval/bval pair since this is the format Verilog simulators using inside, thus it does not result in performance hit.

But in my case using VPI with GHDL it seems that I'm going to have extra performance hit?

Do we have any workaround for GHDL for this? Other than VPI ineterface? i.e. what about VHPIDIRECT, does it allow interfacing C code directly to DUT VHDL code without any intermediate VHDL testbench? (which is what I _can_ do with VPI and GHDL but unfortunately I see some potential performance issues with my planned usage)

tmeissner, 2023-04-22:
If you want to access C functions from a VHDL testbench, you could use VHPIdirect. I use this interface in my cryptocores project: https://github.com/tmeissner/cryptocores/tree/master/aes/sim/vhdl

r2com, 2023-04-22:
> If you want to access C functions from a VHDL testbench, you could use VHPIdirect. I use this interface in my cryptocores project: https://github.com/tmeissner/cryptocores/tree/master/aes/sim/vhdl

no its not what I want. I was saying that I want a complete C/C++ verification with just VHDL DUT code, without any intermediate HDL testbench. It is doable with VPI and GHDL. But not sure if its doable with VHPIDIRECT

tgingold, 2023-04-23:
I don't think that the format has a meaningful performance impact.  That's just a binary conversion.

VPI is certainly not as efficient as direct HDL code.

closed 2023-04-23 by r2com