ghdl/ghdl #2421: VHDL2008: Assign to an unconstrained output using an aggregate and others

Paebbels, 2023-04-22:
**Description**
When assigning to an unconstrained out port using an aggregate and others, GHDL refuses this.

**Expected behavior**
I think this is a before 2008 behavior and needs an update.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use     ieee.std_logic_1164.all;

entity comp is
	port (
		output : out std_logic_vector
	);
end entity;

architecture a1 of comp is
begin
	output <= (others => '0');
	-- output <= (output'range => '0');  -- gives no error
end architecture;

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity top is
end entity;

architecture a2 of top is
	signal sig : std_logic_vector(7 downto 0);
begin
	inst : entity work.comp
		port map (
			output => sig
		);
end architecture;

```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `5d316963`

**Additional context**
This code works in Riviera-PRO and Vivado Synthesis.

--------------
/cc @stefanunrein

Paebbels, 2023-04-22:
@tgingold I added the error message to the post.

tgingold, 2023-04-22:
I think this is the effect of LRM08 9.3.3.3 case e).

I don't think it has changed between LRM93 and LRM08.

Paebbels, 2023-04-22:
Would this be then a candidate for `frelaxed`?

tgingold, 2023-04-22:
Why not, as I think the rule is useless.

tgingold, 2023-04-23:
In fact it is already part of -frelaxed!

Paebbels, 2023-04-23:
Ups.

closed 2023-04-23 by Paebbels