ghdl/ghdl #2422: VHDL2008: Use type conversion on unconstrained formal

Paebbels, 2023-04-22:
**Description**
When converting a type in a port map on the formal side and the formal is unconstrained, an error is raised.

I understand that normal function and conversion function might not be reversible, but a type conversion looking like a function is not a function, right?

Moreover, for such conversions of closely related types, it's know that input and output are structurally alike, thus in my case arrays have the same number of elements.

**Expected behavior**
I think this is a before 2008 behavior and needs an update.

This code works in Riviera-PRO and Vivado Synthesis.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity comp is
	port (
		output : out unsigned
	);
end entity;

architecture a1 of comp is
begin
	output <= (7 downto 0 => '0');  -- not using others due to issue #2421
end architecture;

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity top is
end entity;

architecture a2 of top is
	signal sig : std_logic_vector(7 downto 0);
begin
	inst : entity work.comp
		port map (
			std_logic_vector(output) => sig
		);
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `5d316963`

**Additional context**
```
$ /c/Tools/GHDL/4.0.0.dev0-mingw64-llvm/bin/ghdl.exe -a --std=08 example.vhdl
example.vhdl:31:53: type of formal conversion must be fully constrained
                        std_logic_vector(output) => sig
                                                    ^
C:\Tools\GHDL\4.0.0.dev0-mingw64-llvm\bin\ghdl.exe: compilation error
```

--------------
/cc @stefanunrein

tgingold, 2023-04-23:
Here GHDL follows LRM08 5.3.2.2 e) 3)

Paebbels, 2023-04-26:
I'll do some tests with Vivado and Riviera and come back to you, I might know how they solved that problem.

Paebbels, 2023-04-27:
LRM08 5.3.2.2 e) 3) second item:  
> For an interface object or subelement whose mode is out, buffer, inout, or linkage, if the formal part includes a conversion function or a type conversion, then the parameter subtype of that function or the type mark of the type conversion shall define a constraint for the index range corresponding to the index range of the object, and the index range is obtained from that constraint; otherwise, the index range is obtained from the object denoted by the actual designator.

Here we have a type conversion with a typemark, so:  
> then the parameter subtype of that function or the type mark of the type conversion shall define a constraint for the index range corresponding to the index range of the object

I see, in such a case the output width (subtype of that function) is derived from the conversions input ...

Ohohoh, the input here is the formal, but in my case the constraint comes from the actual. Is that the problem?

tgingold, 2023-04-27:
Yes.  How could the bounds of the formal be determined.  To be honest, that could be possible for a type conversion, but not for a function conversion.

stefanunrein, 2023-04-27:
I tested what @Paebbels suspected. If a type conversion is wrapped by a conversion function, Riviera prints an error.

Modified code:
```vhdl
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity comp is
	port (
		output : out unsigned
	);
end entity;

architecture a1 of comp is
begin
	output <= (7 downto 0 => '0');
end architecture;

library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity top is
end entity;

architecture a2 of top is
	function convert(val : unsigned) return std_logic_vector is
	begin
		return std_logic_vector(val);
	end function;

	signal sig : std_logic_vector(7 downto 0);
begin
	inst : entity work.comp
		port map (
			convert(output) => sig
		);
end architecture;
```

Error message:
> \# ACOM: Error: COMP96_0758: R:/test.vhdl : (37, 4): Either the formal parameter of the "convert" conversion function or the "output" port passed as the actual parameter to that function must be fully constrained.

tgingold, 2023-04-30:
Ok, so what should be done ?

Paebbels, 2023-04-30:
I had a discussion with @stefanunrein and such a feature is essential for code using unconstrained ports and lots of VHDL-2008 features. It reduces the amount of generic that are needed in an entity / in an instantiation.

We also see that other vendors have already implemented that feature for type conversions and they throw an error on conversion functions.

As type conversions are a feature fully controlled by the tool and it's only possible on closely related types, it should be a safe feature with no drawbacks. I'll then work on the standard and legalize it in the next revision.

Can you add this under `frelaxed` or do you prefer another flag for such extensions?

-----------
/cc @JimLewis

tgingold, 2023-04-30:
No, we can keep the same flag.

JimLewis, 2023-04-30:
If what you need is a conversion that sizes the output, have you tried:
```vhdl
architecture a2 of top is
	function convert(val : unsigned) return std_logic_vector is
	begin
		return std_logic_vector(val);
	end function;
        subtype ByteType is std_logic_vector(7 downto 0);
	signal sig : std_logic_vector(7 downto 0);
begin
	inst : entity work.comp
		port map (
			ByteType(output) => sig
		);
end architecture;
```

Then if the previous works, maybe the following will also:
```vhdl
	inst : entity work.comp
		port map (
			sig'subtype(output) => sig
		);
```

stefanunrein, 2023-05-01:
@tgingold this issue is not solved with commit 635824fb. Running the initial code from @Paebbels, I get now this bug report:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\Projekte\udp-ethernet-stack\PL\ghdl_bug\
Command line:
C:\mingw\usr\local\bin\ghdl1-llvm.exe --std=08 --work=PoC -frelaxed -Wno-hide -Wno-elaboration -Wno-port -Wno-specs -Wno-pure -PC:\
mingw\usr\local\lib/ghdl\ieee\v08\ -PC:\mingw\usr\local\lib/ghdl\ -c -o test.o test.vhdl
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : no field Type_Mark
Load address: 0x7ff7a5770000
Call stack traceback locations:
0x7ff7a5abf876 0x7ff7a582da22 0x7ff7a59b0ea3 0x7ff7a59b12cb 0x7ff7a59b2170 0x7ff7a59a4030 0x7ff7a59da542 0x7ff7a59db56c 0x7ff7a59db
781 0x7ff7a59db8e5 0x7ff7a59a362d 0x7ff7a59aa5d2 0x7ff7a597ff21 0x7ff7a5aa49b6 0x7ff7a5aa678d 0x7ff7a5aa7ed3 0x7ff7a5772c8c 0x7ff7a
57712ec 0x7ff7a5771404 0x7fff53817602 0x7fff54bc269f
******************************************************************

```

tgingold, 2023-05-02:
That's weird as the initial test is now part of the testsuite and passes.
(Note: in your first message the test was named ent.vhd, has it slightly changed ?)

stefanunrein, 2023-05-02:
The name was changed to `test.vhdl`. 

I saw now, that I didn't copy the initial example, but the modified example with the conversion-function. This one is crashing.

Furthermore, this feature is still crashing when used with an unconstrained port:
```vhdl
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity comp is
	port (
		output : out unsigned
	);
end entity;

architecture a1 of comp is
begin
end architecture;

library ieee;
use     ieee.std_logic_1164.all;

entity top is
	port(
		sig : out std_logic_vector
	);
end entity;

architecture a2 of top is
begin
	inst : entity work.comp
		port map (
			std_logic_vector(output) => sig
		);
end architecture;
```

Do you want for both of these problems a separate and new issue?

tgingold, 2023-05-02:
No, it's ok.  I will have a look.

tgingold, 2023-05-03:
I cannot reproduce the crash and neither can CI.  But I will investigate.

stefanunrein, 2023-05-03:
I tested the code from https://github.com/ghdl/ghdl/issues/2422#issuecomment-1530985741 right now with the commit 586f2ef5.

It still crashes:
```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 12.2.0
Target: x86_64-w64-mingw32
C:\Projekte\udp-ethernet-stack\PL\ghdl_bug\
Command line:
C:\mingw\usr\local\bin\ghdl1-llvm.exe --std=08 --work=PoC -frelaxed -Wno-hide -Wno-elaboration -Wno-
port -Wno-specs -Wno-pure -PC:\mingw\usr\local\lib/ghdl\ieee\v08\ -PC:\mingw\usr\local\lib/ghdl\ -c
-o test.o test.vhdl
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : trans-chap4.adb:305
Load address: 0x7ff755990000
Call stack traceback locations:
0x7ff755ce00f6 0x7ff755c8296c 0x7ff755c913d5 0x7ff755c91c5c 0x7ff755c17de9 0x7ff755c17f8d 0x7ff755c1
9105 0x7ff755c19c96 0x7ff755cbd63b 0x7ff755cb8a41 0x7ff755cb924b 0x7ff755c6da78 0x7ff755c5f94c 0x7ff
755cc558c 0x7ff755cc700d 0x7ff755cc8753 0x7ff755992ca2 0x7ff7559912ec 0x7ff755991404 0x7ffc1b0b7602
0x7ffc1cd2269f
******************************************************************

```

I'm building it locally with these commands:
```
* manually deleting everithing in the folder `build`
* ../configure --with-llvm-config
* make
* make install
```
I'm installing it over the old version. Might this be a problem?

tgingold, 2023-05-03:
Ok, I have now a reproducer.

closed 2023-05-07 by tgingold

Paebbels, 2023-05-22:
I created https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/294 to modify the LRM.

commit 96423737 (2023-05-01): vhdl-sem_assocs: relax constrained on type conversion of formal

Fix #2422