ghdl/ghdl #2423: SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:2819

Str1ien, 2023-04-27:
**Description**
When trying to generate the Makefile to compile the project with "ghdl --gen-makefile --ieee=synopsys --workdir=WORK testbench > Makefile", I get this exception. The different components are distributed in different files and it used to compile just fine without the last component we added in class, a memory subsystem made of a cache memory, 1 data memory, 1 scratch memory and another component that acts as master of a semi-syncronous bus.

**How to reproduce?**
See the files attached
[src.zip](https://github.com/ghdl/ghdl/files/11345947/src.zip)

**Context**
- OS: WSL 2 (Ubuntu) on Windows 10

**Original Bug block**
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (Ubuntu 1.0.0+dfsg-6) [Dunoon edition]
Compiled with GNAT Version: 10.3.1 20211117
Target: x86_64-linux-gnu
/mnt/d/Unizar/Informatica/2º/2o Cuatri/AOC_2/P2/src_pruebas_mips/
Command line:
/usr/bin/ghdl-llvm --gen-makefile --ieee=synopsys --workdir=WORK testbench
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:2819
Call stack traceback locations:
0x7f30da4df542 0x7f30da805bc8 0x7f30da823e9f 0x7f30da976eac 0x7f30da97726d 0x7f30da977332 0x7f30da976249 0x7f30da976986 0x7f30da97727d 0x7f30da977332 0x7f30da976249 0x7f30da977a47 0x7f30da9855ec 0x7f30da9a6460 0x7f30da9a5b9b 0x7f30da8c1c4a 0x7f30da9c304d 0x7f30da7660a9 0x7f30da059d8e 0x7f30da059e3e 0x7f30da764e83 0xfffffffffffffffe
******************************************************************

tgingold, 2023-04-27:
I cannot reproduce the crash.

What are the steps to reproduce it ?

tmeissner, 2023-04-27:
@Str1ien  GHDL release: 1.0.0 is very old, could you try it again with a more recent version?

Str1ien, 2023-04-27:
@tgingold this is what we were tought on how to set up and generate the makefile:
- Make a directory to store the .o files
- ghdl -i --ieee=synopsys -fexplicit --workdir=WORK *.vhd
- ghdl --gen-makefile --ieee=synopsys --workdir=WORK testbench > Makefile

tgingold, 2023-04-27:
So the bug has certainly been fixed since ghdl 1.0, you should try the latest release.

Str1ien, 2023-04-27:
@tgingold Sure will, thanks the for the help. Btw, I downloaded with apt-get install on wsl 2 (ubuntu), why didn't I get the last version?

closed 2023-04-27 by Str1ien