ghdl/ghdl #2426: Generic creates non-static choice

stefanunrein, 2023-04-29:
**Description**
Using a generic for setting portions of a constant gives the error `not static choice exclude others choice`. 

**Expected behavior**
The generic should be static enough to use it for a constant. This seems to be a false error.

**How to reproduce?**

```vhd :file: ent.vhd

library IEEE;
use     IEEE.std_logic_1164.all;

entity my_entity is
	generic (
		foo   : natural  := 3
	);
end entity;

architecture rtl of my_entity is
	constant my_length       : natural  := 32;
	constant offset         : natural  := 8;
	
	constant test  : std_logic_vector(32 -1 downto 0)        := (32 -1        downto 7      => '1', others => '0'); -- Working
	constant test1 : std_logic_vector(my_length -1 downto 0) := (my_length -1 downto 7      => '1', others => '0'); -- Working
	constant test2 : std_logic_vector(32 -1 downto 0)        := (32 -1        downto foo    => '1', others => '0'); -- Not working
	constant test3 : std_logic_vector(32 -1 downto 0)        := (32 -1        downto offset => '1', others => '0'); -- Working
	constant test4 : std_logic_vector(my_length -1 downto 0) := (my_length -1 downto offset => '1', others => '0'); -- Working
	constant test5 : std_logic_vector(my_length -1 downto 0) := (my_length -1 downto foo    => '1', others => '0'); -- Not working
	constant test6 : std_logic_vector(my_length -1 downto 0) := (offset -1    downto 0      => '1', others => '0'); -- Working
	constant test7 : std_logic_vector(my_length -1 downto 0) := (offset -1    downto foo    => '1', others => '0'); -- Not working
	constant test8 : std_logic_vector(my_length -1 downto 0) := (foo -1       downto 0      => '1', others => '0'); -- Not working

begin

end architecture;

```

**Context**
Please, provide the following information:

- OS: Windows with MinGW-64
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `a3b1311b`

File analyzed with this command:
```
ghdl -a --std=08 --work=PoC -frelaxed -Wno-hide -Wno-elaboration ${file}
```

/cc @Paebbels

tgingold, 2023-04-30:
No, GHDL is correct.  See LRM08 9.3.3.3:

A named association of an array aggregate is allowed to have a choice that is not locally static, or likewise a choice that is a null range, only if the aggregate includes a single element association and this element association has a single choice.

stefanunrein, 2023-04-30:
This is working in my code with Riviera 2022.04 and Vivado 2019.2.1. @tgingold can we add it under `frelaxed`, or should I change my code? @Paebbels what do you think?

tgingold, 2023-04-30:
Unfortunately this is not something that could be easily added to -frelax.

diogratia, 2023-04-30:
When you find what looks like arbitrary semantic rules in the standard it reflects differences in implementation methodology being embraced by the standard. Essentially two or more different tools are using at least two different methods of constructing aggregate values here.

Instead of arbitrary applications of relaxed semantic rules it would be more desirable to have vendor participation in the standardization process where they can collectively decide who to have for lunch or who to allow to call their implementation conforming to the standard. That's been offset by the inability to provide a validation test suite that's universally recognized as conforming to the standard as well as guerilla politics.

Look at the standardized requirement for a separator between identifiers and adjacent abstract literals (lexical elements such as appear in physical literals (syntactical productions). It's a contrast between separate lexical and syntactical analysis where there are cases for example in -2008 with a user defined physical type having units sharing overloaded identifiers with characters used to describe bit literal base where a bit string literal with immediately preceded by an integer length can't be distinguished from a 'run-on' physical literal without enforcing the rule. The language describing the requirement is almost hidden:

15.3 Lexical elements, separators, and delimiters

>In some cases an explicit separator is required to separate adjacent lexical elements (namely when, without
separation, interpretation as a single lexical element is possible). A separator is either a space character
(SPACE or NBSP), a format effector, or the end of a line. ...

Having this rule allowed length integers to be added to based bit string literals.

In the present case for constructing aggregates the use of `-frelaxed` would be considered non portable and suggest you'd test both cases with and without the command line flag and anticipate different results. That likely holds true for other uses as well.

Besides using both test cases as aides in any VASG/P1076/ISAC effort to change the standard you might find the set of relaxed features don't perfectly match between various implementations which suggests feature sets of relaxed rules, e.g. `-fsynopsys` and now potentially other tool vendors while the standardization of VHDL might otherwise allow new semantics to be gracefully added. (And about here you can also see the attractiveness of new standards efforts for radical changes, although these have historically failed for VHDL.)

Here what's unsettling is Riviera-PRO and Vivado apparently making claims through error messages that aren't backed up by the black and white semantics of the standard. And none of this is new to -2008, the same *denotes the subtype* is present in -1993.

diogratia, 2023-04-30:
When you find what looks like arbitrary semantic rules in the standard it reflects differences in implementation methodology being embraced by the standard. Essentially two or more different tools are using at least two different methods of constructing aggregate values here.

Instead of arbitrary applications of relaxed semantic rules it would be more desirable to have vendor participation in the standardization process where they can collectively decide who to have for lunch or who to allow to call their implementation conforming to the standard. That's been offset by the inability to provide a validation test suite that's universally recognized as conforming to the standard as well as guerilla politics.

Look at the standardized requirement for a separator between identifiers and adjacent abstract literals (lexical elements such as appear in physical literals (syntactical productions). It's a contrast between separate lexical and syntactical analysis where there are cases for example in -2008 with a user defined physical type having units sharing overloaded identifiers with characters used to describe bit literal base where a bit string literal with immediately preceded by an integer length can't be distinguished from a 'run-on' physical literal without enforcing the rule. The language describing the requirement is almost hidden:

15.3 Lexical elements, separators, and delimiters

>In some cases an explicit separator is required to separate adjacent lexical elements (namely when, without
separation, interpretation as a single lexical element is possible). A separator is either a space character
(SPACE or NBSP), a format effector, or the end of a line. ...

Having this rule allowed length integers to be added to based bit string literals.

In the present case for constructing aggregates the use of `-frelaxed` would be considered non portable and suggest you'd test both cases with and without the command line flag and anticipate different results. That likely holds true for other uses as well.

Besides using both test cases as aides in any VASG/P1076/ISAC effort to change the standard you might find the set of relaxed features don't perfectly match between various implementations which suggests feature sets of relaxed rules, e.g. `-fsynopsys` and now potentially other tool vendors while the standardization of VHDL might otherwise allow new semantics to be gracefully added. (And about here you can also see the attractiveness of new standards efforts for radical changes, although these have historically failed for VHDL.)