ghdl/ghdl #2427: Setting attribute on type creates error

stefanunrein, 2023-04-29:
**Description**
Setting an attribute on a type, creates an error ` is not of class 'type'`. This is working at least on records.

**Expected behavior**
Creating a type should be of class `type` and therefore don't create this error.

**How to reproduce?**

```vhd :file: ent.vhd
library IEEE;
use IEEE.STD_LOGIC_1164.all;

package pkg is
	attribute Bits        : integer;

	type MY_TYPE is array (5 downto 0) of std_logic_vector(7 downto 0);
	attribute Bits   of MY_TYPE : type is 6 * 8; --Error
	
	type MY_RECORD is record
		sig      : std_logic;
	end record;
	attribute Bits   of MY_RECORD : type is 3 + 3; --No Error
	
end package;
```

**Context**

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `a3b1311b`

Analyzed with:
```
ghdl -a --std=08 --work=PoC -frelaxed -Wno-hide -Wno-elaboration ${file}
```

/cc @Paebbels

diogratia, 2023-04-30:
This really deserves the error messages:
```none
%: ghdl -a --std=08 -frelaxed pkg.vhdl
pkg.vhdl:8:19:error: subtype "my_type" is not of class 'type'
        attribute Bits   of MY_TYPE : type is 6 * 8; --Error
                  ^
pkg.vhdl:7:14:error: "my_type" declares both an anonymous type and a named subtype
        type MY_TYPE is array (5 downto 0) of std_logic_vector(7 downto 0);
             ^
ghdl:error: compilation error
```
Which tell us to pay attention to whether MY_TYPE is the name of type or the subtype for a fully constrained type declaration:

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.all;

package pkg is
	attribute Bits        : integer;
    type MY_TYPE is array (5 downto 0) of std_logic_vector (7 downto 0);
	-- attribute Bits   of MY_TYPE : type is 6 * 8; --Error
    attribute Bits of MY_TYPE : subtype is 6 * 8;
    
	type MY_RECORD is record
		sig      : std_logic;
	end record;
    attribute Bits   of MY_RECORD : type is 3 + 3; --No Error
    
end package;

entity display_attr is
end entity;

architecture foo of display_attr is
    use work.pkg.all;
begin
    process
    begin
        report "attribute Bits of MY_TYPE is " & integer'image(MY_TYPE'Bits);
        report "attribute Bits of MY_RECORD is " & integer'image(MY_RECORD'Bits);
        wait;
    end process;
end architecture;
```

Which when analyzed, elaborated and simulated produces:
```none
%: ghdl -a --std=08 -frelaxed pkg.vhdl
%: ghdl -e --std=08 -frelaxed display_attr
%: ghdl -r display_attr
pkg.vhdl:25:9:@0ms:(report note): attribute Bits of MY_TYPE is 48
pkg.vhdl:26:9:@0ms:(report note): attribute Bits of MY_RECORD is 6
%: 
```
5.3.2 Array types  
5.3.2.1 General  
>A constrained array definition similarly defines both an array type and a subtype of this type:  
>>— The array type is an implicitly declared anonymous type; this type is defined by an (implicit)
unbounded array definition, in which the element subtype indication either denotes the base type of
the subtype denoted by the element subtype indication of the constrained array definition, if that
subtype is a composite type, or otherwise is the element subtype indication of the constrained array
definition, in which the type mark of each index subtype definition denotes the subtype defined by
the corresponding discrete range.  
>>— The array subtype is the subtype obtained by imposition of the index constraint on the array type and, if the element subtype indication of the constrained array definition denotes a fully or partially constrained composite subtype, imposition of the constraint of that subtype as an array element constraint on the array type.  
>  
>If an array definition that defines both an array type and a subtype of that type is given for a type declaration, the simple name declared by this declaration denotes the array subtype.

Seeing the error messages triggered the memory of having read 5.3.2 two days previous.

stefanunrein, 2023-04-30:
This works contradicting to Vivado 2019.2.1 and Riviera-PRO 2022.04. If setting it to subtype, I get there the following error message:

Riviera-PRO 2022.04:
```
The class of named entities denoted by the designators must be the same as that denoted by the entity class.
```

Vivado 2019.2.1:
```
ERROR: 'MY_TYPE' is not a subtype
```

tgingold, 2023-04-30:
So both Vivado and Riviera are wrong.  Let add that to -frelaxed

closed 2023-04-30 by tgingold

diogratia, 2023-04-30:
Here what's unsettling is Riviera-PRO and Vivado apparently making claims through error messages that aren't backed up by the black and white semantics of the standard. And none of this is new to -2008, the same *denotes the subtype* is present in -1993.

stefanunrein, 2023-05-01:
@tgingold i have tested the commit 635824fb. Here i don't get the error message anymore. But it looks like the attribute is still not available:
```
Error: ../../src/stack.pkg.vhdl:73:40: subtype "t_net_mac_address" was not annotated with attribute "bits"
Error:                 0 => T_NET_MAC_ADDRESS'bits,
Error:                                        ^
```

tgingold, 2023-05-02:
Yes, you're right.

commit aeb15c6d (2023-04-30): vhdl-sem_specs: allow attribute specification on a type if -frelaxed

Fix #2427