ghdl/ghdl #2431: Library naming to use -PDIR "Add DIR in the library search path"

cweickhmann, 2023-05-04:
**Is your feature request related to a problem? Please describe.**
When using the `-P` flag, `ghdl` (versions 1 through 3) refuse to include the library I placed elsewhere. Example:

* My project: `~/myproject`
* VHDL-Extras Library: `~/vhdl-extras`

I built VHDL-Extras and have a file `extras-obj08.cf` in `~/vhdl-extras`. The library is called `extras`.

When invoking, I receive an error

```bash
~/myproject$ ghdl -a --std=08 -P../vhdl-extras myfile.vhdl

myfile.vhdl:1:9:error: cannot find resource library "extras"
library extras;
             ^
```

However, if I just copy or link the file `extras-obj08.cf` into `~/myproject`, it just works (with or without the `-P` flag).

**Describe the solution you'd like**
I suspect, there are naming conventions for the compiled library (the `.cf` file) that are not documented. It would be nice to clarify this in the documentation. Also, giving a minimal working example in the docs would be very nice.

**Describe alternatives you've considered**
Or maybe enlighten me, if I've misunderstood something entirely ;-)

cweickhmann, 2023-05-08:
I think, it's got to do with the `-a` flag I used, and also used to build the `extras` library. If I use `-i` instead, then it does work.

It now *does* work, and here is how it does:

## Import units of `extras`

```bash
~/extras $ LIST_OF_FILES="binaryio ..." # and so on, it just skips a few files that cause trouble with ghdl
~/extras $ ghdl -i -g --std=08 --work=extras -frelaxed ${LIST_OF_FILES}
```

Now, I have a file `extras-obj08.cf` in `~/extras`.

## "Build" My Stuff and Run Simulation

I place simulation results in the sub-folder `simulation`, hence the additional `--workdir` flag. This is optional.

```bash
~/myproject $ ghdl -i -g --std=08 -P../extras -frelaxed --workdir=simulation/ --work=work file1.vhdl file1_tb.vhdl
~/myproject $ ghdl -m -g --std=08 -P../extras -frelaxed --workdir=simulation/ --work=work file1.vhdl
~/myproject $ ghdl -m -g --std=08 -P../extras -frelaxed --workdir=simulation/ --work=work file1_tb.vhdl
~/myproject $ ghdl -r -g --std=08 -P../extras -frelaxed --workdir=simulation/ --work=work file1_tb.vhdl --wave=simulation/file1.ghw
```

So, in short: using `-i` and `-m` gets me where I want to go. Using `-a` and `-e` does not. And I think it is because of the way I would have to generate the `extras-obj08.cf` file, which I apparently fail to create properly.

umarcor, 2023-05-08:
@cweickhmann please provide the failing reproducer. That is, the sequence of `-a` and `-e` commands that do fail.

OTOH, the syntax of `-m` (http://ghdl.github.io/ghdl/using/InvokingGHDL.html#make-m) does not include a filename at the end of the command. It is expecting a primary/top unit, the elaboration target. I wonder why/how the commands you showed above did not fail.