ghdl/ghdl #2433: Argument max-stack-alloc is not passed to executable

stefanunrein, 2023-05-05:
**Description**
When having large data structures, GHDL reports this error:
```
error: declaration of a too large object (884567 > --max-stack-alloc=1024 KB)
```

When setting this argument to `ghdl -r` or `ghdl --elab-run` the following is reported:
```
C:\mingw\usr\local\bin\ghdl.exe: unknown command option '--max-stack-alloc=1024000'
```

Executing the TB-executable created from GHDL is just working fine:
```
$ ./udp_eth_tb.exe --max-stack-alloc=2024000
```

My guess is, that the parameter is not passed through GHDL internally.

**Expected behaviour**
This parameter should be passed correctly to the executable.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd

library IEEE;
use     IEEE.std_logic_1164.all;
use     IEEE.numeric_std.all;

entity my_entity is
end entity;

architecture rtl of my_entity is
begin
	assert false report "Dummy." severity Note;

end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
echo "Analyzing..."
ghdl -a --std=08 --work=test ent.vhd
echo "Elaborating..."
ghdl -e --std=08 --work=test my_entity
echo "Simulation..."
ghdl -r --std=08 --work=test --max-stack-alloc=1024000 my_entity
```
**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `586f2ef5`

umarcor, 2023-05-05:
@stefanunrein, try `ghdl -r --std=08 --work=test my_entity --max-stack-alloc=1024000`. If that works, pay attention to `[options...]` (before the entity|primary unit) and `[simulation_options...]` (after the entity|primary|secondary) in https://ghdl.github.io/ghdl/using/InvokingGHDL.html#run-r.

stefanunrein, 2023-05-05:
It is indeed working when changing the order. Thank you @umarcor for pointing me to the solution

closed 2023-05-05 by stefanunrein

Paebbels, 2023-05-05:
@tgingold could we print a hint like this:
`If 'xxx=yyy' is a run parameter, please place it after toplevel name.`

Looks like some people (including my self) stumble from time to time about that problem. So my suggestion is to improve the usability for (new) users.

tgingold, 2023-05-07:
Ah I see. Maybe we could improve the error message.