ghdl/ghdl #2434: Large object check might be wrong

stefanunrein, 2023-05-05:
**Description**
In our code, we have a large record-vector for a configuration. With the following elements:
* Sting with 64 characters
* 32bit unsigned
* Enum with 12 values
* slv, 32bit
* slv, 32bit
* boolean

Counting these together as 1B per char, 1B for enum, 1B per std_logic and 1B for the boolean, i get 844B for the record. A vector is created from this record with 128 elements, which might result in 108kB RAM usage. 

When running the code, this is reported:
```
my_tb.exe:error: declaration of a too large object (884567 > --max-stack-alloc=202400 KB)
```

Now the question is: Is the integer 884567 reported here in `KB` or Byte? A factor of 8 seems to fit, also considering, that there is more in the design. A factor of 8000 looks way to much.

Paebbels, 2023-05-05:
Like Stefan, I suspect there is a kB vs. B problem in the compare and/or error message. Also the simulation is to fast if it would really generate GB of data on the stack ;).

I also suggest to add the unit to the left side and write `KB` as `kB`.

Paebbels, 2023-05-06:
Here a reproducer:

```vhdl
entity e is
end entity;

architecture a of e is
	subtype ConfigString is string(1 to 128);
	
	type RegisterDescription is record
		name1 : ConfigString;   -- 128  128
		name2 : ConfigString;   -- 128  256
		name3 : ConfigString;   -- 128  384
		name4 : ConfigString;   -- 128  512
	end record;
	
	type RegisterConfiguration is array(natural range <>) of RegisterDescription;
	
	function CreateConfig(Count : positive) return RegisterConfiguration is
		variable Config : RegisterConfiguration(0 to Count - 1);
	begin
		return Config;
	end function;
	
	constant RegConfig : RegisterConfiguration := CreateConfig(512);  -- 256 kB (+ overhead)
begin
	assert false report "RegConfig: " & integer'image(RegConfig'length) severity note;
end architecture;
```

Called with:
```bash
ghdl -a --std=08 struct.vhdl
ghdl -e --std=08 e

ghdl -r --std=08 e
.\e.exe:error: declaration of a too large object (256 > --max-stack-alloc=128 KB)
.\e.exe:error: error during elaboration

ghdl -r --std=08 e --max-stack-alloc=256
struct.vhdl:26:9:@0ms:(assertion note): RegConfig: 512
```

I still think the writing of `256` needs to be adjusted to `256 kB` denoting the correct unit.

tgingold, 2023-05-07:
Yes indeed, the unit is missing.

umarcor, 2023-05-08:
Probably related: #822