ghdl/ghdl #2437: Elaboration of entity with array of vectors fails in simulation when debugging flag is set

cweickhmann, 2023-05-12:
**Description**
I am trying to implement a generic tapped delay line. To this end, I want to use an array of signed whose lengths are set by a generic.

My MWE consists of the entity source together with the type definition, and a testbed. I am observing this issue with GHDL 2.0 and 3.0. I use --std=08, i.e. VHDL-2008 and the debug flag `-g`.

As pointed out in the [Stackoverflow thread I posted](https://stackoverflow.com/questions/76229346/elaboration-of-entity-with-array-of-vectors-fails?noredirect=1#comment134438218_76229346), the error disappears when the `-g` flag is omitted.

Analysis and elaboration, or include (`-i`) and make (`-m`) work fine. When running the simulation, it does not complete (see bug error message below).

**Expected behaviour**
`ghdl run` should work regardless of `-g` being set or not.

**How to reproduce?**

```vhd :file: tapped_delay_line.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

package common is
    type signed_array is array(natural range <>) of signed;
end package;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.common.all;

entity tapped_delay_line is
    generic (
        RESET_ACTIVE_LEVEL: std_logic := '0'; -- active low
        REGISTER_FIRST_STAGE: boolean := true;
        NUM_TAPS: positive := 1;
        DATA_WIDTH: positive := 8
    );
    port (
        Clock: in std_logic;
        Reset: in std_logic;
        Enable: in std_logic;
        
        Data: in signed(DATA_WIDTH-1 downto 0);
        Taps: out signed_array(0 to NUM_TAPS-1)(DATA_WIDTH-1 downto 0)
    );
end entity;

architecture behavioural of tapped_delay_line is
begin
    process(Clock)
    begin
        if Enable = '1' and Reset = RESET_ACTIVE_LEVEL then
            if rising_edge(Clock) then
                for i in 2 to NUM_TAPS-1 loop
                    Taps(i) <= Taps(i-1);
                end loop;
                
                Taps(1) <= Data;
            end if;
        else
            for i in 1 to NUM_TAPS-1 loop
                Taps(i) <= to_signed(0, DATA_WIDTH);
            end loop;
        end if;
    end process;
    
end behavioural;
```

```vhd :file: tapped_delay_line_tb.vhdl

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

use std.env.finish;

library work;
use work.common.all;

entity tapped_delay_line_tb is
end entity;

architecture sim of tapped_delay_line_tb is
    constant CLK_PERIOD: time := 0.5 ns;
    constant SIM_DURATION: time := 5 ns;
    constant RESET_ACTIVE_LEVEL: std_logic := '0';
    
    constant DATA_LEN: positive := 8;
    constant NB_TAPS: positive := 4;
    
    signal clk, rst: std_logic := '1';
    
    signal dat: signed(DATA_LEN-1 downto 0) := to_signed(0, DATA_LEN);
    signal taps: signed_array(0 to NB_TAPS-1)(DATA_LEN-1 downto 0);
begin
    tdl: entity work.tapped_delay_line
        generic map(
            RESET_ACTIVE_LEVEL => '0',
            REGISTER_FIRST_STAGE => true,
            NUM_TAPS => NB_TAPS,
            DATA_WIDTH => DATA_LEN
        )
        port map(
            Clock => clk,
            Reset => rst,
            Enable => not rst,
            Data => dat
            -- Taps => taps
        );
    
    data0: process(clk)
    begin
        if rst = RESET_ACTIVE_LEVEL and rising_edge(clk) then
            dat <= dat + 1;
        end if;
    end process;
    
    rst0: process
    begin
        rst <= not RESET_ACTIVE_LEVEL;
        wait for CLK_PERIOD*3;
        rst <= RESET_ACTIVE_LEVEL;
        wait;
    end process;
    
    clock0: process
    begin
        clk <= not clk;
        wait for CLK_PERIOD/2;
    end process;
    
    sim0: process
    begin
        report "Process started...";
        wait for SIM_DURATION;
        report "Waiting ended, finishing...";
        finish;
    end process;
    
end sim;
```

I let GDHL analyse the code and elaborate it like so:

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -frelaxed -g tapped_delay_line.vhdl
ghdl -a --std=08 -frelaxed -g tapped_delay_line_tb.vhdl
ghdl -e --std=08 -frelaxed -g tapped_delay_line
ghdl -e --std=08 -frelaxed -g tapped_delay_line_tb
```

The simulation, however, fails:

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -r --std=08 -frelaxed -g tapped_delay_line_tb
```

**Context**

- OS: Linux Ubuntu 22.04.2 LTS
- Origin:
  - [x] Released binaries: `https://github.com/ghdl/ghdl/releases/download/v3.0.0/ghdl-gha-ubuntu-20.04-mcode.tgz`
  - [x] Released binaries: `https://github.com/ghdl/ghdl/archive/refs/tags/v2.0.0.tar.gz`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 3.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 10.4.0
Target: x86_64-linux-gnu
/home/user/myproject
Command line:
/home/user/myproject/ghdl-3.0.0/build/ghdl_mcode -e --std=08 -frelaxed -g tapped_delay_line_tb
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : ortho_code-dwarf.adb:522
Call stack traceback locations:
0x7f3544e5f542 0x55ac3b187bd6 0x55ac3b188566 0x55ac3b188e7b 0x55ac3b188daf 0x55ac3b18915c 0x55ac3b1896ad 0x55ac3b18757d 0x55ac3b1a3453 0x55ac3b1a35a3 0x55ac3b19662c 0x55ac3b1af319 0x55ac3b4d859c 0x55ac3b4d88a3 0x55ac3b4f9251 0x55ac3b4ba9da 0x55ac3b4adebe 0x55ac3b4bf2de 0x55ac3b4b9560 0x55ac3b50418d 0x55ac3b43e4f8 0x55ac3b42981a 0x55ac3b5055fe 0x55ac3b0bded5 0x7f3544829d8e 0x7f3544829e3e 0x55ac3b0bc353 0xfffffffffffffffe
******************************************************************

```

**Additional context**
I haven't checked personally, but a SO user states the same happens with GHDL 1.0.0.

tmeissner, 2023-05-12:
`-g` isn't a debugging flag. Instead it's used to set generics of simulations top-level. More information in the GHDL docs: https://ghdl.github.io/ghdl/using/Simulation.html#cmdoption-ghdl-gGENERIC

Nevertheless GHDL shouldn't crash when used the flag in an incorrect way.

cweickhmann, 2023-05-12:
Fair point, but then the help text in the ghdl binary should be changed accordingly:

```bash
~/Downloads/ghdl/kernel-6/ghdl-3.0.0/build/ghdl_mcode --help -r
elab-run [OPTS] UNIT [ARCH] [RUNOPTS]
  Elaborate and run design UNIT
  aliases: --elab-run, -r, run
Main options (try --options-help for details):
 --std=XX
   Use XX as VHDL standard (87,93c,93,00,02 or 08)
 --work=NAME
   Set the name of the WORK library
 -PDIR
   Add DIR in the library search path
 --workdir=DIR
   Specify the directory of the WORK library
 -fsynopsys
   Allow to use synopsys packages in ieee library
 -frelaxed
   Relax semantic rules
 -fexplicit
   Gives priority to explicit operator redefinitions
 -g             Generate debugging informations
 -g0            Do not generate any debugging informations
 --debug-be=X   Set X internal debugging flags
 --snap=FILE    Write memory snapshot to FILE
 --expect-failure  Expect analysis/elaboration failure
 --time-resolution=UNIT   Set the resolution of type time
            UNIT can be fs, ps, ns, us, ms, sec or auto

```

See: `-g             Generate debugging informations`.

Or does `-g` change meaning depending on its position (before or after `-r`)?

closed 2023-05-13 by tgingold

tgingold, 2023-05-15:
Even for mcode, -g is used to generate debug information.  But this is mainly for developers.

cweickhmann, 2023-05-16:
I run into this general situation a tad more often, actually. Compiling works, but simulation fails with a pretty cryptic error.  I thought that's what `-g` was for.

What options do I have to pin down the error (say for example `SYSTEM.ASSERTIONS.ASSERT_FAILURE : ortho_code-consts.adb:636`)?

tgingold, 2023-05-16:
For this kind of failures, please file the sources or a reproducer.
If you want to pin down the error, you need to run ghdl within gdb and try to understand the construct which generates the crash.
Creating a small reproducer might be difficult, so it is ok to file all the sources.  However, it is much easier to handle if the sources are standalone (ie do not depend on external sources).

commit 641e6d81 (2023-05-13): ortho_code-dwarf: emit inner element type of subarray.

Fix #2437