ghdl/ghdl #2438: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2667

LHLaurini, 2023-05-12:
**Description**
ghdl raises an assertion when running `synth` with FFs sensible to different edges. See code below.

**Expected behaviour**
Expected not to raise an assertion.

**How to reproduce?**

Shortest MWE I was able to produce (makes no sense, of course):
```vhd :file: test_case.vhdl

library ieee;
use ieee.std_logic_1164.all;

entity test_case is
	port (
		clk : in  std_logic
	);
end entity;

architecture arch of test_case is

	signal a : std_logic;
	signal b : std_logic;

begin

	process (clk)
	begin

		if rising_edge(clk) then
			a <= '0';
		elsif falling_edge(clk) then
			b <= '0';
		end if;

	end process;

end architecture;

```

It works if instead I do
```vhdl
		if rising_edge(clk) then
			a <= '0';
		elsif falling_edge(clk) then
			a <= '0';
		end if;
```

Command line:
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl analyze test_case.vhdl && ghdl elaborate test_case && ghdl synth test_case
```

**Context**
Please, provide the following information:

- OS: Arch Linux x64
- Origin:
  - [ ] Package manager: `version`
  - [x] Released binaries: [`tarball_url`](https://github.com/YosysHQ/oss-cad-suite-build/releases/download/2023-05-12/oss-cad-suite-linux-x64-20230512.tgz) (from oss-cad-suite-build)
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r195.g038f55c37.dirty) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/user/fpga/spi/
Command line:
/home/user/oss-cad-suite/libexec/ghdl synth test_case
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-memories.adb:2667
Call stack traceback locations:
0x7f9558959d12 0x7f9558cf7982 0x7f9558cf7c84 0x7f9558cf86ec 0x7f9558cf9220 0x7f9558cf9bd2 0x7f9558e130d2 0x7f9558e5c73c 0x7f9558e5caaa 0x7f9558e5cc54 0x7f9558e27e58 0x7f9558e28282 0x7f9558e28358 0x7f9558d66cbd 0x7f9558ea8f5a 0x7f9558e6529b 0x7f9558eb1f46 0x7f9558bec38f 0x7f95585280b1 0x7f9558beaf1c 0xfffffffffffffffe
******************************************************************
```

**Additional context**
None for now.

tgingold, 2023-05-13:
Ok it shouldn't crash but this is not usually accepted by synthesis.

LHLaurini, 2023-05-13:
> Ok it shouldn't crash but this is not usually accepted by synthesis.

Hmm, I'm not sure why not.

At least the FPGA family I'm using (iCE40) has FFs sensible to either edge (e.g. `SB_DFF` for rising-edge and `SB_DFFN` for falling-edge), which is what I'd expect this to synthesize to if the FFs were actually connected to outputs. Something like this (generated by yosys):

[image]

If I make both sensible to rising-edge or to falling-edge, things work as expected. The problem is only if I mix them.

Paebbels, 2023-05-13:
IEEE Std. 1076.6-2004 mentions `rising_edge` and `falling_edge` in a single process, but with 2 independent `if`-statements not in an `if`-`elsif`-statement.

Paebbels, 2023-05-13:
The problem is, there is no specification of two functions like `rising_edge` and `falling_edge`, that these functions are mutual exclusive to each other.

LHLaurini, 2023-05-13:
Right, I forgot about the standard. Section "6.1.3.3 Edge-sensitive storage with one or more clocks" seems to be relevant. If I understand it correctly, as long as the check edge conditions are mutually exclusive, the process should be synthesizable.

tgingold, 2023-05-16:
I am testing a fix for the crash.

I simply fear that the reproducer is too simple and the fix may not fix your initial issue.  We will see.

closed 2023-05-21 by tgingold