ghdl/ghdl #2443: Follow up on 2427 - Attribute Specification on Array Types

Paebbels, 2023-05-22:
**Description**
This is a followup on #2427.

When creating a bug report for Aldec, I created a more extensive reproducer and I also checked the LRM on array types. The behavior mentions by @diogratia (an anonymous type and an implicit subtype with that name) applies to:
* *unbounded* array types, as well as
* *unconstrained* array types.

My test now covers all combinations and shows, GHDL is wrong for *unbounded* array types.  
Maybe that's a change from VHDL-1993 to 2002/2008.

> **NOTE** The rules concerning constrained type declarations mean that a type declaration with a constrained array definition such as:
>
>     type T is array (POSITIVE range MIN_BOUND to MAX_BOUND) of ELEMENT;
>
> is equivalent to the sequence of declarations
>
>     subtype index_subtype is POSITIVE range MIN_BOUND to MAX_BOUND;
>     type array_type is array (index_subtype range <>) of ELEMENT'BASE;
>     subtype T is array_type (index_subtype)element_constraint;
>
> where index_subtype and array_type are both anonymous and element_constraint is the constraint that applies to the subtype ELEMENT. Consequently, T is the name of a subtype and all objects declared with this type mark are arrays that have the same index range.
>
> Similarly, a type declaration with an unbounded array definition whose element subtype indication denotes a partially or fully constrained subtype such as
>
>     type T is array (INTEGER range <>) of STRING(1 to 10);
>
> is equivalent to the sequence of declarations
>
>     type array_type is array (INTEGER range <>) of STRING'BASE;
>     subtype T is array_type (open)(1 to 10);

**Expected behavior**
For `MatrixSubType_1` also entity class `subtype` should be requested.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd

entity ent is
end entity;

architecture a of ent is
	attribute TotalBits : integer;

	type ArrayType is array(natural range <>) of bit;
	attribute TotalBits of ArrayType : type is -1 * 1;

	type ArraySubtype is array(natural range 0 to 3) of bit;
	attribute TotalBits of ArraySubtype : subtype is ArraySubtype'length * 1;

	type MatrixType is array(natural range <>) of bit_vector;
	attribute TotalBits of MatrixType : type is -1;

	type MatrixSubType_1 is array(natural range <>) of bit_vector(7 downto 0);
	attribute TotalBits of MatrixSubType_1 : subtype is -1 * MatrixSubType_1'element'length;

	type MatrixSubType_2 is array(natural range 0 to 3) of bit_vector(7 downto 0);
	attribute TotalBits of MatrixSubType_2 : subtype is MatrixSubType_2'length * MatrixSubType_2'element'length;
begin
	assert false report to_string(ArrayType'TotalBits)    severity note;
	
	assert false report to_string(ArraySubtype'TotalBits) severity note;
	
	assert false report to_string(MatrixType'TotalBits) severity note;
	
	assert false report to_string(MatrixSubType_1'element'length) severity note;
	assert false report to_string(MatrixSubType_1'TotalBits) severity note;

	assert false report to_string(MatrixSubType_2'length) severity note;
	assert false report to_string(MatrixSubType_2'element'length) severity note;
	assert false report to_string(MatrixSubType_2'TotalBits) severity note;
end architecture;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS: Windows 11 - MinGW64
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `0a760f16`

**Additional context**
GHDL output:
```
$ /c/Tools/GHDL/4.0.0.dev0-mingw64-llvm/bin/ghdl -a --std=08 type-attributes.vhdl
type-attributes.vhdl:19:19: type "matrixsubtype_1" is not of class 'subtype'
        attribute TotalBits of MatrixSubType_1 : subtype is -1 * MatrixSubType_1'element'length;
                  ^
type-attributes.vhdl:29:55: type "matrixsubtype_1" was not annotated with attribute "totalbits"
        assert false report to_string(MatrixSubType_1'TotalBits) severity note;
                                                      ^
C:\Tools\GHDL\4.0.0.dev0-mingw64-llvm\bin\ghdl.exe: compilation error
```

tgingold, 2023-05-22:
Ah, the creates compatibility issues.
Some vhdl-93 code is not anymore correct in vhdl-08.

Let me see what we can do.

umarcor, 2023-05-22:
What a lovely attitude towards technical "problems" (i.e. challenges)! Never get tired of finding these interactions.

Paebbels, 2023-05-22:
This also explains why Vivado and Riviera-PRO have a wrong behavior, because they never changed it for VHDL-2008. I already filed it for Riviera-PRO, but I assume it won't make it into 2023.04 - currently in beta phase.

Paebbels, 2023-06-15:
The issue is now also filed as an issue at AMD Vivado and was converted today to a change request for 2023.2.

Paebbels, 2025-02-19:
@tgingold what is the state on this one?

As far as I know, it should be implemented in Vivado and Riviera-PRO.

tgingold, 2025-02-20:
I am not sure what can we do.  ghdl simply follows the LRMs.

Paebbels, 2025-02-20:
I think you fixed it already. I found 00834463be0057b444cb0e4fa9be0a1dd4e05eeb and aeb15c6d168fd032689c85fdb42e2a4904bfbd6f

I run the reproducer code and it doesn't give an error. So I assume it's all fixed and we can close it. Or have I missed an aspect to that question?

tgingold, 2025-02-20:
Do let's close it.

closed 2025-02-20 by tgingold