ghdl/ghdl #2448: and/or at portmapping inside a block statement doesn't work

sschmitz86, 2023-05-30:
**Description**
It seems that there are a problem with the port mapping inside a block statement.
I know that the block statement isn't commonly used but sometimes it's very handy to seperate code quick. 
The Workaround is quite easy but still a litte bit annoying.

**Example**

```vhdl
    Testenvironment : block
    begin

        -- clock generate
        --------------------------------------------------------
        clock : clk_i <= not clk_i after (CLK_PERIOD / 2);

        -- watchdog
        --------------------------------------------------------
        watchdog : test_runner_watchdog(runner, WATCHDOG_TIMEOUT);

        -- device under test
        --------------------------------------------------------
        device_under_test : debouncer
            generic map(
                DEBOUNCE_LENGTH => DEBOUNCE_LENGTH)
            port map(
                clk_i   => clk_i,
                reset_i => reset_i,
                key_i   => key_1 or key_2,
                key_o   => key_o);

    end block;
```

this one crashes -> ghdl.exe terminated by unhandled exception

```vhdl
    Testenvironment : block
    begin

        -- clock generate
        --------------------------------------------------------
        clock : clk_i <= not clk_i after (CLK_PERIOD / 2);

        -- watchdog
        --------------------------------------------------------
        watchdog : test_runner_watchdog(runner, WATCHDOG_TIMEOUT);

        -- device under test
        --------------------------------------------------------
        device_under_test : debouncer
            generic map(
                DEBOUNCE_LENGTH => DEBOUNCE_LENGTH)
            port map(
                clk_i   => clk_i,
                reset_i => reset_i,
                key_i   => key_i ,
                key_o   => key_o);

    key_i <= key_1 or key_2;

    end block;
```
this will generate no error.

**Context**

- OS: 
      Windows 10
- GHDL version:
      GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
      Compiled with GNAT Version: 12.2.0
      mcode code generator

diogratia, 2023-05-30:
It'd help if you add the reproducer and any information on the crash. The ability to have a non-static expression as an actual is a feature added in -2008:

**6.5.6.3 Port clauses**

>If the actual part of a given association element for a formal port of a block is the reserved word **inertial**
followed by an expression, or is an expression that is not globally static, then the given association element
is equivalent to association of the port with an anonymous signal implicitly declared in the declarative
region that immediately encloses the block. The signal has the same subtype as the formal port and is the
target of an implicit concurrent signal assignment statement of the form  
>  
>*anonymous* <= E;  
>  
>where E is the expression in the actual part of the given association element. The concurrent signal
assignment statement occurs in the same statement part as the block.

It's possible do so in a block statement reproducer built from your incomplete code:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity debouncer is
    generic (
        DEBOUNCE_LENGTH: natural := 42
    );
    port (
        clk_i:      in  std_logic;
        reset_i:    in  std_logic;
        key_i:      in  std_logic;
        key_o:      out std_logic
    );
end entity;

architecture dummy of debouncer is
begin
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity test_block is
end entity;

architecture foo of test_block is
    signal clk_i:   std_logic := '0';
    constant CLK_PERIOD:   time := 40 ns;
    signal reset_i: std_logic := '0';
    signal key_1:   std_logic := '0';
    signal key_2:   std_logic := '0';
    signal key_o:   std_logic;
    constant DEBOUNCE_LENGTH: natural := 43;
begin

Testenvironment:  block
begin

    -- clock generate
    --------------------------------------------------------
    clock : clk_i <= not clk_i after (CLK_PERIOD / 2) when now < 300 ns;

    -- watchdog
    --------------------------------------------------------
    -- watchdog : test_runner_watchdog(runner, WATCHDOG_TIMEOUT);

    -- device under test
    --------------------------------------------------------
    device_under_test : entity work.debouncer
        generic map(
            DEBOUNCE_LENGTH => DEBOUNCE_LENGTH)
        port map(
            clk_i   => clk_i,
            reset_i => reset_i,
            key_i   => key_1 or key_2, --std=08 
            key_o   => key_o);

    end block;
end architecture;
```
This analyzes, elaborates, and simulates with a recent version of ghdl.

```none
%: ghdl --version
GHDL 4.0.0-dev (3.0.0.r195.g038f55c37) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 llvm 15.0.7 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
```
Your version of ghdl:

```none
%: $GHDL --version
GHDL 3.0.0 (2.0.0.r1417.g7de967c51) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
%: 
%: $GHDL -a testenvironment1.vhdl
testenvironment1.vhdl:55:30:error: actual expression must be globally static
            key_i   => key_1 or key_2,
                             ^
%: $GHDL -a --std=08 testenvironment1.vhdl
%: $GHDL -r --std=08 test_block
%: 
```
While this is the MacOS mcode version it demonstrates that there is nothing in the version of ghdl you are using that will stop the -2008 supported feature from working.  That leaves something relating to your incomplete example, the use of vunit, damage to your ghdl installed image or use of the specific target platform.

sschmitz86, 2023-05-31:
Thank you for your effort, i think you're right !
I run several test with and without vunit. The reproducer are attached

with vunit:
```
python .\run.py

Starting lib.debouncer_tb.reset
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_tb.reset_47c00e56a4f1ffb25f724f724275f107bd059de8\output.txt
pass (P=1 S=0 F=0 T=6) lib.debouncer_tb.reset (0.9 seconds)

Starting lib.debouncer_tb.key_debounce_push
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_tb.key_debounce_push_bb2e4fdf69c7bd50634cf209ecc522570c4aa89c\output.txt
pass (P=2 S=0 F=0 T=6) lib.debouncer_tb.key_debounce_push (1.0 seconds)

Starting lib.debouncer_tb.key_debounce_release
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_tb.key_debounce_release_630f0ff595197b7be82d88c76b63bbc3fec9be33\output.txt
pass (P=3 S=0 F=0 T=6) lib.debouncer_tb.key_debounce_release (1.0 seconds)

Starting lib.debouncer_not_ok_tb.reset
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_not_ok_tb.reset_317fc5de1a0f1c29d66ecabfaeb93ce39b29d532\output.txt

Execution of C:\Program Files\GHDL\bin\ghdl.exe terminated by unhandled exception
raised CONSTRAINT_ERROR : grt-signals.adb:1175 access check failed
Load address: 0x9d0000
Call stack traceback locations:
0xa24705 0xa24985 0xa250f0 0xa253eb 0x9d329a 0xa4e880 0xa4e92a 0xd76773 0xcc1038 0xcb304a 0xd76ed1 0x9d2dea 0x9d12ee 0x76fa00c7 0x77c17b4c 0x77c17b1c
fail (P=3 S=0 F=1 T=6) lib.debouncer_not_ok_tb.reset (0.9 seconds)

Starting lib.debouncer_not_ok_tb.key_debounce_push
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_not_ok_tb.key_debounce_push_1f9f2c3a482d647facc4d07301eeebdb2709f904\output.txt

Execution of C:\Program Files\GHDL\bin\ghdl.exe terminated by unhandled exception
raised CONSTRAINT_ERROR : grt-signals.adb:1175 access check failed
Load address: 0x9d0000
Call stack traceback locations:
0xa24705 0xa24985 0xa250f0 0xa253eb 0x9d329a 0xa4e880 0xa4e92a 0xd76773 0xcc1038 0xcb304a 0xd76ed1 0x9d2dea 0x9d12ee 0x76fa00c7 0x77c17b4c 0x77c17b1c
fail (P=3 S=0 F=2 T=6) lib.debouncer_not_ok_tb.key_debounce_push (0.9 seconds)

Starting lib.debouncer_not_ok_tb.key_debounce_release
Output file: C:\Users\sschmitz\Desktop\debouncer\vunit_out\test_output\lib.debouncer_not_ok_tb.key_debounce_release_3f97e1179f1d448409e879d016c5edbcbb564059\output.txt

Execution of C:\Program Files\GHDL\bin\ghdl.exe terminated by unhandled exception
raised CONSTRAINT_ERROR : grt-signals.adb:1175 access check failed
Load address: 0x9d0000
Call stack traceback locations:
0xa24705 0xa24985 0xa250f0 0xa253eb 0x9d329a 0xa4e880 0xa4e92a 0xd76773 0xcc1038 0xcb304a 0xd76ed1 0x9d2dea 0x9d12ee 0x76fa00c7 0x77c17b4c 0x77c17b1c
fail (P=3 S=0 F=3 T=6) lib.debouncer_not_ok_tb.key_debounce_release (1.0 seconds)

==== Summary ========================================================
pass lib.debouncer_tb.reset                       (0.9 seconds)
pass lib.debouncer_tb.key_debounce_push           (1.0 seconds)
pass lib.debouncer_tb.key_debounce_release        (1.0 seconds)
fail lib.debouncer_not_ok_tb.reset                (0.9 seconds)
fail lib.debouncer_not_ok_tb.key_debounce_push    (0.9 seconds)
fail lib.debouncer_not_ok_tb.key_debounce_release (1.0 seconds)
=====================================================================
pass 3 of 6
fail 3 of 6
=====================================================================
Total time was 5.6 seconds
Elapsed time was 5.7 seconds
=====================================================================
Some failed!

```

run ghdl only works
```
ghdl -a --std=08 debouncer.vhd debouncer_no_vunit_tb.vhd
ghdl -e --std=08  debouncer_no_vunit_tb
ghdl -r --std=08 debouncer_no_vunit_tb --vcd=debouncer.vcd --stop-time=1000ns
C:\Program Files\GHDL\bin\ghdl.exe:info: simulation stopped by --stop-time @997600ps
```

[debouncer.vhd.txt](https://github.com/ghdl/ghdl/files/11610256/debouncer.vhd.txt)
[debouncer_no_vunit_tb.vhd.txt](https://github.com/ghdl/ghdl/files/11610257/debouncer_no_vunit_tb.vhd.txt)
[debouncer_not_ok_tb.vhd.txt](https://github.com/ghdl/ghdl/files/11610258/debouncer_not_ok_tb.vhd.txt)
[debouncer_tb.vhd.txt](https://github.com/ghdl/ghdl/files/11610259/debouncer_tb.vhd.txt)
[run.py.txt](https://github.com/ghdl/ghdl/files/11610260/run.py.txt)

sschmitz86, 2023-05-31:
or maybe not, debouncer_no_vunit_tb should not trigger any error anyway.
This is the one with including the port mapping in the block statement.

```
PS C:\Users\sschmitz\Desktop\debouncer> ghdl -a --std=08 debouncer.vhd debouncer_no_vunit_not_ok_tb.vhd
PS C:\Users\sschmitz\Desktop\debouncer> ghdl -a --std=08 debouncer.vhd debouncer_no_vunit_not_ok_tb.vhd
PS C:\Users\sschmitz\Desktop\debouncer> ghdl -r --std=08 debouncer_no_vunit_not_ok_tb --vcd=debouncer.vcd
C:\Program Files\GHDL\bin\ghdl.exe:error: NULL access dereferenced
  from: [unknown caller]
  from: work.debouncer_no_vunit_not_ok_tb(testbench).testenvironment.device_under_test.key_iINERTIAL at debouncer_no_vunit_not_ok_tb.vhd:65
C:\Program Files\GHDL\bin\ghdl.exe:error: simulation failed
```
[debouncer_no_vunit_not_ok_tb.vhd.txt](https://github.com/ghdl/ghdl/files/11610485/debouncer_no_vunit_not_ok_tb.vhd.txt)

commit b20d25de (2023-06-01): trans-chap4: fix internal association in nested scopes.

Fix #2448
closed 2023-06-01 by tgingold