ghdl/ghdl #2449: Dumping types derived from generics causes crashes

sean-anderson-seco, 2023-05-30:
**Description**

Dumping the signals in the below unit with `--wave` causes the unit to abort with the following backtrace:

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7da4859 in __GI_abort () at abort.c:79
#2  0x00005555555f07c4 in __gnat_last_chance_handler () at ./src/grt/grt-cgnatrts.c:23
#3  0x0000555555646160 in grt.waves.foreach_scalar_signal_number.range_pos_to_val (rti=0x5555556ac400 <std__standard__integer__BT__RTI>, rng=0x5555556ba75c, pos=0, val=...) at ./src/grt/grt-rtis_utils.adb:293
#4  0x000055555564643c in grt.waves.foreach_scalar_signal_number.pos_to_vstring (vstr=..., rti=0x5555556ac400 <std__standard__integer__BT__RTI>, rng=0x5555556ba75c, pos=0) at ./src/grt/grt-rtis_utils.adb:334
#5  0x00005555556467aa in grt.waves.foreach_scalar_signal_number.handle_array_1 (arr_rti=0x5555556aeb20 <ieee__std_logic_1164__std_ulogic_vector__RTI>, index=0) at ./src/grt/grt-rtis_utils.adb:379
#6  0x0000555555646c62 in grt.waves.foreach_scalar_signal_number.handle_any (rti=0x5555556af820 <counter__pkg__c_type__RTI>) at ./src/grt/grt-rtis_utils.adb:454
#7  0x000055555564608f in grt.waves.foreach_scalar_signal_number (ctxt=..., obj_type=0x5555556af820 <counter__pkg__c_type__RTI>, obj_addr=(const system__address) 0x5555556ba7a0, is_sig=true, param=0) at ./src/grt/grt-rtis_utils.adb:507
#8  0x0000555555649c85 in grt.waves.write_signal_numbers (decl=...) at src/grt/grt-waves.adb:878
#9  0x0000555555649e18 in grt.waves.write_hierarchy_el (decl=...) at src/grt/grt-waves.adb:923
#10 0x0000555555649fec in grt.waves.wave_put_hierarchy_1 (inst=..., step=step_hierarchy, match_list=0x0) at src/grt/grt-waves.adb:971
#11 0x000055555564a27c in grt.waves.wave_put_hierarchy_block (inst=..., step=step_hierarchy, match_list=0x0) at src/grt/grt-waves.adb:1043
#12 0x000055555564a3c1 in grt.waves.wave_put_hierarchy (root=..., step=step_hierarchy) at src/grt/grt-waves.adb:1075
#13 0x000055555564cc8d in grt.waves.write_hierarchy (root=...) at src/grt/grt-waves.adb:1670
#14 0x000055555564d0fe in grt.waves.wave_start () at src/grt/grt-waves.adb:1755
#15 0x00005555555f7d40 in grt.hooks.call_start_hooks () at src/grt/grt-hooks.adb:135
#16 0x00005555555d78e8 in <__ghdl_simulation_init> () at src/grt/grt-processes.adb:1128
#17 0x00005555555d7ca5 in grt.processes.simulation () at src/grt/grt-processes.adb:1278
#18 0x00005555555f05b7 in __ghdl_run_through_longjump (func=0x5555555d7c98 <grt.processes.simulation>) at ./src/grt/config/jumps.c:234
#19 0x00005555555f011e in grt.main.run_simul () at src/grt/grt-main.adb:163
#20 0x00005555555f0200 in grt.main.run () at src/grt/grt-main.adb:198
#21 0x000055555560d0a1 in ghdl_main (argc=2, argv=0x7fffffffe038) at src/grt/ghdl_main.adb:63
#22 0x00005555555f0866 in main ()
```

Frame 3 looks like

```
gdb$ frame
Stack level 3, frame at 0x7fffffffd870:
 rip = 0x5555555f6e60 in grt.waves.foreach_scalar_signal_number.range_pos_to_val ./src/grt/grt-rtis_utils.adb:293); saved rip = 0x5555555f713c
 called by frame at 0x7fffffffd930, caller of frame at 0x7fffffffd820
 source language ada.
 Arglist at 0x7fffffffd860, args: rti=0x5555556569e0 <std__standard__integer__BT__RTI>, rng=0x55555566375c, pos=0, val=...
 Locals at 0x7fffffffd860, Previous frame's sp is 0x7fffffffd870
 Saved registers:
  rbx at 0x7fffffffd858, rbp at 0x7fffffffd860, rip at 0x7fffffffd868
rti = 0x5555556569e0 <std__standard__integer__BT__RTI>
rng = 0x55555566375c
pos = 0
val = (
  b1 => true, 
  e8 => 1, 
  e32 => 1, 
  i32 => 1, 
  i64 => 1, 
  f64 => 4.9406564584124654e-324
)
No locals.
gdb$ p *rti
$8 = (
  kind => ghdl_rtik_type_i32, 
  depth => 0, 
  mode => 0, 
  max_depth => 0
)
gdb$ p *rng
$9 = (
  b1 => (
    left => false, 
    right => false, 
    dir => dir_to, 
    len => 10
  ), 
  e8 => (
    left => 0, 
    right => 0, 
    dir => dir_to, 
    len => 10
  ), 
  e32 => (
    left => 0, 
    right => 10, 
    dir => 10, 
    len => 80
  ), 
  i32 => (
    left => 0, 
    right => 10, 
    dir => 10, 
    len => 80
  ), 
  p64 => (
    left => 42949672960, 
    right => 343597383690, 
    dir => 9, 
    len => 0
  ), 
  f64 => (
    left => 2.1219957909652723e-313, 
    right => 1.6975966328216244e-312, 
    dir => 9
  )
)
```

Similarly, dumping the signals with `--vcd` or `--fst` causes a segmentation fault with the following backtrace:

```
#0  0x00005555555eb8cc in grt.vcd.verilog_wire_val (info=..., idx=10) at src/grt/grt-vcd.adb:505
#1  0x00005555555eff16 in grt.fst.fst_put_var (i=0) at src/grt/grt-fst.adb:606
#2  0x00005555555f0259 in grt.fst.fst_cycle () at src/grt/grt-fst.adb:688
#3  0x00005555555886e5 in grt.hooks.call_cycle_hooks () at ./src/grt/grt-hooks.adb:81
#4  <__ghdl_simulation_init> () at src/grt/grt-processes.adb:1164
#5  0x00005555555889a5 in grt.processes.simulation () at src/grt/grt-processes.adb:1278
#6  0x00005555555a0e0d in __ghdl_run_through_longjump (func=0x555555588998 <grt.processes.simulation>) at ./src/grt/config/jumps.c:234
#7  0x00005555555a0974 in grt.main.run_simul () at src/grt/grt-main.adb:163
#8  0x00005555555a0a56 in grt.main.run () at src/grt/grt-main.adb:198
#9  0x00005555555bd8f7 in ghdl_main (argc=2, argv=0x7fffffffe038) at src/grt/ghdl_main.adb:63
#10 0x00005555555a10bc in main ()
```

Frame 0 looks like

```
gdb$ frame
Stack level 0, frame at 0x7fffffffddb0:
 rip = 0x5555555eb8cc in grt.vcd.verilog_wire_val (src/grt/grt-vcd.adb:505); saved rip = 0x5555555ecc3d
 called by frame at 0x7fffffffde00
 source language ada.
 Arglist at 0x7fffffffdda0, args: info=..., idx=10
 Locals at 0x7fffffffdda0, Previous frame's sp is 0x7fffffffddb0
 Saved registers:
  rbx at 0x7fffffffdd78, rbp at 0x7fffffffdda0, r12 at 0x7fffffffdd80, r13 at 0x7fffffffdd88, r14 at 0x7fffffffdd90, r15 at 0x7fffffffdd98, rip at 0x7fffffffdda8
info = (
    vtype => vcd_stdlogic_vector, 
    val => vcd_effective, 
    ptr => (system.address) 0x555555664a50, 
    vec_range => 0x555555664a0c
  )
idx = 10
```

**Expected behaviour**

No crash.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

package pkg is
	generic (LENGTH: positive);
	subtype c_type is std_logic_vector(LENGTH - 1 downto 0);
end;

entity counter is
end;

architecture rtl of counter is
	package pkg is new work.pkg generic map (LENGTH => 10);
	signal c: pkg.c_type;
begin
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl_gcc -a counter.vhd
ghdl_gcc --elab-run counter
./counter --wave=counter.ghw # or --vcd=counter.vcd
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin: Built from sources: 7d32c7b87c27bec344db2442e06997d4152523d3

umarcor, 2023-08-28:
Analysing requires `--std=08`, since generic packages are used.

On MSYS2 I get a crash when analysing the file:

```
# ghdl -a --std=08 counter.vhd 
get_object_type_staticness: cannot handle IIR_KIND_PACKAGE_HEADER (counter.vhd:2:9)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues     
GHDL release: 4.0.0-dev (3.0.0.r147.g6c56631a7) [Dunoon edition]  
Compiled with GNAT Version: 13.1.0
Target: x86_64-w64-mingw32
F:\antmicro\f4pga\fpga-topwrap\tmp-ghdl\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o counter.o counter.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Load address: 0x7ff64d7c0000
Call stack traceback locations:
0x7ff64d8a3b60 0x7ff64da04acb 0x7ff64da04f52 0x7ff64da0e4f2 0x7ff64d9f3dee 0x7ff64d9f414f 0x7ff64d9d9518 0x7ff64d9d964b 0x7ff64d9d9953 0x7ff64d9db939 0x7ff64d9dbe67 0x7ff64d9ee866 0x7ff64d9f1d1a 0x7ff64d9f2046 0x7ff64d9e2f66 0x7ff64d9e436e 0x7ff64d9babd9 0x7ff64dad9cfe 0x7ff64dadb716 0x7ff64dadcb47 0x7ff64d7c2d0c 0x7ff64d7c12ec 0x7ff64d7c1404 0x7fff13df7612 0x7fff144826af
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl --version
GHDL 4.0.0-dev (3.0.0.r147.g6c56631a7) [Dunoon edition]
 Compiled with GNAT Version: 13.1.0
 llvm 16.0.2 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Using a more recent version on a container with mcode backend, analysis and elaboration succeeds but run crashes:

```
# docker run --rm -itv /$(pwd)://wrk -w //wrk gcr.io/hdl-containers/ghdl bash
root@a342a980c2b9:/wrk# ghdl -a --std=08 counter.vhd 
root@a342a980c2b9:/wrk# ghdl --elab-run --std=08 counter --wave=counter.ghw

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (v3.0.0-424-g6a50056cb) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/wrk/
Command line:
ghdl --elab-run --std=08 counter --wave=counter.ghw
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-rtis_utils.adb:293 invalid data
******************************************************************
root@a342a980c2b9:/wrk# ghdl version
GHDL 4.0.0-dev (v3.0.0-424-g6a50056cb) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Similar container with LLVM backend does not crash, but execution is aborted and the waveform is empty:

```
# docker run --rm -itv /$(pwd)://wrk -w //wrk gcr.io/hdl-containers/ghdl/llvm bash
root@fd33a0f66afc:/wrk# ghdl version
GHDL 4.0.0-dev (v3.0.0-424-g6a50056cb) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 llvm 11.0.1 code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
root@fd33a0f66afc:/wrk# ghdl -a --std=08 counter.vhd 
root@fd33a0f66afc:/wrk# ghdl -e --std=08 counter
root@fd33a0f66afc:/wrk# ./counter --wave=counter.ghw
Aborted
root@fd33a0f66afc:/wrk# ls -la
total 1181
drwxrwxrwx 1 root root    4096 Aug 28 15:01 .
drwxr-xr-x 1 root root    4096 Aug 28 15:00 ..
-rwxr-xr-x 1 root root 1187072 Aug 28 15:00 counter
-rwxr-xr-x 1 root root       0 Aug 28 15:01 counter.ghw
-rwxr-xr-x 1 root root    6712 Aug 28 15:00 counter.o
-rwxr-xr-x 1 root root     256 Aug 28 14:11 counter.vhd
-rwxr-xr-x 1 root root    5272 Aug 28 15:00 e~counter.o
-rwxr-xr-x 1 root root     215 Aug 28 15:00 work-obj08.cf
```

dalex78, 2026-09-04:
# Issues #2449, #656 and #2416 — dumping a signal whose subtype comes from a package instance

Analysis by an AI agent. The root cause is pinned; the fix needs an RTI format change and is not attempted. What is fixed here is that the failure is no longer silent on gcc and llvm.

## The reproducer (shared by #2449 and #656)

```vhdl
package lfsr is
	generic (TAPS: bit_vector);
	subtype lfsr_t is bit_vector(TAPS'range);
end;

entity counter is
end;

architecture rtl of counter is
	package pkg is new work.lfsr generic map (TAPS => 10D"0");
	signal c: pkg.lfsr_t;
begin
end;
```

The design runs fine. Ask for a waveform and it does not:

| backend | `--wave` | `--vcd` | `--fst` |
|---|---|---|---|
| mcode | `CONSTRAINT_ERROR : grt-rtis_utils.adb:293 invalid data` + bug banner | `NULL access dereferenced` | passes |
| llvm-jit | same as mcode | `NULL access dereferenced` | `NULL access dereferenced` |
| gcc, llvm | SIGABRT, **no output at all** | `NULL access dereferenced` | `NULL access dereferenced` |

## #656 is the same bug

#656 is titled "RTIs for package instances" — tgingold renamed it after reproducing Ravenwater's design — and its signature is `grt-rtis_utils.adb:261 invalid data`, the same procedure as above at the line it had in 0.35 (`Range_Pos_To_Val`; it is 293 today, and a later reporter on the same thread quotes 292). Its closing note is:

> Note: this requires RTIs for package instances.

The reproducer above produces that signature. So #2449 and #656 are one defect, and it is exactly the one tgingold named.

Both reporters on #656 arrived the same way, through a design that runs until a waveform is asked for. Ravenwater said so explicitly:

> Isolated it to be related to the argument `--wave=posit_add.ghw`, that is, removing the wave generation yields a working simulator.

## #2416 is the same bug, in its library-level shape

#2416's signature is different -- `grt-waves.adb:831 access check failed`. In the version reported (3.0.0-dev, and identically in the `v2.0.0` tag) that line is the body of `Get_Signal_Number`:

```ada
   --  This is only valid during write_hierarchy.
   function Get_Signal_Number (Sig : Ghdl_Signal_Ptr) return Natural
   is
      function To_Integer_Address is new Ada.Unchecked_Conversion
        (Ghdl_Signal_Ptr, Integer_Address);
   begin
      return Natural (To_Integer_Address (Sig.Alink));
   end Get_Signal_Number;
```

"access check failed" there means `Sig` is not a signal: the wave writer was handed a pointer that does not point at one, so this is a garbage signal pointer rather than a garbage range. Its reporter never posted a reproducer -- brameret suspected records (*"Is it related to #2324? I have records in my VHDL"*), tried them and failed (*"For now, failed to reproduce with a simpler example including records"*), and the title blames the stack size. Nothing runnable ever reached the issue.

It is the same defect, reached through a package instantiated at **library level** instead of inside the architecture:

```vhdl
library ieee; use ieee.std_logic_1164.all;
package gp2 is
  generic (W : natural);
  subtype word is std_logic_vector (W - 1 downto 0);
end package;

package inst2 is new work.gp2 generic map (W => 4);

[... 79 lines cut ...]
void
__gnat_last_chance_handler (void)
{
  abort ();
}
```

GNAT calls it with the source location and line of the failed check, and both were being thrown away. That is why the table above says "no output at all" for gcc and llvm: the simulation died on the same `CONSTRAINT_ERROR` as mcode, and printed nothing. It now prints what GNAT passed:

```
ghdl: internal error: grt-rtis_utils.adb:293
Please report this bug on https://github.com/ghdl/ghdl/issues
```

This is not specific to #2449. Any internal error in grt reached from a gcc- or llvm-built simulation was silent in the same way; it is diagnosable now.

## Testing

`testsuite/gna/issue2449` runs the design without a dumper (which must work) and then with `--wave` and `--vcd`, asserting that they still fail -- so the day the RTI work is done, the test fails and says so rather than quietly passing. It deliberately does not match on a message, since the three backends fail differently. `libinst.vhd` covers the library-level shape, ie #2416's signature.

`sanity`, `gna` and `synth` pass on mcode, llvm and llvm-jit.

## Recommendation

Keep #2449, #656 and #2416 open as one issue for the RTI work; #656 already carries the right title for it. Before closing #2416 as a duplicate, it is worth asking brameret whether their design instantiates a generic package, since the connection is inferred from a matching signature and not from their sources.