ghdl/ghdl #2450: Ghdl with LLVM not handing custom output filename (-o ) to ghdl1-llvm

cweickhmann, 2023-05-31:
**Description**
It seems, ghdl compiled with LLVM does not hand the user defined output filename (from `-o` flag) to `ghdl1-llvm` upon analysis.

**Expected behaviour**
When invoking e.g. `ghdl -a --std=08 -v -o myownfilename.o file1.vhdl`, the output object file should be named `myownfilename.o`. Instead it's named whatever the source file is named.

**How to reproduce?**

```vhd :file: file1.vhd
entity some_entity is
end entity;

architecture rtl of some_entity is
begin
  process
  begin
    report "Hello!";
  end process;
end rtl;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a --std=08 -v -o file1_somename.o file1.vhdl
```

Output is:
```
/opt/ghdl/3.0.0/bin/ghdl1-llvm --std=08 -P/opt/ghdl/3.0.0/lib/ghdl/ieee/v08/ -P/opt/ghdl/3.0.0/lib/ghdl/ -fpic -c -o file1.o file1.vhdl
file1.vhdl:6:3:warning: infinite loop for this process without a wait statement [-Wno-wait]
  process
  ^
```

However, if I run the `ghdl1-llvm` command directly, modifying it as follows:
```sh :image: ghdl/ghdl:bullseye-mcode
ghdl1-llvm --std=08 -P/opt/ghdl/3.0.0/lib/ghdl/ieee/v08/ -P/opt/ghdl/3.0.0/lib/ghdl/ -fpic -c -o myownfilename.o file1.vhdl
```
it does work as expected.

**Workaround**: Either by using the `ghdl1-llvm` command directly, or simply renaming the output file.

**Context**

- GHDL 3.0.0 (tarball) [Dunoon edition]
  - Compiled with GNAT Version: 12.1.0
  - llvm 14.0.0 code generator
- OS: Ubuntu 22.04.2 LTS, Linux 6.1.0-1013-oem
- Origin:
  - [x] Released binaries: `https://github.com/ghdl/ghdl/archive/refs/tags/v3.0.0.tar.gz`

cweickhmann, 2023-06-01:
So, having the package inside it named differently does not pose a problem? Then indeed it's not a problem at all.
But what's the `-o` flag for if it's not working?

tgingold, 2023-06-02:
I will have a look and fix it (if easy).

tgingold, 2023-08-27:
I close this ticket as future versions may not anymore generate object code.  So any change would become useless.

closed 2023-08-27 by tgingold