ghdl/ghdl #2451: not a formal parameter

vicencb, 2023-06-01:
This code:
```vhdl
entity test is end entity;
architecture arch of test is
    signal s :bit;
    procedure proc1 is begin s <= '1'; end procedure;
begin
  process
    procedure proc2 is begin s <= '1'; end procedure;
  begin
    wait;
  end process;
end architecture;
```
when built with `ghdl -c test.vhd -e test`
reports `error: signal "s" is not a formal parameter` for proc1 and no error for proc2.

I think the expected behaviour for proc1 should be the same as for proc2.

Tested with: `GHDL 4.0.0-dev (3.0.0.r233.ga752f5971) [Dunoon edition]`

diogratia, 2023-06-02:
IEEE Std 1076-2008  
10.5.2.2 Executing a simple assignment statement  

>If a given procedure is declared by a declarative item that is not contained within a process statement, and if
a simple waveform assignment statement appears in that procedure, then the target of the simple waveform
assignment shall be a formal parameter of the given procedure or of a parent of that procedure, or an
aggregate of such formal parameters. Similarly, if a given procedure is declared by a declarative item that is
not contained within a process statement, and if a signal is associated with an inout or out mode signal
parameter in a subprogram call within that procedure, then the signal so associated shall be a formal
parameter of the given procedure or of a parent of that procedure.  
>  
>NOTE 1—These rules guarantee that the driver affected by a simple waveform assignment statement is always statically determinable if the simple waveform assignment appears within a given process (including the case in which it appears within a procedure that is declared within the given process). In this case, the affected driver is the one defined by the process; otherwise, the simple waveform assignment shall appear within a procedure, and the affected driver is the one passed to the procedure along with a signal parameter of that procedure. Simple force assignments and simple release assignments, on the other hand, do not involve drivers. Hence, the target of such an assignment occurring in a procedure not contained with a process statement need not be a signal parameter of the procedure.

Also see IEEE Std 1076-1993  8.4.1 Updating a projected output waveform which conveys the same requirements and Note. The requirement and Note are present in 8.4.1 of -1987 as well.

vicencb, 2023-06-02:
I checked the Std 1076-2008 document before posting the issue, but looked at the wrong place (4.3 Subprogram bodies).
Thank you for showing me the relevant part that applies to this case.
GHDL is clearly conformant, so, closing.

closed 2023-06-02 by vicencb