ghdl/ghdl #2453: pyunit tests fail during at the end of ghdl's build with python-pyvhdlmodel 0.26

OpusElectronics, 2023-06-05:
Context: ArchLinux, x86_64, AUR packages ghdl-mcode-git and  python-pyvhdlmodel-git
Latest git revision of both ghdl and python-pyvhdlmodel

Error message:
```
=========================================================== FAILURES ============================================================
___________________________________________________ CompileOrder.test_Encoder ___________________________________________________

self = <pyunit.dom.StopWatch.CompileOrder testMethod=test_Encoder>

    def test_Encoder(self):
        print()
        design = Design()
        design.LoadDefaultLibraries()
        t1 = time_perf_counter()
        for lib, file in self._stopwatchFiles:
            library = design.GetLibrary(lib)
            document = Document(self._sourceDirectory / file)
            design.AddDocument(document, library)
            print(dedent("""\
                file: {}
                  libghdl processing time: {:5.3f} us
                  DOM translation time:    {:5.3f} us
                """
                ).format(
                    document.Path,
                    document.LibGHDLProcessingTime * 10**6,
                    document.DOMTranslationTime * 10**6,
                )
            )
        pyGHDLTime = time_perf_counter() - t1
    
        design.Analyze()
    
        toplevel = [root.Value.Identifier for root in design.HierarchyGraph.IterateRoots()]
    
        print(dedent("""
            pyGHDL:
              sum:                       {:5.3f} us
            Analysis:
              default library load time: {:5.3f} us
              dependency analysis time:  {:5.3f} us
            Toplevel:                    {toplevel}
            Compile order:\
            """
            ).format(
                pyGHDLTime * 10**6,
                design._loadDefaultLibraryTime * 10**6,
                design._analyzeTime * 10**6,
                toplevel=", ".join(toplevel)
            )
        )
        for i, vertex in enumerate(design.IterateDocumentsInCompileOrder()):
>           print(f"  {i:<2}: {vertex.Value.Path.relative_to(Path.cwd())}")
E           AttributeError: 'Document' object has no attribute 'Value'

pyunit/dom/StopWatch.py:160: AttributeError
```

With python-pyvhdlmodel 0.25, all tests pass.

Paebbels, 2023-06-05:
GHDL - to be specific `pyGHDL.dom` - is locked to `pyVHDLModel` v0.25.x (see https://github.com/ghdl/ghdl/blob/master/pyGHDL/dom/requirements.txt#L3)

This constrained is not reflected by the maintainers of AUR packages. I know from previous complains the same is true for `pyTooling`. GHDL has a version limit to one major version, so a breaking change (major increment) doesn't effect GHDL. As I understood the complaint, this is also not reflected by AUR packages.

I'm not into AUR packages. Can this constrained be added?

There is ongoing work to update `pyGHDL.dom` with new features and fixes as well as bumping the `pyVHDLModel` version to v0.26 or v0.27.

OpusElectronics, 2023-06-05:
Thanks! I didn't quite know what the status was vs. pyVHDLModel 0.26.
(It's a bit unfortunate that they would break compatibility just by upping minor versions, but that's another story. These days, it looks like every project has a bit of its own way of handling versioning, and while the 25 and 26 in 0.25 and 0.26 pretty much look like minor versions to me, they may not for them. Or maybe the project is still too young to be stable.)

The issue with the AUR packages is a bit tricky. Those are -git packages, the PKGBUILD files have a specific revision attached to them (and indeed for pyVHDLModel it's a 0.25.x), but will still pull the latest git revisions when you install them.

I'm relatively familiar with AUR packages, but not enough for this particular point.
I suppose the -git packages are *meant* to pull the latest git revisions anyway and if they break, you're kinda on your own. Problem is that there is currently no "non-git" AUR package for either. I'll see how it should be done and will report back if that can help other users.

OpusElectronics, 2023-06-05:
So as has been suggested from a user on the AUR page for ghdl-mcode-git, the immediate workaround is just to bypass the final checks using the --nocheck option when building the package. Not ideal, but should be temporary.
Otherwise we would have to write a specific PKGBUILD for pyVHDLModel 0.25.

Paebbels, 2023-06-05:
pyVHDLModel is written by me :). Yes the project is not stable because not all VHDL constructs are modelled. For this to finalize, pyGHDL.dom - as an instance of the model - should be 98% completed. Otherwise it's hard to tell if a model of such a complexity works out.

Changing just the minor version on non backward compatible changes is according to semantic versioning because v1.0 is not reached. In such a case the API can get breaking changes. As a precaution, version limits have been added to GHDL, which work well in Windows and other distributions using wheel packages from PyPI.

The problem is clearly on the AUR / pacman side, because version constraints are not reflected by the package. If AUR would use PIP and PyPI the problem wouldn't exist because wheel packages adhere to version constraints. Why is it build from sources?

If the AUR package builds against latest Git, it will always break if one repository is ahead of the other. That problem is completely independent of semantic versioning, because version number are ignored.

Don't get me wrong I want to support and help solving the issue but the always Git strategy isn't working well for tightly coupled packages.

OpusElectronics, 2023-06-05:
You're right, but it's the whole idea behind "-git" AUR packages, to build with the latest git revision. And yes, it's quite easy to break this approach when dependencies are involved.

One can always set some dependencies to a fixed version in AUR packages instead of pulling them themselves from their latest git. It's just that in case of GHDL and pyVHDLModel, there just aren't any other packages available than the "-git" ones, in AUR or outside of it. It's all done by volunteers!

I could probably have a go at maintaining such more stable Arch packages when I'm a bit more familiar with PKGBUILD stuff, since I'm actively using GHDL. It may take a little while though.

Paebbels, 2023-06-06:
@skoehler any ideas from you about constraining the python package version?

I was working on an update for GHDL when I updated pyVHDLModel. But then I noticed a problem in pyTooling. So here I'm working on v5.0 with a new feature and bugfixes. The increment is needed because the bugfix breaks the API for some special case users. As I wrote more and more tests I discovered more problems (shouldn't have added more test cases 😄). I need to do it right otherwise pyTooling might need another breaking increment... So pyVHDLModel is blocked by that fix/change. If don't I can immediately roll out the matching pyGHDL changes.

I'm also open the improve my workflow considering the 3 Python packages. Please note pyVHDLModel is also used by other packages like pyVHDLParser (also mine 😊).

Paebbels, 2023-06-06:
What about the wheel approach? Can an AUR package install a wheel? If so GHDL comes from AUR and git latest sources and Python support packages from PyPI.

marzoul, 2023-06-06:
Hi, in AUR packages there is also the possibility of restricting dependencies by specifying version suffix in the array of dependencies. for `ghdl-mcode-git` It is already the case with `'python-pyvhdlmodel>=0.12'`.
The general behavior that I see in Arch packages, when a package known to be a sensitive dependency is upgraded, they create for the sake of the transition one package locked on the previous version (or usable API, whatever) and add the version in the package name, such as `python-pyvhdlmodel025` (note the non-git package name). That package is then marked with `provides=('python-pyvhdlmodel=0.25')` and also marked as conflicting with other variants of the package. That way dependencies can be handled cleanly by the package manager.
In general that is done _after_ a compilation issue such as the present one has been experienced ;-)
Regards

marzoul, 2023-06-06:
How to specify version dependencies in a PKGBUILD :
[https://wiki.archlinux.org/title/PKGBUILD#depends]()

OpusElectronics, 2023-06-06:
Yes, one would have to add a dependency in the 'checkdepends' list (that's how you can specify a range of versions as explained in the article above), so for instance add  'python-pyvhdlmodel<0.26' in the list.

But, that's not all that would be needed to "fix" the problem, as now we'd still need a python-pyvhdlmodel package that can build from a 0.25.x version, and so we'd still need to create an additional package, with a fixed version, for it, instead of a -git package.

As I said, '-git' packages are made to pull the latest revisions, so you're supposed to deal with it as a user. And we can deal with it.

The thing that would be a great improvement would be the ability to downgrade to an earlier revision from a '-git' package, which doesn't seem to be possible at the moment. *That* is the problematic point. Maybe I'm missing how to do it though, but I don't think it's possible.

Since I recently installed it, it did pull python-pyvhdlmodel 0.26 and I had no choice, because I didn't have this package installded previously. But '-git' packages are not updated by default unless you force update them. So, if I had python-pyvhdlmodel 0.25 already installed (which was not my case) and only forced the update of ghdl-mcode-git, it would have not triggered the update of python-pyvhdlmodel, and things would have been fine.