ghdl/ghdl #2455: Bug with incomplete case statement

psi-chuffine, 2023-06-13:
**How to reproduce?**
I get a bug notice for this function, bug is not present if I add an others case to the switch statement.

Bug
```vhd 
  function convert_bcd (
    bcd : std_logic_vector
  )
  return unsigned is

    variable v_ret_val : integer;

  begin

    case bcd'length is

      when 1 to 3 =>

        v_ret_val := to_integer(unsigned(bcd(bcd'high downto 0)));

      when 4 to 7 =>

        v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                     + to_integer(unsigned(bcd(bcd'high downto 4))) * 10;

      when 8 to 9 =>

        v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                     + to_integer(unsigned(bcd(7 downto 4))) * 10
                     + to_integer(unsigned(bcd(bcd'high downto 8))) * 100;

    end case;

    return to_unsigned(v_ret_val, bcd'length);

  end function;
```

No Issue
```vhd 
  function convert_bcd (
    bcd : std_logic_vector
  )
  return unsigned is

    variable v_ret_val : integer;

  begin

    case bcd'length is

      when 1 to 3 =>

        v_ret_val := to_integer(unsigned(bcd(bcd'high downto 0)));

      when 4 to 7 =>

        v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                     + to_integer(unsigned(bcd(bcd'high downto 4))) * 10;

      when 8 to 9 =>

        v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                     + to_integer(unsigned(bcd(7 downto 4))) * 10
                     + to_integer(unsigned(bcd(bcd'high downto 8))) * 100;

      when others =>

        v_ret_val := 0;

    end case;

    return to_unsigned(v_ret_val, bcd'length);

  end function;
```

**Context**
Please, provide the following information:

- OS: Ubuntu (WSL2)
- Origin:  OSS-CAD-Suite

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
SBY 14:03:01 [irig_core_bmc] base: ******************** GHDL Bug occurred ***************************
SBY 14:03:01 [irig_core_bmc] base: Please report this bug on https://github.com/ghdl/ghdl/issues
SBY 14:03:01 [irig_core_bmc] base: GHDL release: 4.0.0-dev (3.0.0.r199.g71981ac7f.dirty) [Dunoon edition]
SBY 14:03:01 [irig_core_bmc] base: Compiled with unknown compiler version
SBY 14:03:01 [irig_core_bmc] base: Target: x86_64-linux-gnu
SBY 14:03:01 [irig_core_bmc] base: /mnt/c/HDL/irig-b-receiver/bench/formal/irig_core/irig_core_bmc/src/
SBY 14:03:01 [irig_core_bmc] base: Command line:
SBY 14:03:01 [irig_core_bmc] base: Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
SBY 14:03:01 [irig_core_bmc] base: Exception information:
SBY 14:03:01 [irig_core_bmc] base: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-vhdl_stmts.adb:1718
SBY 14:03:01 [irig_core_bmc] base: ******************************************************************
```

psi-chuffine, 2023-06-13:
I believe this is a synthesis issue.

Reproducer:
``` vhd :file: test_case.vhd
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

entity test is
port (
    i_clk : in    std_logic;
    i_data: in std_logic_vector(10 downto 0);
    o_data: out unsigned(10 downto 0)
);
end entity test;

architecture rtl of test is

    function convert_bcd (
        bcd : std_logic_vector
      )
      return unsigned is
    
        variable v_ret_val : integer;
    
      begin
    
        case bcd'length is
    
          when 1 to 3 =>
    
            v_ret_val := to_integer(unsigned(bcd(bcd'high downto 0)));
    
          when 4 to 7 =>
    
            v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                         + to_integer(unsigned(bcd(bcd'high downto 4))) * 10;
    
          when 8 to 9 =>
    
            v_ret_val := to_integer(unsigned(bcd(3 downto 0)))
                         + to_integer(unsigned(bcd(7 downto 4))) * 10
                         + to_integer(unsigned(bcd(bcd'high downto 8))) * 100;

          --when others => -- Uncomment to make functional
            --v_ret_val := 0;
        end case;
    
        return to_unsigned(v_ret_val, bcd'length);
    
      end function;

  begin

    process (i_clk) is
    begin
        o_data <= convert_bcd(i_data);
    end process;

end architecture;
```

SBY
```vhd :file: test.sby
[tasks]
bmc d_25

[options]
bmc: mode bmc

[engines]
smtbmc

[script]
ghdl --std=08 test_case.vhd -e test
prep -top test

[files]
test_case.vhd
```
Command:
sby --yosys "yosys -m ghdl" -f test.sby

Paebbels, 2023-06-14:
`'length` returns an integer. So the range is 2**31 elements. A case must cover all cases. So others is needed.

tgingold, 2023-06-14:
Yes sure.  Thank you for the issue.