ghdl/ghdl #2456: Formal verification doesn't find assert within function

Topi-ab, 2023-06-14:
**Description**
Assert within function (which is called) isn't catched during formal run.

**Expected behaviour**
Assert being generated.

**How to reproduce?**
Example project in:
https://github.com/Topi-ab/formal_assert_in_function_tester/

**Context**
Please, provide the following information:

- OS: Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Latest hdlc/formal:all docker image (June 14, 2023)

Note: The synthesized netlist seems to have postponed assert correctly in place, for some reason sby doesn't catch it.

tgingold, 2023-06-14:
Using z3 and commenting out the process, I don't get any error.

What is the output of sby ?

Topi-ab, 2023-06-14:
With commented process the output of sby (with "smtbbc z3") is:

SBY 17:21:20 [assert_in_function_tester_cover] Removing directory '/root/assert_in_function_tester_cover'.
SBY 17:21:20 [assert_in_function_tester_cover] Copy '/root/assert_in_function_tester.vhdl' to '/root/assert_in_function_tester_cover/src/assert_in_function_tester.vhdl'.
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: smtbmc z3
SBY 17:21:20 [assert_in_function_tester_bmc] Removing directory '/root/assert_in_function_tester_bmc'.
SBY 17:21:20 [assert_in_function_tester_bmc] Copy '/root/assert_in_function_tester.vhdl' to '/root/assert_in_function_tester_bmc/src/assert_in_function_tester.vhdl'.
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: smtbmc z3
SBY 17:21:20 [assert_in_function_tester_cover] base: starting process "cd assert_in_function_tester_cover/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 17:21:20 [assert_in_function_tester_bmc] base: starting process "cd assert_in_function_tester_bmc/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 17:21:20 [assert_in_function_tester_cover] base: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_cover] prep: starting process "cd assert_in_function_tester_cover/model; yosys -m ghdl -ql design_prep.log design_prep.ys"
SBY 17:21:20 [assert_in_function_tester_bmc] base: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_bmc] prep: starting process "cd assert_in_function_tester_bmc/model; yosys -m ghdl -ql design_prep.log design_prep.ys"
SBY 17:21:20 [assert_in_function_tester_cover] prep: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_cover] smt2: starting process "cd assert_in_function_tester_cover/model; yosys -m ghdl -ql design_smt2.log design_smt2.ys"
SBY 17:21:20 [assert_in_function_tester_cover] smt2: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: starting process "cd assert_in_function_tester_cover; yosys-smtbmc -s z3 --presat -c --noprogress -t 50  --append 0 --dump-vcd engine_0/trace%.vcd --dump-yw engine_0/trace%.yw --dump-vlogtb engine_0/trace%_tb.v --dump-smtc engine_0/trace%.smtc model/design_smt2.smt2"
SBY 17:21:20 [assert_in_function_tester_bmc] prep: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_bmc] smt2: starting process "cd assert_in_function_tester_bmc/model; yosys -m ghdl -ql design_smt2.log design_smt2.ys"
SBY 17:21:20 [assert_in_function_tester_bmc] smt2: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: starting process "cd assert_in_function_tester_bmc; yosys-smtbmc -s z3 --presat --noprogress -t 50  --append 0 --dump-vcd engine_0/trace.vcd --dump-yw engine_0/trace.yw --dump-vlogtb engine_0/trace_tb.v --dump-smtc engine_0/trace.smtc model/design_smt2.smt2"
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: ##   0:00:00  Solver: z3
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Solver: z3
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: ##   0:00:00  Status: passed
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assumptions in step 0..
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_cover] engine_0: Status returned by engine: pass
SBY 17:21:20 [assert_in_function_tester_cover] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:21:20 [assert_in_function_tester_cover] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:21:20 [assert_in_function_tester_cover] summary: engine_0 (smtbmc z3) returned pass
SBY 17:21:20 [assert_in_function_tester_cover] summary: engine_0 did not produce any traces
SBY 17:21:20 [assert_in_function_tester_cover] DONE (PASS, rc=0)
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assertions in step 0..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assumptions in step 1..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assertions in step 1..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assumptions in step 2..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assertions in step 2..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assumptions in step 3..
...
...
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assertions in step 48..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assumptions in step 49..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Checking assertions in step 49..
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: ##   0:00:00  Status: passed
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: finished (returncode=0)
SBY 17:21:20 [assert_in_function_tester_bmc] engine_0: Status returned by engine: pass
SBY 17:21:20 [assert_in_function_tester_bmc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:21:20 [assert_in_function_tester_bmc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 17:21:20 [assert_in_function_tester_bmc] summary: engine_0 (smtbmc z3) returned pass
SBY 17:21:20 [assert_in_function_tester_bmc] summary: engine_0 did not produce any traces
SBY 17:21:20 [assert_in_function_tester_bmc] DONE (PASS, rc=0)

Topi-ab, 2023-06-14:
As I understand, it should throw an assert at clock #5.

tgingold, 2023-06-14:
Ah yes, thank you for the hint.

tgingold, 2023-06-15:
Should now be fixed.  Thank you for the issue and your patience.

closed 2023-06-15 by tgingold

Topi-ab, 2023-06-16:
Was this included in the latest (available on June 16 2023) docker push?

Tested with that release, and no difference seen compared to previous release.

tgingold, 2023-06-16:
Which version of ghdl do you have ?

Topi-ab, 2023-06-16:
ghdl --version
GHDL 4.0.0-dev (v3.0.0-266-g3ce6e7562) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tgingold, 2023-06-16:
Yes, it's too old.  HEAD is at r305

umarcor, 2023-06-18:
@Topi-ab I triggered CI to update hdlc/formal images. The current version is:

```
$ docker run --rm -it hdlc/formal ghdl version
Emulate Docker CLI using podman. Create /etc/containers/nodocker to quiet msg.
GHDL 4.0.0-dev (v3.0.0-307-gb9bae14a9) [Dunoon edition]
 Compiled with GNAT Version: 9.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2023 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

In the future, if you need/want some update after Tristan pushes some fix here, please ping me.
By default, images in hdl/containers are updated once a week, but I can manually trigger then if needed.

Topi-ab, 2023-06-18:
Now it seems to be working.

Thanks a lot for the effort, this tool is getting better and better...

commit 0f2ab810 (2023-06-15): synth-environment: propagate enable wires.  Fix #2456

Add comments