ghdl/ghdl #2457: cocotb cosimulation with GHDL

firaitz, 2023-06-16:
Hi, I am trying to cosimulate a VHDL design using GHDL and Cocotb. This design needs a library of altera called lpm. I have already seen that these types of libraries are supported in GHDL, by running a configuration script on the path of the cosimulations workspace. I suppose that the script should create the needed libraries in the working path. However, when I try to run that script the following error message appears:
`~$ /usr/local/lib/ghdl/vendors/compile-altera.sh --all
Loading environment...
[ERROR] Path '/opt/altera/16.0/quartus/eda/sim_lib' does not exist.
`
Does anyone know how to solve this problem?

firaitz, 2023-06-16:
Okay, I was finally able to specify the correct path where the needed files were, and a new folder named altera has been created on the cosimulation working path. However, there is still a problem with the compilation. When executing the make file there is still an error caused by the non-recognition of the lpm library, which is already created by the last script mentioned. Here it is an example of the makefile. 

SIM ?= ghdl
EXTRA_ARGS += --std=08 

TOPLEVEL_LANG ?= vhdl

HDL_LIBRARY += ($(PWD)/altera/lpm)

VHDL_SOURCES += $(PWD)/example.vhd
VHDL_SOURCES += $(PWD)/top.vhd

TOPLEVEL = top

MODULE = pythontb

include $(shell cocotb-config --makefiles)/Makefile.sim

Paebbels, 2023-06-16:
I don't know how to setup creepy makefiles for CoCoTb, but I assume the `$(PWD)/altera/lpm` value is passed as `-P$(PWD)/altera/lpm` to GHDL.

GHDL will find libraries based on there name or by specifying the correct library and version number. so either `$(PWD)/altera` or `$(PWD)/altera/lpm/v08`. I suggest to user the former style. Here it will also find other Intel libraries if needed.

firaitz, 2023-06-20:
I have finally solved the problem with your given help. Thank you. The main problem here was an incorrect compilation of the libraries that were needed for my VHDL files. In fact, by running `/usr/local/lib/ghdl/vendors/compile-altera.sh --altera --source "/home/iraitz/intelFPGA_lite/17.0/quartus/eda/sim_lib" --vhdl2008` a folder with the libraries is created on the working path. Then by passing the extra argument `-P$(PWD)/altera/lpm/v08` to GHDL in the Makefile everything works great.

closed 2023-06-20 by firaitz