ghdl/ghdl #2459: translate_range_length: cannot handle IIR_KIND_REVERSE_RANGE_ARRAY_ATTRIBUTE

reprographix, 2023-06-23:
**Description**
It seems like GHDL is unable to cope with such expressions:
```
(var'reverse_range => '0');
```

**Expected behaviour**
I expect this piece of code to work properly as it does in other simulators and synthesizers.

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end entity;

architecture a of ent is
   signal sig : std_logic_vector(1 downto 0);
   variable var : std_logic_vector(0 to 1);
begin
  sig <= (var'reverse_range => '0');
end;
```

```sh :image: ghdl/ghdl:bullseye-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `3.0.0.r307.gb9bae14a9`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
translate_range_length: cannot handle IIR_KIND_REVERSE_RANGE_ARRAY_ATTRIBUTE (xxx:746:90)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 4.0.0-dev (3.0.0.r307.gb9bae14a9) [Dunoon edition]
Compiled with GNAT Version: 9.4.0
Target: x86_64-linux-gnu
xxx
Command line:
/usr/local/bin/ghdl -r --workdir=sim_build -Psim_build --work=work xxx --vpi=xxx/cocotb/libs/libcocotbvpi_ghdl.so -t 1fs
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x56146a689a68 0x56146a8c7fd1 0x56146a8d2687 0x56146a8d1db0 0x56146a8d3ed9 0x56146a8d3f92 0x56146a8d79d5 0x56146a8d7c59 0x56146a8cc595 0x56146a8cca4f 0x56146a8cbeeb 0x56146a8cc033 0x56146a8cbf77 0x56146a8cb290 0x56146a8cf55b 0x56146a8d7e28 0x56146a8d17f2 0x56146a88f306 0x56146a89fc1a 0x56146a89fd9d 0x56146a913b89 0x56146a8bfe15 0x56146a9152b6 0x56146a91541b 0x56146a87c726 0x56146a87b568 0x56146a888516 0x56146a924223 0x56146a859485 0x56146a848691 0x56146a92569a 0x56146a3d7237 0x7f2e10b7d081 0x56146a3d52cc 0xfffffffffffffffe
******************************************************************
```

**Additional context**

This fix seems to work, but I'm not sure it is the right thing to do:
```diff
diff --git a/src/vhdl/translate/trans-chap7.adb b/src/vhdl/translate/trans-chap7.adb
index 8ac6ab85a..dc0825fc6 100644
--- a/src/vhdl/translate/trans-chap7.adb
+++ b/src/vhdl/translate/trans-chap7.adb
@@ -779,6 +779,8 @@ package body Trans.Chap7 is
             return Translate_Range_Expression_Length (Expr);
          when Iir_Kind_Range_Array_Attribute =>
             return Chap14.Translate_Length_Array_Attribute (Expr, Null_Iir);
+         when Iir_Kind_Reverse_Range_Array_Attribute =>
+            return Chap14.Translate_Length_Array_Attribute (Expr, Null_Iir);
          when others =>
             Error_Kind ("translate_range_length", Expr);
       end case;
```

tgingold, 2023-06-24:
Yes, you're patch is correct.  I will apply it.

Thank you for the issue!

tgingold, 2023-06-25:
However, the reproducer is not correct (you cannot declare non-shared variables in a block)

closed 2023-06-25 by tgingold

reprographix, 2023-06-26:
You're right about the reproducer, I should have wrapped things in a process. Thanks for integrating the patch so quick!

commit 5c8c2887 (2023-06-25): trans-chap7.adb(translate_range_length): handle reverse_range

Fix #2459